OctoPrint/OctoPrint-FirmwareUpdater #125: Upgrade ANET A8

sroselli, 2020-03-26:
Hi when upgrade a firmware marlin on the anet a8 (original) i receive this message :

Flashing failed
Avrdude says: 'not in sync: resp=0xd6
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x6f
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x80
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xbf
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xae
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xd6
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x02
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x21
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xcc
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xd6

avrdude done. Thank you.'

Please Help me

cocanut, 2020-03-30:
I recently also updated Marlin and have similar issues:
```
# Software
Marlin 2.0.5.2
OctoPi 0.17.0 (installed via sd card flash)
Octoprint 1.4.0
Firmware-Updater 1.6.1 (installed via plugin manager)
avrdude 6.3-20171130 (installed via 'apt install avrdude')

# Settings
Flash method: avrdude
AVR MCU: ATmega1284p
Path to avrdude: /usr/bin/avrdude
AVR programmer type: arduino
Baudrate: 115200 (taken from OctoPi's webinterface)
```

Unfortunately I did not pay attention since which Marlin version the issue showed up first. I remember for sure that it worked on Marlin 1.9.4.

I tried with both enabled and disabled setting `Disable write verification`.

sroselli, 2020-04-05:
well, you tried **sudo chmod 777 /dev/ttyUSB*** without success.

1) I not a try with command chmod, i try.
2) I try with a baudrate of 57600 and 115200 but don't works

jan-gerard, 2020-04-18:
Do you see the "startup screen" on display of the Anet when you start the firmware upload? 
When I do this manually, over a wifi connection, I have to reset the Anet just before starting the upload. If I release the reset button too soon or too late, I get the same error you describe. 
If the Anet does not reboot, that's what you get, so check that it reboots when you start updating. Normally, if you use the CH340 over the USB connection, the reset line is triggered by the CH340 interface output.

sroselli, 2020-04-28:
I tried to release the reset a moment before pressing the firmware update button and I did it by changing Baud but it is useless.
The printer displays seems to reset and immediately afterwards the sync errors that I posted above appear. I noticed the note when the start ì shows the display turns off and on again signals the welcome message

sroselli, 2020-05-09:
> Viene visualizzata la "schermata di avvio" sul display dell'Anet quando si avvia il caricamento del firmware?
> Quando lo faccio manualmente, tramite una connessione wifi, devo ripristinare Anet appena prima di iniziare il caricamento. Se rilascio il pulsante di ripristino troppo presto o troppo tardi, ottengo lo stesso errore che descrivi.
> Se Anet non si riavvia, è quello che ottieni, quindi controlla che si riavvii quando inizi l'aggiornamento. Normalmente, se si utilizza CH340 tramite la connessione USB, la linea di ripristino viene attivata dall'uscita dell'interfaccia CH340.

Hai qualche idea per poter provare ?

jan-gerard, 2020-05-09:
> Hai qualche idea per poter provare ?

Yes.
I have connected the reset line of the printer to one of the pins of a ESP3D module I use to update the firmware. The module is connected to the serial Tx and Rx lines of the printer, and now also to the reset line. I can trigger the printer to reset just before I start the firmware upload. https://github.com/luc-github/ESP3D/issues/444 The timing is critical, so I had to tune the delay a bit, but now it works 80% of the time.

sroselli, 2020-05-09:
> luc-github/ESP3D#444 <https://github.com/luc-github/ESP3D/issues/444> The

I'm try ...tanks for a hint

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mail
priva di virus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

-- 

_________________________________

*Sergio Domenico Roselli**Ministero Università e Ricerca*

*Mail :* *sergio.roselli@gmail.com <sergio.roselli@gmail.com>*

*Cell.:* *366 5484974 *

*Il presente messaggio, inclusi gli eventuali allegati, ha natura
strettamente personale e potrebbe contenere informazioni confidenziali
e/o riservate e pertanto il suo contenuto ai sensi della norma vigente
(DL 196/03 ed art. 616 cp) non può essere in nessuna forma anche
verbale, divulgata a terzi diversi dal destinatario, salve espressa
autorizzazione del mittente del messaggio. *

Il mittente del messaggio non si assume alcuna responsabilità per eventuali
danni o perdite derivanti dalla presenza di virus.
*P*rima di stampare questa mail, pensa all'impatto sull'ambiente

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mail
priva di virus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2020-11-22 by benlye