OctoPrint/OctoPrint-FirmwareUpdater #126: Flashing failed Avrdude returned code 1

tuboil, 2020-04-13:
**Hardware Setup**
Board type "Creality V2.1" and printer model "CR10" updated (S).

**Describe the problem**
Flashing failed Avrdude returned code 1 

**Log Files**
`~/.octoprint/logs/`
...
2020-04-13 15:33:14,249 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-04-13 15:33:25,093 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 6311}
2020-04-13 15:34:18,151 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2020-04-13 15:34:18,164 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2020-04-13 15:34:18,199 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2020-04-13 15:34:18,218 - octoprint.plugins.firmwareupdater - INFO - Running '/usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyUSB0 -D -U flash:w:/tmp/tmpeu721d:i' in /usr/bin
2020-04-13 15:34:19,394 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Avrdude returned code 1.
2020-04-13 15:34:19,401 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyUSB0, baudrate=115200, profile={'heatedChamber': False, 'name': 'Default', 'color': 'default', 'axes': {'y': {'speed': 6000, 'inverted': False}, 'x': {'speed': 6000, 'inverted': False}, 'z': {'speed': 200, 'inverted': False}, 'e': {'speed': 300, 'inverted': False}}, 'heatedBed': True, 'volume': {'origin': 'lowerleft', 'formFactor': 'rectangular', 'depth': 200.0, 'width': 200.0, 'custom_box': False, 'height': 200.0}, 'model': 'Generic RepRap Printer', 'id': '_default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}}
2020-04-13 15:34:19,474 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial port"
2020-04-13 15:34:19,478 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"
2020-04-13 15:34:19,487 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-13 15:34:20,598 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-13 15:34:20,861 - octoprint.plugins.tracking - INFO - Sent tracking event printer_safety_warning, payload: {'printer_safety_warning_type': u'firmware-unsafe', 'printer_safety_check_name': u'creality_cr10s'}
2020-04-13 15:34:26,391 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-13 15:34:26,394 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2020-04-13 15:34:26,417 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
2020-04-13 15:34:26,421 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
2020-04-13 15:34:26,469 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-13 15:34:26,560 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.0 From Archive"
2020-04-13 15:34:26,800 - octoprint.plugins.tracking - INFO - Sent tracking event printer_connected, payload: {u'printer_baudrate': 115200, u'printer_port': '/dev/ttyUSB0', 'firmware_name': u'Marlin 1.1.0 From Archive'}
2020-04-13 15:35:54,356 - octoprint.server.util.flask - INFO - Passively logging in user pi from ::ffff:192.168.10.237
2020-04-13 15:35:54,359 - octoprint.access.users - INFO - Logged in user: pi
2020-04-13 15:35:54,531 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.10.237
2020-04-13 15:35:55,087 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.10.237
2020-04-13 15:35:55,230 - octoprint.server.util.flask - INFO - Passively logging in user pi from ::ffff:192.168.10.237
2020-04-13 15:35:55,232 - octoprint.access.users - INFO - Logged in user: pi
2020-04-13 15:35:57,416 - octoprint.server.util.sockjs - INFO - User pi logged in on the socket from client ::ffff:192.168.10.237
...

 `~/.octoprint/logs/plugin_firmwareupdater_console.log`
...
2020-04-13 15:34:18,219 
2020-04-13 15:34:18,220 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyUSB0 -D -U flash:w:/tmp/tmpeu721d:i
2020-04-13 15:34:19,324 > 
2020-04-13 15:34:19,326 > avrdude: Version 6.3-20171130
2020-04-13 15:34:19,327 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2020-04-13 15:34:19,329 >          Copyright (c) 2007-2014 Joerg Wunsch
2020-04-13 15:34:19,330 > 
2020-04-13 15:34:19,331 >          System wide configuration file is "/etc/avrdude.conf"
2020-04-13 15:34:19,332 >          User configuration file is "/home/pi/.avrduderc"
2020-04-13 15:34:19,333 >          User configuration file does not exist or is not a regular file, skipping
2020-04-13 15:34:19,335 > 
2020-04-13 15:34:19,336 >          Using Port                    : /dev/ttyUSB0
2020-04-13 15:34:19,337 >          Using Programmer              : wiring
2020-04-13 15:34:19,338 >          AVR Part                      : ATmega2560
2020-04-13 15:34:19,339 >          Chip Erase delay              : 9000 us
2020-04-13 15:34:19,341 >          PAGEL                         : PD7
2020-04-13 15:34:19,342 >          BS2                           : PA0
2020-04-13 15:34:19,343 >          RESET disposition             : dedicated
2020-04-13 15:34:19,344 >          RETRY pulse                   : SCK
2020-04-13 15:34:19,345 >          serial program mode           : yes
2020-04-13 15:34:19,346 >          parallel program mode         : yes
2020-04-13 15:34:19,347 >          Timeout                       : 200
2020-04-13 15:34:19,349 >          StabDelay                     : 100
2020-04-13 15:34:19,350 >          CmdexeDelay                   : 25
2020-04-13 15:34:19,351 >          SyncLoops                     : 32
2020-04-13 15:34:19,352 >          ByteDelay                     : 0
2020-04-13 15:34:19,353 >          PollIndex                     : 3
2020-04-13 15:34:19,354 >          PollValue                     : 0x53
2020-04-13 15:34:19,356 >          Memory Detail                 :
2020-04-13 15:34:19,357 > 
2020-04-13 15:34:19,358 >                                   Block Poll               Page                       Polled
2020-04-13 15:34:19,360 >            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
2020-04-13 15:34:19,361 >            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
2020-04-13 15:34:19,362 >            eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
2020-04-13 15:34:19,363 >            flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
2020-04-13 15:34:19,364 >            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2020-04-13 15:34:19,366 >            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2020-04-13 15:34:19,367 >            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2020-04-13 15:34:19,368 >            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2020-04-13 15:34:19,370 >            calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
2020-04-13 15:34:19,371 >            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
2020-04-13 15:34:19,372 > 
2020-04-13 15:34:19,373 >          Programmer Type : Wiring
2020-04-13 15:34:19,375 >          Description     : Wiring
2020-04-13 15:34:19,376 >          Programmer Model: AVRISP
2020-04-13 15:34:19,377 >          Hardware Version: 15
2020-04-13 15:34:19,378 >          Firmware Version Master : 2.10
2020-04-13 15:34:19,379 >          Vtarget         : 0.0 V
2020-04-13 15:34:19,380 >          SCK period      : 0.1 us
2020-04-13 15:34:19,382 > 
2020-04-13 15:34:19,383 > avrdude: AVR device initialized and ready to accept instructions
2020-04-13 15:34:19,384 > avrdude: Device signature = 0x1e9801 (probably m2560)
2020-04-13 15:34:19,385 > avrdude: reading input file "/tmp/tmpeu721d"
2020-04-13 15:34:19,386 > avrdude: invalid record at line 281 of "/tmp/tmpeu721d"
2020-04-13 15:34:19,387 > avrdude: read from file '/tmp/tmpeu721d' failed
2020-04-13 15:34:19,389 > 
2020-04-13 15:34:19,390 > avrdude done.  Thank you.
2020-04-13 15:34:19,391 > 
2020-04-13 15:34:19,392 > 
...

**Additional context**
Plugin Installation
Install manually using this URL: https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/master.zip

benlye, 2020-04-13:
Avrdude says the file is corrupt.  Have you verified that it isn't?  Can you flash the same file via a any other method?  Can you run the avrdude from the command line of your server?  Does it succeed?

The plugin isn't doing anything fancy with the file - it gets copied to the tmp path then AVRDUDE gets called....

tuboil, 2020-04-13:
In fact, Avrdude says the file is corrupt : 
2020-04-13 15:34:19,387 > avrdude: read from file '/tmp/tmpeu721d' failed

and manually on the server:
pi@octopi:/usr/bin $ avrdude -v -q -p m2560 -c wiring -P /dev/ttyUSB0 -D -U flash:w:/tmp/tmpeu721d:i

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/pi/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : wiring
avrdude: stk500v2_ReceiveMessage(): timeout
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
avrdude: stk500v2_ReceiveMessage(): timeout
^C
pi@octopi:/usr/bin $

benlye, 2020-04-14:
Completely different error.  You need to resolve whatever issues are preventing avrdude from talking to your printer before you'll be able to make the plugin work.

Was the printer on?  Connected to OctoPrint?  (Disconnect it before trying to flash.)

tuboil, 2020-04-14:
Yes, my printer was "power on*, yes it was connected to OctoPrint.

~/.octoprint/logs/serial.log
...
2020-04-14 22:49:05,780 - Connection closed, closing down monitor
2020-04-14 22:49:05,796 - Changing monitoring state from "Operational" to "Offline"
2020-04-14 22:49:07,270 - Connecting to: /dev/ttyUSB0
2020-04-14 22:49:07,293 - Changing monitoring state from "Offline" to "Opening serial port"
2020-04-14 22:49:07,308 - Connected to: Serial<id=0x6937f210, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2020-04-14 22:49:07,309 - Changing monitoring state from "Opening serial port" to "Connecting"
2020-04-14 22:49:07,330 - Send: N0 M110 N0*125
2020-04-14 22:49:08,413 - Recv: start
2020-04-14 22:49:08,427 - Recv: echo: External Reset
2020-04-14 22:49:08,432 - Recv: Marlin
2020-04-14 22:49:08,438 - Send: N0 M110 N0*125
2020-04-14 22:49:08,440 - Recv: echo: **Last Updated: 2015-12-01 12:00** | Author: (CR-10Slanguage)
2020-04-14 22:49:08,443 - 
2020-04-14 22:49:08,444 - !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
2020-04-14 22:49:08,444 - Your printer's firmware is known to lack mandatory safety features (e.g.
2020-04-14 22:49:08,446 - thermal runaway protection). This is a fire risk.
2020-04-14 22:49:08,447 - 
2020-04-14 22:49:08,447 - Learn more at https://faq.octoprint.org/warning-firmware-unsafe
2020-04-14 22:49:08,448 - !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
2020-04-14 22:49:08,449 - 
2020-04-14 22:49:08,450 - 
2020-04-14 22:49:08,455 - Recv: Compiled: Jun  2 2018
2020-04-14 22:49:08,466 - Recv: echo: Free Memory: 892  PlannerBufferBytes: 1232
2020-04-14 22:49:08,474 - Recv: echo:Hardcoded Default Settings Loaded
2020-04-14 22:49:08,488 - Recv: echo:Steps per unit:
2020-04-14 22:49:08,491 - Recv: echo:  M92 X80.00 Y80.00 Z400.00 E93.00
2020-04-14 22:49:08,499 - Recv: echo:Maximum feedrates (mm/s):
2020-04-14 22:49:08,503 - Recv: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
2020-04-14 22:49:08,507 - Recv: echo:Maximum Acceleration (mm/s2):
2020-04-14 22:49:08,551 - Recv: echo:  M201 X500 Y500 Z100 E5000
2020-04-14 22:49:08,554 - Recv: echo:Accelerations: P=printing, R=retract and T=travel
2020-04-14 22:49:08,562 - Recv: echo:  M204 P500.00 R500.00 T1000.00
2020-04-14 22:49:08,567 - Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
2020-04-14 22:49:08,570 - Recv: echo:  M205 S0.00 T0.00 B20000 X5.00 Z0.40 E5.00
2020-04-14 22:49:08,574 - Recv: echo:Home offset (mm):
2020-04-14 22:49:08,577 - Recv: echo:  M206 X0.00 Y0.00 Z0.00
2020-04-14 22:49:08,581 - Recv: echo:Material heatup parameters:
2020-04-14 22:49:08,583 - Recv: echo:  M145 M0 H185 B0 F0
2020-04-14 22:49:08,585 - Recv: echo:  M145 M1 H240 B0 F0
2020-04-14 22:49:08,587 - Recv: echo:PID settings:
2020-04-14 22:49:08,590 - Recv: echo:  M301 P22.20 I1.08 D114.00 C100.00 L20
2020-04-14 22:49:08,592 - Recv: echo:Filament settings: Disabled
2020-04-14 22:49:08,595 - Recv: echo:  M200 D3.00
2020-04-14 22:49:08,599 - Recv: echo:  M200 D0
2020-04-14 22:49:08,601 - Recv: echo:Z-Probe Offset (mm):
2020-04-14 22:49:08,604 - Recv: echo:  M851 Z-0.20
2020-04-14 22:49:12,088 - Recv: echo:SD init fail
2020-04-14 22:49:14,192 - Recv: echo:SD init fail
2020-04-14 22:49:14,200 - Recv: ok
2020-04-14 22:49:14,216 - Send: N0 M110 N0*125
2020-04-14 22:49:14,219 - Changing monitoring state from "Connecting" to "Operational"
2020-04-14 22:49:14,280 - Recv: ok
2020-04-14 22:49:14,286 - Send: N0 M110 N0*125
2020-04-14 22:49:14,304 - Recv: ok
2020-04-14 22:49:14,308 - Send: N1 M115*39
2020-04-14 22:49:14,397 - Recv: FIRMWARE_NAME:Marlin 1.1.0 From Archive SOURCE_CODE_URL:http:// ... PROTOCOL_VERSION:1.0 MACHINE_TYPE:www.cxsw3d.com EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
2020-04-14 22:49:14,423 - Recv: ok
2020-04-14 22:49:14,437 - Send: M21
2020-04-14 22:49:16,544 - Recv: echo:SD init fail
2020-04-14 22:49:16,556 - Recv: ok
2020-04-14 22:49:19,221 - Send: M105
2020-04-14 22:49:19,244 - Recv: ok T:21.6 /0.0 B:21.7 /0.0 T0:21.6 /0.0 @:0 B@:0
2020-04-14 22:49:24,218 - Send: M105
2020-04-14 22:49:24,229 - Recv: ok T:21.6 /0.0 B:21.7 /0.0 T0:21.6 /0.0 @:0 B@:0
2020-04-14 22:49:29,220 - Send: M105
2020-04-14 22:49:29,232 - Recv: ok T:21.6 /0.0 B:21.6 /0.0 T0:21.6 /0.0 @:0 B@:0
[... 142 lines cut ...]
avrdude: stk500v2_ReceiveMessage(): timeout
         Vtarget         : 0.8 V
         SCK period      : 0.1 us
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD
avrdude: reading input file "/tmp/tmpYSin_z"
avrdude: can't open input file /tmp/tmpYSin_z: No such file or directory
avrdude: read from file '/tmp/tmpYSin_z' failed
avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D8, L:FF)

avrdude done.  Thank you.

pi@octopi:~ $ 

> too much timeout ... there's some kind of bad communications.

Any ideas?

benlye, 2020-04-15:
In your 'manual' test the upload was never attempted because the file wasn't found:

```
avrdude: reading input file "/tmp/tmpYSin_z"
avrdude: can't open input file /tmp/tmpYSin_z: No such file or directory
```

You can't just try to flash with the same path as the plugin used because the temp file gets removed.  You need to copy your firmware file to the raspberry pi manually and then try to flash it.

tuboil, 2020-04-15:
Thanks...
My mistake was a wrong interpretation of the **Firmware Updater**.

Indeed, I was using the following option:
"... from URL " : https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/master.zip `Flash from URL` 

In fact, I've solved my problem with:
"... from file " :  "CR-10S 1.1.6 latest firmware/CR-10STF1.1.6 resume printing and filament sensor.hex" `Flash from File` 

I found it in the file "Creality_Latest Marlin1.1.6 firmware_CR-10S.rar" I found it in the file provided by the manufacturer. (> https://creality.com/cr-10s-3d-printer-p00097p1.html)

~/.octoprint/logs/plugin_firmwareupdater_console.log
...
2020-04-15 20:27:05,975 
2020-04-15 20:27:05,977 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyUSB0 -D -U flash:w:/tmp/tmpAwYPcL:i
2020-04-15 20:27:07,115 > 
2020-04-15 20:27:07,117 > avrdude: Version 6.3-20171130
2020-04-15 20:27:07,118 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2020-04-15 20:27:07,119 >          Copyright (c) 2007-2014 Joerg Wunsch
2020-04-15 20:27:07,120 > 
2020-04-15 20:27:07,122 >          System wide configuration file is "/etc/avrdude.conf"
2020-04-15 20:27:07,123 >          User configuration file is "/home/pi/.avrduderc"
2020-04-15 20:27:07,124 >          User configuration file does not exist or is not a regular file, skipping
2020-04-15 20:27:07,125 > 
2020-04-15 20:27:07,127 >          Using Port                    : /dev/ttyUSB0
2020-04-15 20:27:07,128 >          Using Programmer              : wiring
2020-04-15 20:27:07,129 >          AVR Part                      : ATmega2560
2020-04-15 20:27:07,130 >          Chip Erase delay              : 9000 us
2020-04-15 20:27:07,131 >          PAGEL                         : PD7
2020-04-15 20:27:07,132 >          BS2                           : PA0
2020-04-15 20:27:07,134 >          RESET disposition             : dedicated
2020-04-15 20:27:07,135 >          RETRY pulse                   : SCK
2020-04-15 20:27:07,136 >          serial program mode           : yes
2020-04-15 20:27:07,137 >          parallel program mode         : yes
2020-04-15 20:27:07,138 >          Timeout                       : 200
2020-04-15 20:27:07,140 >          StabDelay                     : 100
2020-04-15 20:27:07,141 >          CmdexeDelay                   : 25
2020-04-15 20:27:07,142 >          SyncLoops                     : 32
2020-04-15 20:27:07,143 >          ByteDelay                     : 0
2020-04-15 20:27:07,145 >          PollIndex                     : 3
2020-04-15 20:27:07,146 >          PollValue                     : 0x53
2020-04-15 20:27:07,147 >          Memory Detail                 :
2020-04-15 20:27:07,148 > 
2020-04-15 20:27:07,150 >                                   Block Poll               Page                       Polled
2020-04-15 20:27:07,151 >            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
2020-04-15 20:27:07,152 >            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
2020-04-15 20:27:07,153 >            eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
2020-04-15 20:27:07,155 >            flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
2020-04-15 20:27:07,156 >            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2020-04-15 20:27:07,157 >            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2020-04-15 20:27:07,158 >            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2020-04-15 20:27:07,159 >            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2020-04-15 20:27:07,161 >            calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
2020-04-15 20:27:07,162 >            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
2020-04-15 20:27:07,163 > 
2020-04-15 20:27:07,164 >          Programmer Type : Wiring
2020-04-15 20:27:07,165 >          Description     : Wiring
2020-04-15 20:27:07,166 >          Programmer Model: AVRISP
2020-04-15 20:27:07,168 >          Hardware Version: 15
2020-04-15 20:27:07,169 >          Firmware Version Master : 2.10
2020-04-15 20:27:07,170 >          Vtarget         : 0.0 V
2020-04-15 20:27:07,171 >          SCK period      : 0.1 us
2020-04-15 20:27:07,172 > 
2020-04-15 20:27:07,174 > avrdude: AVR device initialized and ready to accept instructions
2020-04-15 20:27:07,175 > avrdude: Device signature = 0x1e9801 (probably m2560)
2020-04-15 20:27:07,176 > avrdude: reading input file "/tmp/tmpAwYPcL"
2020-04-15 20:27:07,177 > avrdude: writing flash (135728 bytes):
2020-04-15 20:27:07,178 > 
2020-04-15 20:27:41,144 > avrdude: 135728 bytes of flash written
2020-04-15 20:27:41,145 > avrdude: verifying flash memory against /tmp/tmpAwYPcL:
[... 2 lines cut ...]
2020-04-15 20:27:41,147 > avrdude: reading on-chip flash data:
2020-04-15 20:27:41,148 > 
2020-04-15 20:28:10,893 > avrdude: verifying ...
2020-04-15 20:28:10,895 > avrdude: 135728 bytes of flash verified
2020-04-15 20:28:10,896 > 
2020-04-15 20:28:10,898 > avrdude done.  Thank you.
2020-04-15 20:28:10,899 > 
2020-04-15 20:28:10,900 > 
2020-04-15 20:28:10,907 **Flashing successful.**

~/.octoprint/logs/octoprint.log
...
2020-04-15 20:27:05,928 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
[... cut ...] dth': 200.0, 'custom_box': False, 'height': 200.0}, 'model': 'Generic RepRap Printer', 'id': '_default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}}
2020-04-15 20:28:11,029 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial port"
2020-04-15 20:28:11,037 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"

Thank you very much for your work and your patience.

closed 2020-04-17 by benlye