OctoPrint/OctoPrint-FirmwareUpdater #127: Reset board just before firmware upload via ESP3D

jan-gerard, 2020-04-14:
**Hardware Setup**
Anet A6 (Marlin 2.0) with ESP8266 running ESP3D.

**Describe the problem**
I can flash the firmware of my printer via the ESP3D (using the OctoPrint-FirmwareUpdater plugin), buth then I have to press the reset button (below the LCD) just before starting the firmware upload.
It would be great if this plugin would reset the printer just before starting the firmware upload.
However, I tried with the G997 code, but that can only be enabled in the printer for a hand full of board configurations, not for the Anet. However, I'm sure Marlin could reset the board through some native Arduino command. And couldn't this plugin achieve it somehow?

**Log Files**
I have no log-files, as it is a manual action of pressing buttons...

**Additional context**
I'd like to automate this by a soft-reset via this plugin.

benlye, 2020-04-15:
The plugin can only use the methods made available by Marlin.  You'll need top open a request there if you want them to provide a new method.

closed 2020-04-15 by benlye

jan-gerard, 2020-04-18:
I found a hardware solution:
I soldered a connection from the reset pin of my printer board (happened to be on the same connector as the serial interface I use for ESP3D) to one of the outputs of my esp8266-F board that acts as the ESP3D interface. So GPIO13 is connected to printer reset.
No I had to trigger this output in order to reset my board. It resets on the positive edge of the reset line. I used the "system command" field of FirmwareUpdater for that, by executing the following system command 
`wget http://192.168.1.108/command?plain=[ESP201]P13%20V0;wget http://192.168.1.108/command?plain=[ESP201]P13%20V1; wget http://192.168.1.108/command?plain=[ESP201]P13%20PULLUP=NO; sleep .1`
Direcly after that, OctoPrint-FirmwareUpdater calls /usr/bin/avrdude to upload the printer firmware. 
It works fine! I first pull down the reset line, then back up, and then set it to high-impedance input mode, have 100 ms delay, and that's it.
The system call `wget` to the ESP3D board with the parameters `?plain=[ESP201]P13%20Vx` switches the GPIO13 terminal to the value of `x`.