OctoPrint/OctoPrint-FirmwareUpdater #131: [Feature Request]  Add support for LPC1769 boards.

Zilveron, 2020-04-28:
Would it be possible to add support for the boards based upon the LPC1769 such as the SKR v1.4 and 1.4 Turbo?  Thank you.

ForlornWinter, 2020-04-28:
I also came here looking for a solution for an LPC1769-based board (1.4 Turbo). Given that the SKR 1.1 Pro is compatible (since it uses the same flash-and-reset procedure as the LPC1768) and the LPC1769 updates the same way as LPC1768-based boards, it may already be compatible. If the maintainer of this plugin doesn't have access to the hardware and needs information, let me know-- I'll provide what information I can to help confirm compatibility.

Glought, 2020-04-28:
It does work fine on the SKR v1.4 Turbo just follow this [https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#lpc1768-boards](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#lpc1768-boards) and in your marlin configuration_adv.h file make sure "SDCARD_CONNECTION" is set to ONBOARD. 
Also on your Octoprint settings "GCODE Scripts" add M22 to "After connection to printer is established" it tells Marlin to release the sd card so the firmwareUpdater plugin can write to the sd card.

Zilveron, 2020-04-28:
Thanks for the information.  I thought about using the LPC1768 flashing method but wasn't sure it would work and didn't want to chance it.  Though after the near flawless print I got overnight, I might be done tweaking until I'm ready to bring the second extruder I'm adding online.

Glought, 2020-04-28:
the LPC1768 and the LPC1769 are the same just the LPC1769 is running at a higher frequency  100MHz vs 120MHz

closed 2020-05-11 by benlye