OctoPrint/OctoPrint-FirmwareUpdater #132: connection attempts called from outside breaks flashing

Mirarkitty, 2020-05-02:
When

1) flashing is ongoing
and 
2) OctoPrint gets a connect command 

OctoPrint will send data over the serial port, breaking the flash (or the verify).

If possible, OctoPrint should probably not try to reconnect and send data while flashing is ongoing. Can OctoPrint be put in a state where it doesn't meddle with the connection while flashing?

(Yes, yes, I know this is a very minor issue.)

paulroberthill, 2020-05-20:
I think I have the same issue. If I kill OctoPrint and run a manual flash with avrdude it works.

benlye, 2020-05-23:
Why is OctoPrint trying to reconnect?  Do you have another plugin that is causing that to happen automatically?  I've never had it happen on my printer.

I'm not sure if the API has a way to block or disable connection attempts but I'll look into it.

paulroberthill, 2020-05-23:
I just tried disabling all plugins. In OctoPrint if I hit disconnect it will reconnect immediately!
"Auto-connect on server startup" is not enabled.

I assume this is the issue. Very odd!

Mirarkitty, 2020-06-09:
I noticed the issue because I had an external script (tied to my home automation) that wanted to reconnect the printer. I haven't spotted any external status (REST API) about that octoprint is flashing, that would be nice too...

paulroberthill, 2020-06-18:
I found my issue. I have a touchscreen attached to my PI and run OctoScreen. Disabling OctoScreen fixes my issue.

Mirarkitty, 2020-12-13:
This doesn't seem fixed, why is it closed?

benlye, 2020-12-13:
@Mirarkitty what kind of board are you updating?

Mirarkitty, 2020-12-29:
The Prusa MK3S board.
[image]

Mirarkitty, 2020-12-29:
Since the plugin knows when it starts and stops the flashing (or so it seems), signalling that the connection is busy so do not connect the printer would be very nice.

benlye, 2021-01-03:
@foosel can you give any guidance here?  Is there a way I can have the Firmware Updater plugin tell OctoPrint not to try to reconnect to the printer while a firmware update is in progress?

foosel, 2021-01-25:
Sorry, the notification got buried. OctoPrint should normally never reconnect on its own. However, there is currently no official way to prevent a connect command from going through if one comes in. I could add a hook for that so that your plugin can block this in such a case. In the meantime an option would be a monkey patch of the connect endpoint on the printer object, however that would be really just a stopgap solution (monkey patching in general should be a last resort kinda thing).

I'll look into a hook for y'all (and everyone else).

benlye, 2021-04-05:
1.11.0b4 in the Devel release branch will now block serial connections while a flash is in progress by default.
* Requires OctoPrint >=1.6.0 (works with RC1)
* On by default but can be disabled in the plugin settings

benlye, 2021-05-16:
Firmware Updater v1.11.0 has been released.

closed 2021-05-16 by benlye