OctoPrint/OctoPrint-FirmwareUpdater #135: Printrboard using CDC bootloader

romanr9999, 2020-05-23:
Hello.
I'd like to flash a Printrboard that currently has a CDC bootloader installed.
I've been playing around with the setup and it seems you only currently support DFU bootloaders on Printrboard.
Thanks.

benlye, 2020-05-23:
I can look into it.  The original ask for Printrboard support was for DFU, and as that seemed to be the default that's what I added support for.  I still have the Teensy 2.0++ I used to test that out, so I should be able to figure out the CDC bootloader as well.

Do you have a link to the bootloader you're using?  How do you update the firmware on the board now (tools, commands, etc.)?

benlye, 2020-05-23:
I poked around a little more.  I guess this is the CDC bootloader:
https://reprap.org/wiki/Teensylu#Alternate_Bootloader

I flashed it to my test board successfully.  If you could give me a pointer to the process you currently use to flash your board it will save me some time.

benlye, 2020-05-23:
I'm pretty sure it will be trivial to make this work - it looks like you flash the board with the Arduino IDE which uses avrdude.  Can you confirm how you flash your board currently?  If you use Arduino, a copy of the verbose output during upload would be super helpful.

Have you flashed your board from a Raspberry Pi before?  Just curious because I saw a warning about doing that here:
https://reprap.org/wiki/Printrboard#Loading_Firmware_.28Linux.29

romanr9999, 2020-05-23:
Hello.
Thanks for the quick answer.
I've never updated this board before. I have it supported in Arduino IDE but I wasn't able to compile Marlin 2.0 in the Arduino IDE. I did compile it in VSCode and that's why I need a way to upload the HEX file.

romanr9999, 2020-05-23:
I guess you've already seen this, but just in case.
There's some info on the board and the upload process in https://reprap.org/wiki/Printrboard

benlye, 2020-05-23:
OK.  You should be able to configure the plugin to flash your board.

Go into the settings and pick avrdude as your flash method, then pick any AVR MCU and flash method. Set and test the path to avrdude (install it if you don't already have it).

Then expand the Advanced Settings and go down to the  'Command Line' and change it to this (hard coding the MCU and programmer you need):

`{avrdude} -v -q -p at90usb1286 -c avr109 -P {port} -D -C {conffile} -b {baudrate} {disableverify} -U flash:w:{firmware}:i`

That *should* work - those settings work for my AT90USB development board.

I don't know if your Printrboard has a button you need to press before you can flash it.  My dev board does - it wouldn't flash until I pressed that button to boot it into the bootloader, I guess.

If it works I'll add the MCU and programmer as options and release an update - it won't change anything other than give you the options you hardcoded as menu choices..  Let me know how it goes.

romanr9999, 2020-05-23:
That's great! I'll give it a try and let you know.

> I don't know if your Printrboard has a button you need to press before you can flash it. My dev board does - it wouldn't flash until I pressed that button to boot it into the bootloader, I guess.

According to the Printrboard revision (mine is D) you have to either install or remove a jumper to boot the board into the bootloader.
In my case I have to install the jumper and press the reset button and it goes into bootloader, I already verified that.

I connected the board directly to my computer and tried avrdude from the command line and for some reason I couldn't get the board and avrdude to sync. I'm not sure why yet. The command line I used was pretty much the same you mentioned.

I'll try again both from the commando line and following your instructions and will get back with the results.

Thanks again!!!!!

romanr9999, 2020-05-23:
Hello Again.
I have good news and bad news.
The good ones are that following your instructions I was able to download a copy of the original flash through avrdude from the command line.
I run:
     avrdude -v -q -p at90usb1286 -c avr109 -PCOM8 -D -Cavrdude.conf -b115200 -U flash:r:firmware.hex:i

That worked like a charm.
Following that (and before I saved the session log) I tried to upload my compiled Marlin firmware. The process took way too long and the board stop responding. Now the board seems bricked because it's not recognized when plugged into the PC. I'm thinking there might be a file format issue.
I run: 
   avrdude -v -q -p at90usb1286 -c avr109 -PCOM8 -D -Cavrdude.conf -b115200 -U flash:w:firmware.hex:i

I'll try to recover it.

So, the procedure works as you said and a Printrboard with CDC bootloader can be flashed with your plugin.

benlye, 2020-05-24:
That is a bummer. If you have an ISP you should be able to recover it. I used a USBasp to flash the AT90USB bootloader on my board.

romanr9999, 2020-05-24:
I have a few tools although I honestly don't remember how to use them anymore :(
I have a AVR-USB-JTAG and an Atmel ICE. I also know I've built and used and Arduino as an ISP.
Trying to google how to use them now.

romanr9999, 2020-05-25:
Ok.
Burnt the bootloader with an arduino board. 
The only thing now is that when i plug in the board to the computer, i hear the sound for the new device but i don't see the the board as a COM port. I can't find the board in the device list. weird

closed 2020-12-05 by github-actions[bot]