OctoPrint/OctoPrint-FirmwareUpdater #137: LPC1768 - usbmount instructions not working

Syco54645, 2020-05-28:
I am attempting to get the fw updater working for my skr 1.4 turbo and the sd card is not mounting on the pi. I have tried releasing the sd card from the printer and restarting the service and it still does not work. Any help here would be appreciated.

kn-dima, 2020-05-29:
Same issue with SKR v1.3

KevMags, 2020-05-29:
Can you elaborate adding line to fstab for a newb?

kn-dima, 2020-05-29:
1. Get access to pi command line (i am connecting through SSH with Putty. User name pi, default password raspberry)
2. enter "sudo nano /etc/fstab" and it will ask for password
3. after entering password you will see file system table, something like:
`proc            /proc           proc    defaults          0       0
PARTUUID=6c586e13-01  /boot           vfat    defaults          0       2
PARTUUID=6c586e13-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
`
4. copy string from my message above and insert into table
use alt+m,alt+u for undo if something looks wrong
press ctrl+x for exit, then Y or N for write file or not.
5. sudo reboot

Syco54645, 2020-05-30:
> found bypass through editing fstab
> 
> 1. Assign label to sdcard, without spaces
> 2. Add line to fstab
>    `LABEL=SKR_1_3 /media/usb0 vfat sync,noexec,nodev,noatime,nodiratime,gid=pi,dmask=0022,fmask=0111 0 2 `
> 3. change Firmware updater "Path to firmware folder" option to /media/usb0
> 4. reboot pi
> 
> Now you can update firmware.
> After update you must connect to printer and enter M22 command before you can update firmware again.

Why does the label need to be added?

kn-dima, 2020-05-30:
partition name is changing after each firmware update i.e. when after reboot you have /dev/sda1, after firmware update it can change to /dev/sdb1.
uuid and partuuid is unusable if you want replace card.
For me, partition label is a simpliest way to identify sdcard partition.

kn-dima, 2020-05-30:
Oops! without nofail option pi cant start after power off if printer USB not attached and sdcard not released in Marlin.
fstab line should be
```
LABEL=SKR_1_3         /media/usb0     vfat    nofail,sync,noexec,nodev,noatime,nodiratime,gid=pi,dmask=0022,fmask=0111 0 0
```

KevMags, 2020-05-30:
Aaaaaaand, that got it.  Thanks, this saves me from disassembling the printer to reflash the SKR.

KevMags, 2020-05-30:
AAAAAnd, problem back again.  It has worked a few times, the times that it hasn't it disconnects the SKR from the pi when starting upload leaving the message "Waiting for SD card to mount on host...".

kn-dima, 2020-05-31:
Can you 

> AAAAAnd, problem back again. It has worked a few times, the times that it hasn't it disconnects the SKR from the pi when starting upload leaving the message "Waiting for SD card to mount on host...".
> Reset before flash diabled

check that you still can connect to printer
enter M22 in octoprint terminal or release card (release media) from marlin menu
get access to pi command line
enter blkid /dev/sd*
you should see skr partition
```
pi@octopi:~ $ blkid /dev/sd*
/dev/sda1: SEC_TYPE="msdos" LABEL="SKR_1_3" UUID="349C-A708" TYPE="vfat" PARTUUID="0396c84e-01"
```
ls /media/usb0
```
pi@octopi:~ $ ls /media/usb0
FIRMWARE.CUR
```

kn-dima, 2020-05-31:
After removing usbmount, I had to change options fmask and dmask to 0000.
That variant also work on clean installation without install usbmount and setting PrivateMounts and MountFlags:

```
LABEL=SKR_1_3         /media/skr_1_3  vfat    nofail,sync,noexec,nodev,noatime,nodiratime,dmask=0000,fmask=0000 0 0
```

KevMags, 2020-05-31:
After shuffling 8 SD cards at 3:00 am I finally got it straightened out this morning.  All works well now.
Thanks for your effort.

Syco54645, 2020-09-12:
Sorry for my absence, life got in the way. @kn-dima I have tried everything that you have suggested above and none of it is working for me. There is no mountpoint created at /media/skr_1_3.

bill-orange, 2020-10-16:
@Syco54645 I just got my Firmware updater on the PI to work with an SKR1.3 board.  I did not find the Label= line to be necessary.  What is necessary however is sending a M22 from the terminal to release the card before testing.

RFBomb, 2020-11-15:
I am also very recently having this issue. 
Previously, it worked flawlessly following the instructions on the install page. I don't know if this is related to the issues everyone else in this thread is having. 

My issue is that the SD Card is being shown as 'Read Only FileSystem' when I go to check it out. M21 / M22 change nothing. 
OctoPrint can see the SD Card, can view the files on the SD Card, but is mounted as read only. I have even tried 2 separate SD cards. both have the same issue. 
[image]

fordaj, 2021-01-12:
The change that ended up working for me was running the line mentioned in [this forum](https://community.octoprint.org/t/firmwareupdater-and-skr-1-3-lpc1768/15894/14):
```
sudo chmod 777 /media/usb0
```
Might be worth adding to [this section](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#sudo-rights) in the documentation. I followed the instructions for a Pi 3 and Pi4, as well as an SKR V1.3 and SKR Mini E3 V1.2, and elevating the permissions of /media/usb0 ended up fixing it for both configurations. Of course, I could be reading something incorrectly, and the error could be between the chair and the keyboard!

benlye, 2021-01-19:
Please see my comments here about changing permissions on /media/usb0:

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/175#issuecomment-760949800

Spoiler: it's not a good idea.

benlye, 2021-01-30:
I've updated the documentation and tested the described process on an SKR 1.4 Turbo board.  It works exactly as described/expected when properly configured.

closed 2021-01-30 by benlye