OctoPrint/OctoPrint-FirmwareUpdater #138: Firmware updater PI has problems with SKR1.4 using usbmount

TxBillbr, 2020-05-30:
**Hardware Setup**
Octoprint 1.40 -> SKR 1.4 Turbo
Uname: Linux octopi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
**Describe the problem**
Attempted to setup firmware updater per the instructions on the plugin github. usbmount did not seem to mount the usb.

**Log Files**
dmesg snip

[    1.619293] usb 1-1.4: New USB device found, idVendor=1d50, idProduct=6029, bcdDevice= 1.00
[    1.619328] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.619359] usb 1-1.4: Product: Marlin USB Device
[    1.619383] usb 1-1.4: Manufacturer: marlinfw.org
[    1.619408] usb 1-1.4: SerialNumber: 0D00201CAF37A0805BBFB7B4F50020C3
[    1.619649] systemd[1]: Detected architecture arm.
[    1.628490] usb-storage 1-1.4:1.2: USB Mass Storage device detected
[    1.629159] scsi host0: usb-storage 1-1.4:1.2
[    1.629724] systemd[1]: Set hostname to <octopi>.
[    1.636809] systemd[1]: Failed to bump fs.file-max, ignoring: Invalid argument
[    1.931871] systemd[1]: File /lib/systemd/system/systemd-journald.service:12 configures an IP firewall (IPAddressDeny=any), but the local system does not support BPF/cgroup based firewalling.
[    1.931933] systemd[1]: Proceeding WITHOUT firewalling in effect! (This warning is only shown for the first loaded unit using IP firewalling.)
[    1.966844] systemd[1]: /etc/systemd/system/systemd-udevd.service.d/override.conf:3: Unknown lvalue 'Mountflags' in section 'Service', ignoring
[    2.265607] random: systemd: uninitialized urandom read (16 bytes read)
[    2.273887] random: systemd: uninitialized urandom read (16 bytes read)
[    2.274485] systemd[1]: Listening on udev Control Socket.
[    2.275232] random: systemd: uninitialized urandom read (16 bytes read)
[    2.275546] systemd[1]: Listening on initctl Compatibility Named Pipe.
[    2.276776] systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.
[    2.643644] scsi 0:0:0:0: Direct-Access     Marlin   SDCard 01        1.0  PQ: 0 ANSI: 0 CCS
[    2.696548] sd 0:0:0:0: [sda] Attached SCSI removable disk

Octoprint snip when using /tmp as flash location - tells me plugin is working as expected.
2020-05-29 20:33:51,259 - octoprint.startup - INFO - ******************************************************************************
2020-05-29 20:33:51,260 - octoprint.startup - INFO - Starting OctoPrint 1.4.0
2020-05-29 20:33:51,261 - octoprint.startup - INFO - ******************************************************************************
2020-05-29 20:33:51,575 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2020-05-29 20:33:51,577 - octoprint.startup - INFO - Blacklist processing done, adding 1 blacklisted plugin versions: roomtemp (any)
2020-05-29 20:33:51,606 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-05-29 20:33:52,242 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 16 mixin implementations, 29 hook handlers
2020-05-29 20:33:52,341 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2020-05-29 20:33:52,355 - octoprint.server - INFO - Intermediary server started
2020-05-29 20:33:52,355 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-05-29 20:33:52,418 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 16 mixin implementations, 29 hook handlers
2020-05-29 20:33:52,435 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2020-05-29 20:33:52,436 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2020-05-29 20:33:52,439 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: u"Need(method='role', value=u'plugin_action_command_prompt_interact')")
2020-05-29 20:33:52,440 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: u"Need(method='role', value=u'plugin_announcements_read')")
2020-05-29 20:33:52,440 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: u"Need(method='role', value=u'plugin_announcements_manage'), Need(method='role', value=u'plug
in_announcements_read')")
2020-05-29 20:33:52,441 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: u"Need(method='role', value=u'plugin_appkeys_admin')")
2020-05-29 20:33:52,442 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: u"Need(method='role', value=u'plugin_backup_access')")
2020-05-29 20:33:52,443 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: u"Need(method='role', value=u'plugin_logging_manage')")
2020-05-29 20:33:52,444 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: u"Need(method='role', value=u'plugin_pi_support_check')")
2020-05-29 20:33:52,444 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: u"Need(method='role', value=u'plugin_pluginmanager_manage')")
2020-05-29 20:33:52,446 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: u"Need(method='role', value=u'plugin_pluginmanager_install'), Need(method='role', value=u'pl
ugin_pluginmanager_manage')")
2020-05-29 20:33:52,446 - octoprint.server - INFO - Added new permission from plugin printer_safety_check: PLUGIN_PRINTER_SAFETY_CHECK_DISPLAY (needs: u"Need(method='role', value=u'plugin_printer_safety_check_display')")
2020-05-29 20:33:52,447 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: u"Need(method='role', value=u'plugin_softwareupdate_check')")
2020-05-29 20:33:52,448 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: u"Need(method='role', value=u'plugin_softwareupdate_update')")
2020-05-29 20:33:52,457 - octoprint.access.users - INFO - Migrating user billbr to new granular permission system
2020-05-29 20:33:52,484 - octoprint.plugins.softwareupdate - INFO - Version cache was created for another version of OctoPrint, not using it
2020-05-29 20:33:53,036 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2020-05-29 20:33:53,587 - octoprint.util.pip - INFO - Version of pip is 19.3.1
2020-05-29 20:33:53,588 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2020-05-29 20:33:53,589 - octoprint.util.pip - INFO - ==> pip ok -> yes
2020-05-29 20:33:53,592 - octoprint.plugin.core - INFO - Initialized 16 plugin implementation(s)
2020-05-29 20:33:53,644 - octoprint.plugin.core - INFO - 16 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
[... 205 lines cut ...]
ID_VENDOR_ENC=Marlin\x20\x20
ID_VENDOR_ID=1d50
ID_MODEL=SDCard_01
ID_MODEL_ENC=SDCard\x2001\x20\x20\x20\x20\x20\x20\x20
ID_MODEL_ID=6029
ID_REVISION=1.0
ID_SERIAL=Marlin_SDCard_01_0D00201CAF37A0805BBFB7B4F50020C3-0:0
ID_SERIAL_SHORT=0D00201CAF37A0805BBFB7B4F50020C3
ID_TYPE=disk
ID_INSTANCE=0:0
ID_BUS=usb
ID_USB_INTERFACES=:020200:0a0000:080650:
ID_USB_INTERFACE_NUM=02
ID_USB_DRIVER=usb-storage
DEVLINKS=/dev/disk/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.2-scsi-0:0:0:0 /dev/disk/by-id/usb-Marlin_SDCard_01_0D00201CAF37A0805BBFB7B4F50020C3-0:0
ID_PATH=platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.2-scsi-0:0:0:0
ID_PATH_TAG=platform-fd500000_pcie-pci-0000_01_00_0-usb-0_1_4_1_2-scsi-0_0_0_0
TAGS=:systemd:
USEC_INITIALIZED=3492256
run: '/usr/share/usbmount/usbmount add'
Unload module index
Unloaded link configuration context.

running usbmount did not work.
Found this issue that indicates usbmount is  an orphan project.  https://raspberrypi.stackexchange.com/questions/100312/raspberry-4-usbmount-not-working

CaptainSweatpants, 2020-09-21:
This broke my serial connection to my printer from octoprint as well. Disabling it returned connection abilities to my SKR 1.4 turbo; I haven't found a solution yet to update firmware via Octoprint with this BTT SKR 1.4 T board. For now, the only way i can update Marlin is to enable usbmount, then plug it into windows and the firmware folder will open there. I drop firmware.bin and reboot the printer. then once updated, back to ssh'ing over to the RPi and disabling usbmount so octoprint can connect again. 

This isn't efficient at all.

YHRen, 2020-10-03:
Hi I have experienced the same problem after the update. SKR 1.4 T with Pi4 connected by GPIO pins. 

@CaptainSweatpants  can you `scp` the firmware ? I will give it a try after the current print is done.

benlye, 2020-11-27:
Duplicate of #149.  Closing this one because the other issue has more conversation.

closed 2020-11-27 by benlye