OctoPrint/OctoPrint-FirmwareUpdater #146: Unable to flash to skr V2.0, avrdude errors

TwoToneEddy, 2020-08-18:
Hi,
I've tried to flash a hex file to my SKR mini e3 V2.0 using this plugin and i'm not having much luck, here are the errors I get:

`Avrdude says: 'not in sync: resp=0x20
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x54
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x32
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x35
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x2e
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x33
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x2f`

Could this be something to do with not being in DFU mode? I've installed stm32flash.
Thanks

benlye, 2020-08-18:
You need to properly configure the plugin to flash your board - you are trying to use avrdude (a tool for flashing 8-bit) MCU's to flash a 32-bit board.

I think the details you need are here:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#lpc1768-boards

With more information specific to the SKR Mini E3 in this issue:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/103

TwoToneEddy, 2020-08-19:
OK i've followed those instructions for the lpc1768. I still get the avrdude messages, why would it still be using avrdude?

benlye, 2020-08-19:
Because that is still the configured option, I guess.

Can you post screenshots of your settings?

TwoToneEddy, 2020-08-19:
Sorry yes you're right, I didn't even see there was a settings page. Should I select lpc1768 or stm32flash?

TwoToneEddy, 2020-08-19:
[image]

This doesn't look right, it doesn't look like it's mounting the SD card

benlye, 2020-08-19:
So you need to follow the instructions in the documentation...

TwoToneEddy, 2020-08-19:
Do you mean the parts about installing usbmount etc:

sudo apt-get install usbmount
Change FS_MOUNTOPTIONS in /etc/usbmount/usbmount.conf:

FS_MOUNTOPTIONS="-fstype=vfat,gid=pi,uid=pi,dmask=0022,fmask=0111"

Add the following to sudo systemctl edit systemd-udevd
[Service]
PrivateMounts=no
MountFlags=shared

sudo systemctl daemon-reload
sudo service systemd-udevd --full-restart

I've done this already

benlye, 2020-08-19:
I think you'll have to find a group for your printer and ask there. I don't have that board so can't really help. All I know is that others have been able to make it work. I asked them to provide documentation updates but I guess nobody did.

TwoToneEddy, 2020-08-19:
Ok i'll do some digging and if i'm successful i'll let you know how i did it. Should I be compiling to a bin or a hex? I would assume bin.

TwoToneEddy, 2020-08-19:
I figured this out, you need to be sure to use the STM32F103RC_btt_512K_USB platformio environment, not the STM32F103RC_btt_512K environment.