OctoPrint/OctoPrint-FirmwareUpdater #149: BTT SKR 1.4 Turbo -

gordonpt, 2020-09-21:
**Hardware Setup**
BTT SKR 1.4 Turbo

**Describe the problem**
Unable to upload firmware.

**Log Files**
pi@raspberrypi:~ $ ls /media/usb
firmware.bin
pi@raspberrypi:~ $ ls /media/usb0
firmware.bin
pi@raspberrypi:~ $ 

The message Flashing successful appear, but it doesn't flash the firmware.
the /media/USB it always available, even if i detached the serial connection, it should be the onboard sd card content?

LTrainExpress, 2020-09-23:
I have the same issue.

1) I can copy my firmware manually via scp to /media/usb0 and manually reset my board for the flash to work. This makes me think it isn't a USB issue. 
2) If I manually copy my firmware as described above and tell OctoPrintFirmwareUpdater to reboot the board before flashing, sometimes the board comes up, sometimes I need to pull the SD card. I think the file isn't fully flushed. Obviously when the board doesn't come up OctoPrintFirmwareUpdater complains that it cannot write the firmware file.
3) If I use OctoPrint-FirmwareUpdater as intended, the file is claimed to be written without error, but I monitor /media/usb and never see the file, so when the board reboots the update does not occur.

I was wondering if this was a permission issue, but I have it wide open and I assume that OFU would complain it cannot write if that were the case.

gawainXX, 2020-10-25:
what method are you using for flashing it? I see that LPC1768 is available but not LPT1769 which the BTT 1.4 Turbo is based on

CTCA10, 2020-11-14:
Hello,
To let it working with usbmount I had only to chmod usb0 to 777.
usb is by me pointing to usb0.

nikos1671, 2020-11-23:
any news on this Board??

benlye, 2020-11-27:
Unless someone with the board can contribute code or documentation (I'm not sure what is required) there's not a lot I can do...

nikos1671, 2020-11-27:
i have this board how can i help??

benlye, 2020-11-27:
> i have this board how can i help??

Can you describe how you manually flash the board and explain what would be needed in the plugin to be able to do the same?  I'm not sure I understand why this board is very different to flashing an lpc1768 board.

Also, I just added support for running pre-flash gcode commands.  Would running `M22` before the `M997` command help?

nikos1671, 2020-11-27:
Manual update: just inserting the microsd on the board and then power off power on and the new firmware is loading.
i have try m22 with m997, the board its making reboot i can see on the lcd screen but no new firmware is installed.

benlye, 2020-11-27:
Well, unless someone with a board can figure it out and provide me the details (or someone provides me a board) there's not a lot I can do.

gawainXX, 2020-11-27:
Manual update process via PC involves connecting the device via USB, the SD
card on the skr 1.4turbo will then appear in the list of drives, you then
upload the firmware file and then power cycle the controller.

I'll see if I can get the exact step by step and filename/format later
today.

benlye, 2020-11-27:
> Manual update process via PC involves connecting the device via USB, the SD card on the skr 1.4turbo will then appear in the list of drives, you then upload the firmware file and then power cycle the controller. I'll see if I can get the exact step by step and filename/format later today.

It's the power-cycle of the controller which presents the problem.  

It's easy enough to copy the file to the SD card, but if a software reset doesn't trigger the update we're a bit stuck.  Can you find out if there's another way to trigger the firmware upgrade once the file is copied?

benlye, 2020-11-27:
`M997` is supposed to work for LPC176x boards.
https://marlinfw.org/docs/gcode/M997.html

I'm guessing that the vendor of this board may have done something in the bootloader so that it only updates after a power cycle.  In that case you might need to look at hooking up a GPIO pin to reset the board (if there is a reset pin you can connect to).

Again, without access to the board, all I can do is guess...

nikos1671, 2020-11-27:
I have copy manual the bin in to the sdcard then i have execute the M997 printer reboot but no new firmware. then i take the sdcard and i check on my pc the bin file was still there.Next step was again the sdcard on the board power cycling the board and the firmware was there. so it mean board must be power cycle to take effect. :-(

gawainXX, 2020-11-27:
Being able to upload it directly to the board without having to connect it
directly to a PC would be helpful.   I'll tinker around and see if the RPI
auto mounts the sd media to a path. (i'm also wondering if it's possible to
transfer the firmware file via winscp)

nikos1671, 2020-11-27:
The only work around will be something like this [https://de.aliexpress.com/item/32972603577.html?spm=a2g0o.cart.0.0.18863c00jzMckn&mp=1](url)  or just a wifi plug and openhab it would work :-) just sent a command from the plugin to close and open the plug.

gawainXX, 2020-11-27:
Link doesn't work for me, is it some sort of relay?

nikos1671, 2020-11-27:
> Link doesn't work for me, is it some sort of relay?
> […](#)

Try now

benlye, 2020-11-28:
I'm thinking about adding a 'custom' flash method which would let you configure any arbitrary command or script to run as the flash method.  With something like that you could write a bash script which copied firmware file wherever it needed to go, and then control your relay (or smart plug), or even GPIO pins if that were an option to trigger the reset.

That's about all I can think of doing in the plugin for now for this board.

nikos1671, 2020-11-28:
Thats sound good..

thisiskeithb, 2020-12-02:
> `M997` is supposed to work for LPC176x boards (it's the method the plugin uses for LPC1768 boards).
> https://marlinfw.org/docs/gcode/M997.html
> 
> I'm guessing that the vendor of this board may have done something in the bootloader so that it only updates after a power cycle. In that case you might need to look at hooking up a GPIO pin to reset the board (if there is a reset pin you can connect to).
> 
> Again, without access to the board, all I can do is guess...

There's nothing special about the bootloader on these boards. Updating the SKR 1.4 Turbo (LPC1769) is no different than the SKR 1.4 (LPC1768).

I've also used this plugin on the SKR E3 Turbo (LPC1769) and it works.

I have found that BigTreeTech's TFTs can interfere with updating the motherboard firmware when in "Touch Mode" in case anyone is running that combination.

benlye, 2020-12-02:
> There's nothing special about the bootloader on these boards. Updating the SKR 1.4 Turbo (LPC1769) is no different than the SKR 1.4 (LPC1768).

That's not exactly what others are saying.  The posts above say that this board does not initiate a firmware update after a simple reset via `M997`, it requires a hard reset or power cycle.  Other SKR boards will perform the firmware update after an `M997` command.

If you know differently and want to write some documentation, that would be great.

thisiskeithb, 2020-12-02:
Like I said, I've tested it on the LPC1769-based boards from BigTreeTech and it works, so I don't know why it doesn't work for others.

I just tested the latest Marlin `bugfix-2.0.x` ([Permit SD EEPROM emulation on E3V2 (#20353) - a4d6908](https://github.com/MarlinFirmware/Marlin/commit/a4d6908d556afd77cf66ef7a8adc7c34fee81674)) on an SKR E3 Turbo and SKR 1.4 Turbo and both reboot & flash firmware correctly using `M997`.

This works identically to LPC1768-based boards like the SKR 1.3 and 1.4.

I'd need more details, preferably zipped copies for their `Configuration.h` & `Configuration_adv.h` from Marlin as well as a short description of attached hardware to see if there's something weird with configs or possibly a TFT as that will cause flashing to not occur if in "Touch Mode".

It'd also help to try the following:
- Use a new, high-quality SD card (like a Sandisk), that is <8GB
- Format the SD card as FAT32
- Manually delete `firmware.cur` before attempting to flash as some of BTT's bootloaders can't overwrite it and fails to flash

benlye, 2020-12-02:
So there are no changes required to the documentation at all?

I can just close this and the other open issues for SKR boards? That would be nice.

CBDesignS, 2020-12-02:
that could explain a lot then @thisiskeithb I can not even get the pi to see the sd card in tft mode. will try in marlin mode later if this slow print ever finishes. My board in under the printer out of the way and is a real pain to get to when updating firmware. I was really hoping this plugin would make my life a little easier and you may well have just cracked the nut for me . I keep forgetting tft mode is emulating a controller and is still very buggy.

thisiskeithb, 2020-12-02:
> So there are no changes required to the documentation at all?

Shouldn't be, no. I'm pinging some other folks with SKR 1.4 Turbos to see if they can reproduce the issue.

> I can just close this and the other open issues for SKR boards? That would be nice.

I took a quick glance at the issues list and the other open SKR issue I see is for an SKR Pro (https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/114) which is STM32F4/a different MCU.

Oh! I just remembered this "Toggle for SD mount on boot" PR https://github.com/MarlinFirmware/Marlin/pull/20099 which gives you a new `SD_IGNORE_AT_STARTUP` option.

> SD Card mounting on boot can be disabled. This is useful if you regularily want to flash a firmware via internal SD-Card Reader with the FirmwareUpdater Plugin in Octoprint. The mounting of the SD on a RPi requires you to have the SD card not mounted in Marlin.

thisiskeithb, 2020-12-02:
> I keep forgetting tft mode is emulating a controller and is still very buggy.

I can easily reproduce lockups/weird behavior when trying to update the motherboard firmware when in "Touch Mode" every time I try it, so I tend to stick with "Marlin Mode". It may not be as pretty, but you can access most of Marlin's settings 🙂

CBDesignS, 2020-12-02:
Oh! I just remembered this "Toggle for SD mount on boot" PR MarlinFirmware/Marlin#20099 which gives you a new SD_IGNORE_AT_STARTUP option. I wondered what it was referencing as it was not in marlin from my last build at the start of november.

thisiskeithb, 2020-12-02:
> I wondered what it was referencing as it was not in marlin from my last build at the start of november.

You need to be running the latest `bugfix-2.0.x` since `2.0.7.2` was released at the end of October and `SD_IGNORE_AT_STARTUP` was added 21 days ago.

CBDesignS, 2020-12-02:
I have just downloaded the latest bugfix an hour or so ago and noticed quite a few changes. I normally just file compaire and change without reading.

parskofamily, 2020-12-28:
I would like to mention that I have been having this issue as well.  Identical.  Can't flash.  The test button works to find /media/usb0, so I was able to get that part working.  But, I am having an issue of not being able to "unmount the SD card".

I have gone through this thread above, I've changed permissions to usb0, commented out "SD_IGNORE_AT_STARTUP", but I still can't get it working.  I have a homemade printer I'm converting from a Mega/Ramps setup (printed beautifully yesterday before this).  I am upgrading to the identical drivers and setup to baseline, but with the SKR1.4Turbo.  So far, everything else seems to be working like the RAMPS.  But, I make a LOT of firmware changes, and this add-on has been wonderful with the previous setup.

I imagine it is something simple.  I'm sorry, I want to add more, but it's very late and I'm tired.  If anyone has any suggestions for me to try in the morning, please advise.

My point of posting is to note that with this board, there is more that needs to be done to get this working, I think.  I hope I'm wrong and I'm just tired.

benlye, 2020-12-28:
> I would like to mention that I have been having this issue as well. Identical. Can't flash. The test button works to find /media/usb0, so I was able to get that part working. But, I am having an issue of not being able to "unmount the SD card".

You don't have the same issue.  You most likely need to fix your sudo rights, as explained in the [documentation](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#sudo-rights), and in #172.

If that doesn't help:
1. Make sure you have the latest version of the plugin
2. Try to flash
3. Post your octoprint.log file

benlye, 2021-01-30:
I've updated the documentation and tested the described process on an SKR 1.4 Turbo board.  It works exactly as described/expected when properly configured.

closed 2021-01-30 by benlye