OctoPrint/OctoPrint-FirmwareUpdater #150: [Feature Request] RUMBA32 Upload (STM32 microcontroller support)

Celcius1, 2020-10-12:
I use a RUMBA32 board, but I need to connect the Boot and Reset pins to gpio to put the board into upload mode, so to enter upload mode the boot pin needs to be pulled high, and reset pulled low, and the general process to enter upload mode is hold boot press reset, whilst still holding boot, then release boot approx 1 to 2 sec after pressing reset. The board will be in upload mode waiting for a file.

So is it possible to get this change made to the plugin so I can get the updater to put the board into upload so I can upload the firmware from octoprint not an external computer, also does your plugin support the STM programming executable?

vhol, 2020-11-01:
Simple, +1.
It will be nice to map BOOT and RESET signals to GPIO pins of Raspberry Pi. Thank you.

benlye, 2020-11-27:
It seems like you could use a pre-flash shell script to set the GPIO pins as needed with [gpio](http://wiringpi.com/the-gpio-utility/).

Aside from that, what does it take to update the firmware?

closed 2021-02-06 by github-actions[bot]