OctoPrint/OctoPrint-FirmwareUpdater #151: timeout communicating with programmer

ricky0178, 2020-11-09:
**Hardware Setup**
Megatronics 2.0 with delta custom-made 3d printer running marlin

**Describe the problem**
I'm trying to update the marlin software by flashing the firmware.hex file generated by platform.io
After configuring the plugin trying to flash it will give me a timeout error on avrdude, by saying it got a timeout

**Log Files**
octoprint.log
`2020-11-09 17:06:30,495 - octoprint.startup - INFO - ******************************************************************************
2020-11-09 17:06:30,498 - octoprint.startup - INFO - Starting OctoPrint 1.4.2
2020-11-09 17:06:30,499 - octoprint.startup - INFO - ******************************************************************************
2020-11-09 17:06:31,515 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: offline
2020-11-09 17:06:31,516 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is not working
2020-11-09 17:06:31,518 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is not working
2020-11-09 17:06:31,610 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-11-09 17:06:33,567 - octoprint.plugin.core - INFO - Found 19 plugin(s) providing 19 mixin implementations, 34 hook handlers
2020-11-09 17:06:33,936 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2020-11-09 17:06:33,991 - octoprint.server - INFO - Intermediary server started
2020-11-09 17:06:33,992 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2020-11-09 17:06:34,128 - octoprint.plugin.core - INFO - Found 19 plugin(s) providing 19 mixin implementations, 34 hook handlers
2020-11-09 17:06:34,164 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2020-11-09 17:06:34,611 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2020-11-09 17:06:34,615 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: u"Need(method='role', value=u'plugin_action_command_notification_show')")
2020-11-09 17:06:34,617 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: u"Need(method='role', value=u'plugin_action_command_notification_clear')")
2020-11-09 17:06:34,618 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: u"Need(method='role', value=u'plugin_action_command_prompt_interact')")
2020-11-09 17:06:34,620 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: u"Need(method='role', value=u'plugin_announcements_read')")
2020-11-09 17:06:34,622 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: u"Need(method='role', value=u'plugin_announcements_manage'), Need(method='role', value=u'plugin_announcements_read')")
2020-11-09 17:06:34,624 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: u"Need(method='role', value=u'plugin_appkeys_admin')")
2020-11-09 17:06:34,625 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: u"Need(method='role', value=u'plugin_backup_access')")
2020-11-09 17:06:34,627 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: u"Need(method='role', value=u'plugin_firmware_check_display')")
2020-11-09 17:06:34,629 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: u"Need(method='role', value=u'plugin_logging_manage')")
2020-11-09 17:06:34,630 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: u"Need(method='role', value=u'plugin_pi_support_check')")
2020-11-09 17:06:34,632 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: u"Need(method='role', value=u'plugin_pluginmanager_manage')")
2020-11-09 17:06:34,634 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: u"Need(method='role', value=u'plugin_pluginmanager_install'), Need(method='role', value=u'plugin_pluginmanager_manage')")
2020-11-09 17:06:34,636 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: u"Need(method='role', value=u'plugin_softwareupdate_check')")
2020-11-09 17:06:34,637 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: u"Need(method='role', value=u'plugin_softwareupdate_update')")
2020-11-09 17:06:34,771 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2020-11-09 17:06:36,535 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2020-11-09 17:06:37,667 - octoprint.util.pip - INFO - Version of pip is 19.3.1
2020-11-09 17:06:37,669 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2020-11-09 17:06:37,670 - octoprint.util.pip - INFO - ==> pip ok -> yes
2020-11-09 17:06:37,679 - octoprint.plugin.core - INFO - Initialized 19 plugin implementation(s)
2020-11-09 17:06:37,696 - octoprint.plugin.core - INFO - 19 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  File Check (2020.8.7) (bundled) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_file_check
|  Firmware Check (2020.9.23) (bundled) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmware_check
|  Firmware Updater (1.7.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmwareupdater
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Login UI (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/loginui
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2020-11-09 17:06:37,715 - octoprint.environment - INFO - Detected environment is Python 2.7.16 under Linux (linux2). Details:
|  hardware:
|      cores: 4
|      freq: 1200.0
|      ram: 915709952
|  os:
[... 190 lines cut ...]
2020-11-09 17:07:32,355 >          SCK period      : 0.1 us
2020-11-09 17:07:32,356 > 
2020-11-09 17:07:32,357 > avrdude: AVR device initialized and ready to accept instructions
2020-11-09 17:07:32,358 > avrdude: Device signature = 0x1e9801 (probably m2560)
[... cut ...]
2020-11-09 17:07:53,035 > 
2020-11-09 17:07:58,041 > avrdude: stk500v2_ReceiveMessage(): timeout
2020-11-09 17:07:58,056 > 
2020-11-09 17:08:03,027 > avrdude: stk500v2_ReceiveMessage(): timeout
2020-11-09 17:08:03,029 > 
2020-11-09 17:08:08,035 > avrdude: stk500v2_ReceiveMessage(): timeout
2020-11-09 17:08:08,036 > 
2020-11-09 17:08:13,042 > avrdude: stk500v2_ReceiveMessage(): timeout
2020-11-09 17:08:13,044 > avrdude: stk500v2_getsync(): timeout communicating with programmer
2020-11-09 17:08:13,046 > 
`

benlye, 2020-11-09:
Are you really using an STK500v2 programmer?  Most mega2560 boards are flashed with a 'wiring' programmer.  What happens if you change that?

Also, if you're going to paste log files, please use code tags around the text - it makes it much easier to read.  Otherwise, you can just attach the files rather than pasting the content.

ricky0178, 2020-11-09:
I've always used the STK500v2 programmer with avrdude, but i sure can try with wiring. Unfortunately I can't right now, but tomorrow i'll test it and i'll update my post. 
Also, sorry for the log file, i'll attach future logs as files

benlye, 2020-11-10:
If you're familiar with avrdude's command line, all you have to do is find a command which works on your OctoPrint server.  Once you know what that is, you make sure that the plugin is using the same command.  You can completely edit the flash tool command line in advanced settings.

I don't have your board so there's not really much more I can do to help - the problem is between avrdude and your board, the plugin doesn't do anything more than transfer the file and run the command.

closed 2020-11-22 by benlye