OctoPrint/OctoPrint-FirmwareUpdater #152: Issues when attempting to flash Creality v1.1.4 mainboard.

Will-wastelander, 2020-11-22:
**Hardware Setup**
Creality v1.1.4 Creality Ender 3 Pro

**Describe the problem**
Bootloader has been installed, Marlin 2.0.7.x has already been flashed, and is working, but I am updating some settings in the firmware, and want to flash the new hex file. When attempting to flash newly compiled marlin 2.0.7.x, I get the following error.

Flashing failed
Avrdude says: 'not in sync: resp=0x65
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x63
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x68
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x6f
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x73
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x74
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x61
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x72
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x74

avrdude done. Thank you.'
Printer Display

**Log Files**

[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5578885/octoprint.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5578886/plugin_firmwareupdater_console.log)

**Additional context**

[image]

Will-wastelander, 2020-11-22:
I get the following when running the command manually.

/usr/bin/avrdude -v -q -p m1284p -c arduino -P /dev/ttyUSB0 -D -b 115200 -U flash:w:./firmware.hex:i

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/pi/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 5.0
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: reading input file "./firmware.hex"
avrdude: writing flash (124768 bytes):
avrdude: stk500_loadaddr(): (a) protocol error, expect=0x14, resp=0x65

avrdude: stk500_paged_write(): (a) protocol error, expect=0x14, resp=0x63
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
[... 128 lines cut ...]
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync
avrdude: stk500_cmd(): programmer is out of sync

benlye, 2020-11-22:
This isn't a problem with the plugin - you need to solve whatever problem is preventing the avrdude command from working from your OctoPrint host.

closed 2020-11-22 by benlye

Will-wastelander, 2020-11-22:
So even tho I have arguing selected, and its trying to stk500 to program
the board, its not a bug ?

benlye, 2020-11-22:
> So even tho I have arguing selected, and its trying to stk500 to program the board, its not a bug ?
> […](#)

No.  

You can see in your log that the plugin is running the same avrdude command that you tried on the command line:
`/usr/bin/avrdude -v -q -p m1284p -c arduino -P /dev/ttyUSB0 -D -U flash:w:/tmp/tmplirmewrm:i`

The error your getting is just a standard avrdude error message.  You need to figure out how to flash your board with avrdude then configure the plugin accordingly.  If you've used the Arduino IDE to compile and upload to your board you may be able to get the correct command from the verbose output that the Arduino tools give.