OctoPrint/OctoPrint-FirmwareUpdater #153: [Feature Request]

Zuckme, 2020-11-26:
Regarding my question [here](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/103#issuecomment-733865516),
I would like to be able to umount the SD/USB Drive before M997.

After some research I believe the post-flash command is executed after the M997, so it is too late to umount the drive then.

Thanks a lot and sorry if it is a stupid request.

Keep up with this nice work!

benlye, 2020-11-26:
Right, the post-flash command has to run after M997 because M997 initiates flashing.  I believe you're looking for a pre-flash command.  Therefore this is a duplicate of #113.

I'll take another look at that and see if I can implement it - it shouldn't be too much work, but I need to make sure it handles the printer not being connected when the flash is triggered.

closed 2020-11-26 by benlye

Zuckme, 2020-11-27:
@benlye , I did my homework and analyzed your code.
I have to apologize, according to [here](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/4cd8b5a75180485b525ba4a9151977b93b2935d3/octoprint_firmwareupdater/methods/lpc1768.py#L95) you do umount the drive before M997. So everything is fine, I do not need this feature anymore you are doing everything correctly.
Moreover according to my .log I see no errore only a "Firmware updated", so I can report your plugin seems to work even on a FreeBSD box.
Thanks for all your support.