OctoPrint/OctoPrint-FirmwareUpdater #156: Firmware icon in navbar missing

harpagophytum, 2020-11-27:
Hello,
I updated the firmware and the icon appears in the navigation bar, but not correctly.

[image]

https://fontawesome.com/v4.7.0/icon/microchip
`<i class="fa fa-microchip" aria-hidden="true"></i>`

I think it's located here.
[octoprint_firmwareupdater/templates/firmwareupdater_navbar.jinja2](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/c8902c9b5dbaa44cda6afbcfd47eb197ebb92dbc/octoprint_firmwareupdater/templates/firmwareupdater_navbar.jinja2)

`<i class="fas fa-microchip" title="Firmware Updater"></i>` 

Regards,

benlye, 2020-11-27:
Did you enable it?

harpagophytum, 2020-11-27:
Yes of course.
I restarted the system to.

[image]

How you can see, the icon is activated but its path is missing.

harpagophytum, 2020-11-27:
don't you think the problem is here.

`- fas fa-microchip`
`+ fa fa-microchip`

benlye, 2020-11-27:
Working fine here.
[image]

What version of OctoPrint are you on?  I just remembered I'm on a 1.5 RC.  Will go back to stable and see if that makes a difference.

benlye, 2020-11-27:
OK, think I found the problem.  Will need to release a new version shortly.

benlye, 2020-11-27:
Fixed in v1.7.3.  

I had inadvertently created a dependency on Font Awesome v5, which I happen to have on my system due to another plugin I run.  Changing the syntax to `fa` instead of `fas` makes my plugin use the built-in Font Awesome v4 icons.  Will look at bundling FA5 in the future.

closed 2020-11-27 by benlye

harpagophytum, 2020-11-27:
Ok, Thank you for your responsiveness.