OctoPrint/OctoPrint-FirmwareUpdater #157: [FR] Remember last firmware file

hamster65, 2020-11-28:
I just discovered the navbar icon... very handy.
Now if the plug in would remember the last firmware file, it would only take one click on the navbar icon and then on "flash from file" to start the upgrade.

benlye, 2020-11-28:
What OS are you on?  On Windows it remembers last folder and always opens that when I click Browse button.  Did you know you can click the down arrow and see a list of recent files?

[image]

It's really not many clicks, but I do see your point if you're doing a lot of compiling and flashing.  I'll see if I can implement your feature anyway, but no promises...

hamster65, 2020-11-28:
I am on Linux with KDE. There is a history, too, but you still have to open the dialog. 

There is also https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/74 and https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/39 from 2018 and 2019 where others asked for similar (but not exactly the same) features (including myself).

benlye, 2020-11-28:
This is part of the code I inherited from the previous developer.  I've never really had to look at how it works until now.  I'm still not completely 100% clear on it, but I don't think this will be possible...

The problem is that the file selection box does not return the full path to the file back to the web UI, so I never 'know' it in order to store it.  When the file is selected the data is read from it immediately and passed to the back-end when the flash is started.  All that is retained in the UI is the file name, size, and last modified date.

It would be a massive security hole if the browser just allowed the page to read an arbitrary file from your system when the page is loaded, so even if I could store the full path to the file I wouldn't be able to read the content of it in order to send it to the back-end.  You would still have to do _something_ to open the file and get the data from it.

I can't see how implementing this would be possible without a large re-write of the UI.  Sorry.

hamster65, 2020-11-28:
Thanks for looking into it!

benlye, 2020-12-01:
This doesn't seem possible with the way things work today.  If I can think of a way to do it in the future I'll give it another go.

closed 2020-12-01 by benlye