OctoPrint/OctoPrint-FirmwareUpdater #158: [Feature Request] Umount command in Pre-Flash and Post-Flash Settings

Zuckme, 2020-12-02:
**Hardware Setup**
Octoprint on FreeBSD with Python venv
Board: SKR E3 Mini V2.0

**Describe the problem**
I am experiencing sporadic mount problems after flashing a FW, The SD/USB disk can sporadically not monuted again after a flash.
Looking at the logs I find no issues at all:

`2020-12-01 20:36:01,149 - octoprint.plugins.firmwareupdater - INFO - Executing pre-flash commandline 'mount -t msdosfs /dev/label/sd3dp /mnt/sd_3dp'
2020-12-01 20:36:01,507 - octoprint.plugins.firmwareupdater - INFO - Pre-flash command 'mount -t msdosfs /dev/label/sd3dp /mnt/sd_3dp' returned: 0
2020-12-01 20:36:01,508 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2020-12-01 20:36:01,508 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2020-12-01 20:36:01,520 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2020-12-01 20:36:01,524 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-12-01 20:36:01,528 - octoprint.plugins.firmwareupdater - INFO - Release the firmware lock on the SD Card by sending 'M22' to '/dev/cuaU0'
2020-12-01 20:36:01,641 - octoprint.plugins.firmwareupdater - INFO - Waiting for SD card to be available at '/mnt/sd_3dp'
2020-12-01 20:36:01,643 - octoprint.plugins.firmwareupdater - INFO - Firmware update folder '/mnt/sd_3dp' available for writing after 0.0 seconds
2020-12-01 20:36:01,643 - octoprint.plugins.firmwareupdater - INFO - Copying firmware to update folder '/tmp/tmp4gxqvcjm' -> '/mnt/sd_3dp/firmware.bin'
2020-12-01 20:36:01,657 - octoprint.plugins.firmwareupdater - INFO - Synchronizing cached writes to SD card
2020-12-01 20:36:02,708 - octoprint.plugins.firmwareupdater - INFO - Unmounting SD card: 'sudo umount /mnt/sd_3dp'
2020-12-01 20:36:02,712 - octoprint.plugins.firmwareupdater - INFO - Firmware update reset: attempting to reset the board
2020-12-01 20:36:02,712 - octoprint.plugins.firmwareupdater - INFO - Resetting LPC1768 at '/dev/cuaU0'
2020-12-01 20:36:03,974 - octoprint.plugins.firmwareupdater - INFO - Waiting for LPC1768 at '/dev/cuaU0' to reset
2020-12-01 20:36:04,178 - octoprint.plugins.firmwareupdater - INFO - LPC1768 at '/dev/cuaU0' is resetting
2020-12-01 20:36:07,409 - octoprint.plugins.firmwareupdater - INFO - LPC1768 at '/dev/cuaU0' reset in 3.43 seconds
2020-12-01 20:36:07,409 - octoprint.plugins.firmwareupdater - INFO - Flashing successful.
2020-12-01 20:36:07,412 - octoprint.plugins.firmwareupdater - INFO - Setting run_postflash_gcode flag to true
2020-12-01 20:36:07,447 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/cuaU0, `

So, I started to check the code and sure enough [here](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/4cd8b5a75180485b525ba4a9151977b93b2935d3/octoprint_firmwareupdater/methods/lpc1768.py#L95)

After some tests I believe `'sudo umount ' + lpc1768_path` can not work properly on my system:

`$ mount -t msdosfs /dev/label/sd3dp /mnt/sd_3dp
$ sudo umount /mnt/sd_3dp
-sh: sudo: not found
$ umount /mnt/sd_3dp
$
`
What is even more confusing, I think python is not raising an exception (!) according to the logs.
That's why [I though everything was working](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/153#issuecomment-734938090).

I will try to patch the lpc1768.py locally in my box and report back if the situation improve.

**Feature Request**

It would be nice to have a Umount command in Pre-Flash and Post-Flash Settings since Octoprint could run on systems where `sudo umount` is not supported.

**Log Files**
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5630640/octoprint.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5630641/plugin_firmwareupdater_console.log)

**Additional context**
I could install sudo on FreeBSD but then I need to input the psw to execute the command.

benlye, 2020-12-02:
Can you help me understand why you can't just put the commands you want to run in the existing pre-flash and post-flash system command options?

I suppose I could also make it so that the existing umount command is editable so you could remove `sudo`.

benlye, 2020-12-02:
Ah, actually I re-read that and I think I understand more clearly what you want now.

You want the existing lpc1768 pre-flash `umount` command to be configurable in the settings so that you can remove the `sudo` prefix , which you don't need on your system.  That seems reasonable - you shouldn't have to add your own pre-flash command to unmount the media (although you could).

That makes sense for pre-flash, but what about post-flash - there isn't a post-flash umount today.  When in the sequence do you need that?

And yeah, I'll see if I can figure out why an exception isn't being raised if the umount command fails.

benlye, 2020-12-02:
Also, I don't have an lpc1768 board to test with anymore, so any code you can contribute or test will be awesome.

Zuckme, 2020-12-02:
> That makes sense for pre-flash, but what about post-flash - there isn't a post-flash umount today. When in the sequence do you need that?

Sorry I was referring to the page in the plugin. I need something which can be executed after the firmare upload is completed and before the M997. Currently it is impossible to get access to a custom umount command.

benlye, 2020-12-02:
So if you could edit the unmount command in the settings that would solve your problem (because that is after the copy but before the `M997`)?

That would be easy enough to do.

benlye, 2020-12-02:
[image]

I should have a new version for you to test shortly.

benlye, 2020-12-02:
New pre-release available now.  To upgrade you can do a manual plugin installation from a URL using this package URL:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/1.7.4.zip

The command will default to `sudo umount {mountpoint}`.  `{mountpoint}` will be substituted for the firmware folder path when the command is run.  Deleting the entire command will disable the unmount step in the flash process.

Please let me know if it works for you.

Zuckme, 2020-12-03:
Sorry for the late reply I was working.

I did the update but I believe something went wrong...

[image]

Moreover I can see the magic umont command and I can edit it

[image]

but if I click "Save" nothing happens and I am stuck on the Plugin Configuration page, if I click "Cancel" then I can quit the that page. 

PS: Thank you for your patience and support!

benlye, 2020-12-03:
Ugh - I see the problem - fixing it now.

benlye, 2020-12-03:
Please try this version:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/1.7.5.zip

Zuckme, 2020-12-03:
.... and winner winner chicken dinner! I flashed just once now. It took more than expected but everything went well. No error messages.

here the logs:
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5637744/octoprint.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5637745/plugin_firmwareupdater_console.log)

I will continue to test it to ensure the fix is stable.
Since you were very quick and friendly you earned a donation from my side!

Good job!

benlye, 2020-12-03:
Awesome, good to hear, and thanks so much for the donation!

I'll leave this issue open for a few days in case you run into any problems, then I'll release this build as a regular release.

closed 2020-12-28 by benlye