OctoPrint/OctoPrint-FirmwareUpdater #162: Using this plugin with the SKR Mini E3 V2 (STM32F103RCT6)

isMyUsername, 2020-12-03:
**Hardware Setup**
Ender 3, SKR Mini E3 v2

**Describe the problem**
I've tried every permutation of BOOT0 pin / RESET pin but I'm unable to flash successfully

**Log Files**
Octoprint.log

> 2020-12-02 20:31:30,909 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2020-12-02 20:31:30,911 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2020-12-02 20:31:30,988 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2020-12-02 20:31:31,010 - octoprint.plugins.firmwareupdater - INFO - Running '/usr/bin/stm32flash -v -i dtr,-rts,rts,-dtr -g 0x8000000 -w /tmp/tmp037qdizc /dev/ttyACM0' in /usr/bin
2020-12-02 20:31:31,014 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-12-02 20:31:32,477 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Cannot enter bootloader, check your BOOT0/Reset pins settings and try again.
2020-12-02 20:31:32,485 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyACM0, baudrate=115200, profile={'axes': {'e': {'inverted': False, 'speed': 300}, 'x': {'inverted': False, 'speed': 6000}, 'y': {'inverted': False, 'speed': 6000}, 'z': {'inverted': False, 'speed': 200}}, 'color': 'default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}, 'heatedBed': True, 'heatedChamber': False, 'id': '_default', 'model': 'Ender 3', 'name': 'Ender', 'volume': {'custom_box': False, 'depth': 220.0, 'formFactor': 'rectangular', 'height': 250.0, 'origin': 'lowerleft', 'width': 220.0}}
2020-12-02 20:31:32,585 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"
2020-12-02 20:31:32,587 - octoprint.util.comm - INFO - Connecting to port /dev/ttyACM0, baudrate 115200
2020-12-02 20:31:32,606 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial connection" to "Connecting"
2020-12-02 20:31:32,610 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0

plugin_firmwareupdater_console.log

> 2020-12-02 20:31:31,029 /usr/bin/stm32flash -v -i dtr,-rts,rts,-dtr -g 0x8000000 -w /tmp/tmp037qdizc /dev/ttyACM0
2020-12-02 20:31:32,467 > stm32flash 0.5
2020-12-02 20:31:32,468 > 
2020-12-02 20:31:32,469 > http://stm32flash.sourceforge.net/
2020-12-02 20:31:32,470 > 
2020-12-02 20:31:32,471 > Using Parser : Raw BINARY
2020-12-02 20:31:32,472 > Interface serial_posix: 57600 8E1
2020-12-02 20:31:32,473 > 
2020-12-02 20:31:32,474 > 
2020-12-02 20:31:32,475 > Failed to init device.

**Additional context**
I can update from SD without any issues

benlye, 2020-12-03:
Without a board to experiment with there's really nothing I can do with this.

isMyUsername, 2020-12-03:
Any way I can send you a board or you can tell me what to experiment on? Either way thank you for the plugin, it worked great with the Melzi board

benlye, 2020-12-03:
Sure, you could send me a board if you wanted to.  My email address is available in my GitHub profile if you want to get in touch to work out logistics.

Andy-ABTec, 2020-12-06:
Actually, I had this working on my SKR Mini E3 V2 using the LPC1768 instructions. It was S-L-O-W mounting on OctoPrint but it worked.

A couple of days ago I upgraded OctoPrint to V1.5.1, but what I didn't notice at the time was that FirmwareUpdater also updated... 
```
2020-12-04 18:56:42,944 - octoprint.plugins.softwareupdate - INFO - Starting update of octoprint to 1.5.1...
2020-12-04 18:57:31,070 - octoprint.plugins.softwareupdate - INFO - Update of octoprint to 1.5.1 successful!
2020-12-04 18:57:31,275 - octoprint.plugins.tracking - INFO - Sent tracking event update_successful, payload: {'target': 'octoprint', 'from_version': '1.5.0', 'to_version': '1.5.1'}
2020-12-04 18:57:31,392 - octoprint.plugins.softwareupdate - INFO - Starting update of firmwareupdater to 1.7.5...
2020-12-04 18:57:41,317 - octoprint.plugins.softwareupdate - INFO - Update of firmwareupdater to 1.7.5 successful!
2020-12-04 18:57:41,548 - octoprint.plugins.tracking - INFO - Sent tracking event update_successful, payload: {'target': 'firmwareupdater', 'from_version': '1.7.3', 'to_version': '1.7.5'}
2020-12-04 18:57:41,771 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2020-12-04 18:57:41,774 - octoprint.plugins.softwareupdate - INFO - Restarting...
```
Interestingly, there is nothing in the plugin_firmwareupdater.log after that point...
```
2020-12-04 18:57:39,039 > Building wheels for collected packages: OctoPrint-FirmwareUpdater
2020-12-04 18:57:39,040 > Building wheel for OctoPrint-FirmwareUpdater (setup.py): started
2020-12-04 18:57:41,308 > Building wheel for OctoPrint-FirmwareUpdater (setup.py): finished with status 'done'
2020-12-04 18:57:41,310 > Created wheel for OctoPrint-FirmwareUpdater: filename=OctoPrint_FirmwareUpdater-1.7.5-py3-none-any.whl size=24132 sha256=9ec420e266a72764c29ee188f55661964a62139311a342d861aa2af945e2fe94
2020-12-04 18:57:41,311 > Stored in directory: /tmp/pip-ephem-wheel-cache-j2g86dwq/wheels/60/63/4f/07dd1e3ef9fd0e09d868362c35cd289d7e049b01bcd4a8037c
2020-12-04 18:57:41,312 > Successfully built OctoPrint-FirmwareUpdater
2020-12-04 18:57:41,312 > Installing collected packages: OctoPrint-FirmwareUpdater
2020-12-04 18:57:41,313 > Attempting uninstall: OctoPrint-FirmwareUpdater
2020-12-04 18:57:41,313 > Found existing installation: OctoPrint-FirmwareUpdater 1.7.3
2020-12-04 18:57:41,314 > Uninstalling OctoPrint-FirmwareUpdater-1.7.3:
2020-12-04 18:57:41,314 > Successfully uninstalled OctoPrint-FirmwareUpdater-1.7.3
2020-12-04 18:57:41,314 > Successfully installed OctoPrint-FirmwareUpdater-1.7.5
```
Something to do with _"Make LPC1768 unmount command configurable via advanced settings"_ maybe?

benlye, 2020-12-06:
@Andy-ABTec not sure what you're saying. Have you tried to flash after the update and it didn't work? The plugin normally logs most output to octoprint.log.

Andy-ABTec, 2020-12-06:
Apologies - Trying to get to the bottom of this is quite frustrating...

I'm running Octoprint 1.5.1 with FirmwareUpdater 1.7.5
Both the OctoPrint and FirmwareUpdater Console are set to the DEBUG level
Cleaned out all the logs before rebooting the printer and the pi

Octoprint has the SKR Mini card mounted to /media/usb0 and I can scp the firmware file to it and then reboot the printer manually.
 
The advanced setting reset before flashing is ticked and the command is set to "umount /media/usb0", though I have tried "sudo umount {mountpoint}" as well to no avail

The message I'm getting when using the FirmwareUpdater in OctoPrint is: "Flashing failed: unable to unmount SD Card"

The firmwareupdaterconsole log is empty - no file exists

Have attached the octoprint log
[octoprint (3).log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5649014/octoprint.3.log)

Happy to do whatever I can to help resolve this

benlye, 2020-12-06:
Can you install this version?
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/rc.zip

You need to install it manually via 'install from URL' in the plugin manager. 

It has extra logging for the unmount commands so that you will be able to see why it is failing. 

The umount command has most likely always been failing on your system, but now that failure is being bubbled up to the UI.

Andy-ABTec, 2020-12-06:
Woo-Hoo! thanks for the quick response...

Adding "`pi      ALL=NOPASSWD: /bin/mount,/bin/umount`" to the sudoers file did it! I've just "successfully flashed" again!

One last quick question - is there a way to speed up the mount - an entry in fstab maybe?

bakaufman, 2020-12-06:
> Woo-Hoo! thanks for the quick response...
> 
> Adding "`pi ALL=NOPASSWD: /bin/mount,/bin/umount`" to the sudoers file did it! I've just "successfully flashed" again!
> 
> One last quick question - is there a way to speed up the mount - an entry in fstab maybe?

which sudoers file did you update? creat a new one? I am assuming this is in etc/sudoers.d. Thanks!

benlye, 2020-12-06:
> which sudoers file did you update? creat a new one? I am assuming this is in etc/sudoers.d. Thanks!

I added some new instructions on the documentation... 

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#sudo-rights

bakaufman, 2020-12-06:
@benlye That did it. I got it to work without reboot.

Andy-ABTec, 2020-12-07:
Just run "`sudo visudo`" and add "`pi      ALL=NOPASSWD: /bin/mount,/bin/umount`" to the end of the file...

benlye, 2020-12-07:
> One last quick question - is there a way to speed up the mount - an entry in fstab maybe?

I think that will be dependent on your system and the board.  I don't have the board so can't really test.  You'll have to try a few things and see if you can improve it.

Good to know that this board works though - I'll add it to the documentation.

Andy-ABTec, 2020-12-07:
No worries, I'll come back if I find anything...

cybercatgurrl, 2020-12-16:
> 
> 
> Woo-Hoo! thanks for the quick response...
> 
> Adding "`pi ALL=NOPASSWD: /bin/mount,/bin/umount`" to the sudoers file did it! I've just "successfully flashed" again!
> 
> One last quick question - is there a way to speed up the mount - an entry in fstab maybe?

Using the current version this was the trick that let me flash it successfully. Well, that and the usbmount install guide from README on the main page. Thanks.

closed 2020-12-28 by benlye