OctoPrint/OctoPrint-FirmwareUpdater #164: Cannot save flashing method

ravingDIODE, 2020-12-04:
I am trying to use this plugin to flash my CR-10S Pro V2 and when hitting the settings button to choose the flashing method etc, it won't save the settings.

Hitting the blue save button has no effect whatsoever and closing the dialogue box doesn't save the settings either.

Is this a bug or am I missing something?

[image]

https://prnt.sc/vvi0oe

benlye, 2020-12-04:
It looks like you installed the v1.7.4 pre-release. Can you try v1.7.5?

ravingDIODE, 2020-12-04:
> It looks like you installed the v1.7.4 pre-release. Can you try v1.7.5?

How do I do that? I am using the plugin manager in OctoPrint and it seems it only detects the 1.7.3 and 1.7.4:

[image]

benlye, 2020-12-04:
Click the circled button:
[image]

ravingDIODE, 2020-12-04:
Strange, still gives the same...

[image]

benlye, 2020-12-04:
Try removing it and re-installing anyway.

benlye, 2020-12-04:
There was a new release on GitHub which was tagged as a pre-release.  It seems like the Plugin Manager doesn't show pre-releases in the UI, but pulls the latest pre-release when you install.  I've removed the broken release (1.7.4) and made 1.7.5 a 'proper' release.

ravingDIODE, 2020-12-04:
Thank you very much! Got it updated and working now.

Thanks for the quick feedback and support, really appreciate it!

benlye, 2020-12-04:
Thanks for letting me know.

closed 2020-12-04 by benlye