OctoPrint/OctoPrint-FirmwareUpdater #167: "Unable to unmount SD card" error after last update

RafaelSchridi, 2020-12-04:
**Hardware Setup**
SKR 1.3 on a Ender 5

**Describe the problem**
The UI shows an Error while unmounting. checking the /media/usb it is indeed still mounted, but unmounting the same way with "umount /media/usb" on the pi it works just fine. And if i send a M997 to the printer it updates the firmware just fine.
[image]

**Log Files**
[octoprint.log-4.txt](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5643192/octoprint.log-4.txt)
[plugin_firmwareupdater_console.log-2.txt](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5643193/plugin_firmwareupdater_console.log-2.txt)

**Additional context**
/

benlye, 2020-12-04:
Previous versions of the plugin failed to unmount silently if the unmount command failed.  Now you get an error.  I'm surprised that the sudo command fails, but I can't tell you why that happens on your system (maybe your don't have sudo).

Either way, you've fixed it already.

benlye, 2020-12-04:
I was a bit hasty with the close button.  I think there is probably an edge case where the unmount command will fail if pre-flash reset is enabled but the SD card is not mounted.  I'll need to solve that.

This isn't your problem though because your logs show you're not using pre-flash reset.  I can't explain why the `sudo umount` command fails, but can only guess that it's to do with the user that OctoPrint is running as not having enough rights.

Have you got any other system commands configured, e.g. the system command to shut the host down?  Does it work?

RafaelSchridi, 2020-12-04:
I've only got the Reboot system & Shutdown system setup, if you mean those they're working just fine.

I assume the user that runs the umount command is "pi"? because the tutorial goes through setting up permissions for the "pi" user.

benlye, 2020-12-04:
Can you try setting your unmount command to this:
`sudo umount {mountpoint}  2> >(tee ~/.octoprint/logs/umount.log)`

Then try to flash and `~/.octoprint/logs/umount.log` will contain the error message that umount is giving.

benlye, 2020-12-04:
Scratch that - I thought that would work but it doesn't.  I'll have to add some debug logging to that step and get you to try it out a test build.

Meanwhile, in order to flash your board just clear out the unmount command which will disable it, and your flash should succeed.

Nascire, 2020-12-04:
In my case, I have connected 2 printers, and modified usbmount to create a symlink based on the SD card partition label.

So my path for the firmware is only a symlink: lrwxrwxrwx  1 root root  11 Dec  4 17:25 SKR14 -> /media/usb2
This, of course, cannot be unmounted - I just cleared the command as I do a Pre-Reset, but others may need a combination of symlink removal and unmount?

Maybe this helps :)

RafaelSchridi, 2020-12-04:
By default usbmount creates a symlink from /media/usb to /media/usb0, perhaps this could be the issue? I'm currenly printing, but when it's done I'll try to change the unmount command to point to /media/usb0.

benlye, 2020-12-04:
> In my case, I have connected 2 printers, and modified usbmount to create a symlink based on the SD card partition label.
>
> So my path for the firmware is only a symlink: lrwxrwxrwx 1 root root 11 Dec [...]

Actually, you can unmount a symlink:
```
pi@octopi:/ $
pi@octopi:/ $ ls -al /media/
total 52
drwxr-xr-x 10 root root  4096 Dec  4 16:03 .
drwxr-xr-x 21 root root  4096 Feb 13  2020 ..
lrwxrwxrwx  1 root root     4 Dec  4 16:03 usb -> usb0
drwxr-xr-x  3 pi   pi   16384 Jan  1  1970 usb0
drwxr-xr-x  2 root root  4096 Dec  4 16:03 usb1
drwxr-xr-x  2 root root  4096 Dec  4 16:03 usb2
drwxr-xr-x  2 root root  4096 Dec  4 16:03 usb3
drwxr-xr-x  2 root root  4096 Dec  4 16:03 usb4
drwxr-xr-x  2 root root  4096 Dec  4 16:03 usb5
drwxr-xr-x  2 root root  4096 Dec  4 16:03 usb6
drwxr-xr-x  2 root root  4096 Dec  4 16:03 usb7
pi@octopi:/ $ mount | grep usb
/dev/sda1 on /media/usb0 type vfat (rw,nodev,noexec,noatime,nodiratime,sync,uid=1000,gid=1000,fmask=0111,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
pi@octopi:/ $ sudo umount /media/usb
pi@octopi:/ $ mount | grep usb
pi@octopi:/ $
```

Nascire, 2020-12-04:
> 
> Actually, you can unmount a symlink:
> 

My fault, misstyped the command ... but you led me to another thought - sudo
The default command uses sudo for umount - typing it manually asks for my password, and there is no NOPASSWD in the sudoers-File

benlye, 2020-12-04:
Can you try manually installing this version of the plugin?
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/rc.zip

You need to use 'Install from URL' and paste that URL in.  You should end up with 1.7.6rc0 installed.

Then you should then be able to see the error from the unmount command in the log file.  E.g., it errors for me if I remove `sudo`:
```
2020-12-04 17:35:10,039 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2020-12-04 17:35:10,041 - octoprint.plugins.firmwareupdater - INFO - Printer not connected, not sending pre-flash gcode commands
2020-12-04 17:35:10,055 - octoprint.plugins.firmwareupdater - INFO - Synchronizing cached writes to SD card
2020-12-04 17:35:11,269 - octoprint.plugins.firmwareupdater - INFO - Unmounting SD card: 'umount /media/usb'
2020-12-04 17:35:11,319 - octoprint.plugins.firmwareupdater - ERROR - Error executing unmount command 'umount /media/usb'
2020-12-04 17:35:11,343 - octoprint.plugins.firmwareupdater - ERROR - umount: /media/usb0: umount failed: Operation not permitted.
```

benlye, 2020-12-04:
> > Actually, you can unmount a symlink:
> 
> My fault, misstyped the command ... but you led me to another thought - sudo
> The default command uses sudo for umount - typing it manually asks for my password, and there is no NOPASSWD in the sudoers-File

It could very well be a bad sudo config that is causing the problem.  You should be able to 'sudo' commands on your Pi without being prompted for a password.

RafaelSchridi, 2020-12-04:
Oh it straight throws a stacktrace now:

>2020-12-04 20:46:16,316 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2020-12-04 20:46:18,192 - octoprint.plugins.portlister - INFO - Port Lister () {}
2020-12-04 20:53:42,317 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2020-12-04 20:53:42,336 - octoprint.plugins.firmwareupdater - INFO - Sending pre-flash gcode commands: M22
2020-12-04 20:53:42,360 - octoprint.plugins.firmwareupdater - INFO - Pre-flash delay: 5s
2020-12-04 20:53:42,489 - octoprint.plugins.portlister - WARNING - Not auto connecting because printer is not closed nor in error state.
2020-12-04 20:53:47,364 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2020-12-04 20:53:47,593 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2020-12-04 20:53:47,607 - octoprint.plugins.firmwareupdater - INFO - Release the firmware lock on the SD Card by sending 'M22' to '/dev/ttyACM0'
2020-12-04 20:53:47,635 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-12-04 20:53:47,674 - octoprint.plugins.firmwareupdater - INFO - Waiting for SD card to be available at '/media/usb'
2020-12-04 20:53:47,696 - octoprint.plugins.firmwareupdater - INFO - Firmware update folder '/media/usb' available for writing after 0.0 seconds
2020-12-04 20:53:47,698 - octoprint.plugins.firmwareupdater - INFO - Copying firmware to update folder '/tmp/tmpwBOGBP' -> '/media/usb/firmware.bin'
2020-12-04 20:53:47,846 - octoprint.plugins.firmwareupdater - INFO - Synchronizing cached writes to SD card
2020-12-04 20:53:49,920 - octoprint.plugins.firmwareupdater - INFO - Unmounting SD card: 'sudo umount /media/usb'
2020-12-04 20:53:49,922 - octoprint.plugins.firmwareupdater - ERROR - Error while attempting to flash
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmwareupdater/__init__.py", line 240, in _flash_worker
    if flash_callable(self, firmware=firmware, printer_port=printer_port):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmwareupdater/methods/lpc1768.py", line 127, in _flash_lpc1768
    e = sys.exc_info()[0]
NameError: global name 'sys' is not defined
2020-12-04 20:53:49,930 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyACM0, baudrate=115200, profile={'heatedChamber': False, 'name': 'Ender 5', 'color': 'default', 'axes': {'y': {'speed': 10000, 'inverted': False}, 'x': {'speed': 10000, 'inverted': False}, 'z': {'speed': 10000, 'inverted': True}, 'e': {'speed': 120, 'inverted': False}}, 'heatedBed': True, 'volume': {'origin': 'lowerleft', 'formFactor': 'rectangular', 'depth': 240.0, 'width': 240.0, 'custom_box': False, 'height': 300.0}, 'model': 'Creality Ender 5', 'id': '_default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}}
2020-12-04 20:53:50,052 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"

benlye, 2020-12-04:
I didn't test it on Python 2.7. I guess I'll have to do that.

RafaelSchridi, 2020-12-04:
Do we have control over that? I thought it was up to the plugin devs?

benlye, 2020-12-04:
> Do we have control over that? I thought it was up to the plugin devs?

That would be me. 

What I changed is supposed to be compatible with Python 2.x and 3.x but that doesn't seem to be the case. It's working fine on my Pi, but I run Python 3.

RafaelSchridi, 2020-12-04:
I thought that the plugin ran in python 3 or 2 by itself. 

But I just read this post:
https://octoprint.org/blog/2020/09/10/upgrade-to-py3/
And I'm probably gonna switch to python 3 unless you need me to test some stuff for you.

benlye, 2020-12-04:
Stand by - nearly have a version ready which should work on both.  I'd appreciate you trying it on Python 2 for me.

benlye, 2020-12-04:
Could you try this version - URL is the same but version is now 1.7.6rc1:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/rc.zip

I haven't actually tested it on Python 2.7, but this time it really is supposed to be compatible...

RafaelSchridi, 2020-12-04:
Seems to be working, i'm getting a sudo error which is a problem on my part

```
2020-12-04 22:02:49,865 - octoprint.plugins.firmwareupdater - INFO - Release the firmware lock on the SD Card by sending 'M22' to '/dev/ttyACM0'
2020-12-04 22:02:49,938 - octoprint.plugins.firmwareupdater - INFO - Waiting for SD card to be available at '/media/usb'
2020-12-04 22:02:49,953 - octoprint.plugins.firmwareupdater - INFO - Firmware update folder '/media/usb' available for writing after 0.0 seconds
2020-12-04 22:02:49,956 - octoprint.plugins.firmwareupdater - INFO - Copying firmware to update folder '/tmp/tmpKMO_rp' -> '/media/usb/firmware.bin'
2020-12-04 22:02:49,997 - octoprint.plugins.firmwareupdater - INFO - Synchronizing cached writes to SD card
2020-12-04 22:02:52,061 - octoprint.plugins.firmwareupdater - INFO - Unmounting SD card: 'sudo umount /media/usb'
2020-12-04 22:02:52,190 - octoprint.plugins.firmwareupdater - ERROR - Error executing unmount command 'sudo umount /media/usb'
2020-12-04 22:02:52,192 - octoprint.plugins.firmwareupdater - ERROR - sudo: no tty present and no askpass program specified```

benlye, 2020-12-04:
Phew :)  

I'll leave you to dig into the sudo issue.  Let me know how it goes.

RafaelSchridi, 2020-12-04:
Ok, so I added `020_firmware_updater` to `/etc/sudoers.d/` 
With in it just:
`pi ALL=NOPASSWD: /bin/umount`
And everything is flashing correctly now.  
Might be worth it to add it to the wiki.

Altough the first time it errored out because it was already unmounted while unmounting(???)
```2020-12-04 22:25:57,388 - octoprint.plugins.firmwareupdater - INFO - Unmounting SD card: 'sudo umount /media/usb'
2020-12-04 22:25:57,552 - octoprint.plugins.firmwareupdater - ERROR - Error executing unmount command 'sudo umount /media/usb'
2020-12-04 22:25:57,554 - octoprint.plugins.firmwareupdater - ERROR - umount: /media/usb: not mounted. 
```   

Altough I can't reproduce that anymore.

Anyway, thanks for all the help, really appreciate it.

benlye, 2020-12-04:
Great.  I've added some better handling of umount errors to hopefully ensure we catch that edge case.

I'll make a new official release available in a few days which will include this, and I'll make sure the documentation is updated to make the required `sudo` rights clearer.

closed 2020-12-04 by benlye