OctoPrint/OctoPrint-FirmwareUpdater #172: Firmware updater failing with Octoprint 1.5.1

Testmangh, 2020-12-06:
**Hardware Setup**
SKR 1.3 Ender 3 PRO

**Describe the problem**
While trying to update the printers firmware it fails after the SD card is mounted on the host.

**Log Files**
https://drive.google.com/file/d/10rQfr5qTCYhPvcjfyeOUh57dlsMaNPRT/view?usp=sharing

**Additional context**
https://i.imgur.com/eQfgs3m.png

benlye, 2020-12-07:
Can't access your logs, but you most likely have the same problem (with sudo rights) as described/resolved in #167. 

Follow the instructions here to fix it:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#sudo-rights

Testmangh, 2020-12-07:
I'm gonna try, but it was working flawlessly before updating Octoprint to version 1.5.1

closed 2020-12-07 by Testmangh

benlye, 2020-12-07:
You only *think* it was working flawlessly but an error-handling bug meant it was failing silently.

Upgrading to OctoPrint 1.5.1 also upgraded the plugin to the latest version, where it now errors when the unmount command fails.