OctoPrint/OctoPrint-FirmwareUpdater #173: Flashing failed Timeout communicating with programmer

Laurelinwen, 2020-12-11:
**Hardware Setup**
I am useing a Geeetech A10T 3D Printer with a GT2560 Motherboard, which is an 8-bit processor motherboard that integrates all the functions of Arduino mega2560+Ultimaker and Arduino Mega2560+Ramps1.4.
Geeetech Support added following:
'ATmega' family 8-bit MCUs (RAMPS, Sanguinololu, Melzi, Anet, Creality, Ender.
I am also using a Raspberry Pi 4 B 8GB with Octoprint 1.5.2

**Describe the problem**
I am new to 3D Printing.
I updated the Firmware in the beginning for the BLTouch Copy 3DTouch using direct connection to my Desktop PC via USB and now i want to go back. I want to make it work via the OctoPi with the Firmwareupdater Plugin.
Ofcourse i could try updating Firmware to previous Version without BLTouch Fuction, connecting directly to PC via USB but where is the Fun in it right?

I have the Printer connected via USB to the OctoPi and the OctoPi connected via WiFi to my HomeNetWork. I enter to the OctoPi Surface via http://192.xxx.xxx.xxx. I removed SD Card from Printer (i also tried with SD Card inserted). I had a Webcam running at the OctoPi, tried with and without the Webcam.

I followed the Instructions from:
4 .Board-Specific Configuration
ATmega Boards
Avrdude Installation:
Raspberry Pi
sudo apt-get update
sudo apt-get install avrdude
Avrdude Configuration:
Atmega2560 | wiring | RAMPS, RAMbo, etc.
The output is similar to:
pi@octopi:~ $ which avrdude
/usr/bin/avrdude
pi@octopi:~ $
I did not modify the OctoPi installation except for the Password and Language & Location Settings.
I tryed a run with and without the following:
Add the full path to avrdude in the plugin settings. Did that and it returned Path is valide
Optional advanced settings are available for:
Baud rate -  I used the 250000 wich is shown in the Connection Window of OctoPrint
Avrdude config file - overrides the default config file with a custom one I tried to leave it default and added the Path to the config File. Both returned valide.
Disabling write verification - ignored that
Customizing the avrdude command line - ignored that
Disabling the bootloader warning - disables a warning which is shown the hex filename has 'bootloader' in it -ignopred that

Pre-Flash and Post-Flash Settings - tried both disabled and enabled without further configuration.

Flashing failes all the time with:"Flashing failed
Timeout communicating with programmer"

I am using wiring in the configuration but in the Logfiles there is a confusing Entry:"avrdude: serial_baud_lookup(): Using non-standard baud rate: 250000avrdude: stk500v2_ReceiveMessage(): timeout", maybe i missunderstand it but it looks for me that it is not useing wiring but using stk500v2. 

I hope the Logfiles can help you to help me. Thank you in Advanced.

Best Regards

[plugin_firmwareupdater_console (1).log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5678598/plugin_firmwareupdater_console.1.log)
[octoprint (1).log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5678599/octoprint.1.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5678600/plugin_firmwareupdater_console.log)
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5678601/octoprint.log)

**Log Files**
Attach the OctoPrint log and the Firmware Updater log Log files, which can be accessed under **Settings** -> **Logging** in OctoPrint, or in `~/.octoprint/logs/` on the host.  Both `octoprint.log` and `~/.octoprint/logs/plugin_firmwareupdater_console.log` should be included.

**Additional context**
Add any other context about the problem here.

benlye, 2020-12-11:
Set the baud rate back to 'default'.

Laurelinwen, 2020-12-11:
I will try. I think i did that. In the Plugin and in the OctoPi. But I am not sure so i will try. Thank you so far.

Best regards

benlye, 2020-12-11:
The baud rate that OctoPrint uses for serial communication to the printer will not necessarily be the same as the one that avrdude uses.  You can leave OctoPrint at 2500000, but you should set the plugin to 'default'.

Laurelinwen, 2020-12-11:
That was it. Setting to default made it possible. Many thanks.

Best regards

closed 2020-12-11 by benlye