OctoPrint/OctoPrint-FirmwareUpdater #176: Feature request: Permanent url

cociweb, 2021-01-03:
**Hardware Setup**
N/A

**Describe the problem**
Very usefull addon, esential for remote maintenance. I would like to request a 'feature':
All the time, when a a new firmware build is available, I have to search my url. Basically, my firmware build overwrites the old one, so the url is not changing. It takes some minutes on each build. Can I set somehow permanent url to earn some minutes on each time?

**Log Files**
N/A

**Additional context**
N/A

Thanks in advance,

benlye, 2021-01-03:
You're talking about flash from URL, not flash from file?

cociweb, 2021-01-03:
Yes, I mean flash from URL.

benlye, 2021-01-03:
OK, then it should be easy to do something for you :-)

cociweb, 2021-01-05:
Thank You! That could be awsome!

benlye, 2021-01-13:
Just wanted to let you know I hadn't forgotten this and I've just about got this ready for you to try out.

benlye, 2021-01-14:
If you switch the Firmware Updater plugin to the 'Development' release channel you will be able to install version 1.7.7b2, which includes an option to remember the last-used URL.

[image]

[image]

Please try it out and confirm that it works as you would expect.

cociweb, 2021-01-18:
Hello,
I've tested the version of 1.7.7b3 with different browsers (Firefox, Edge Chromium). Thank you for your time and effort! It's a little step from the add-on point of view, but it's a huge help for me to get closer to the CI.
Verdict:
It seems, that the feature works functionally.  But I've found a little glitch (rather a kitsch):
after the feature deactivation (uncheck the checkbox) requires double config modification. Here is what I've done:

1) check the checkbox-> save (OK)
2) add link to the url field (OK)
3) update the firmware (OK)
4) restart octoprint/system (OK) - I don't think it's neccessary
5) uncheck the checkbox ->save (maybe, It's OK - I don't remember well....)
6) re-check checkbox->save (NOK, it's still unchecked after the save: step6 repeated!->OK)
7) uncheck checkbox->save (NOK: it's still checked after the save: step7 repeated->OK!)
From now, It seems the step6/7  has effect on every 2nd save button press. (I've do not check other configuration saves, only this checkbox)
Br,
cociweb

benlye, 2021-01-18:
Strange, I cannot reproduce that at all (also using Edge).  I followed the steps you gave and at 6 and 7 the box is in the correct state when I go back into the plugin settings.

Could you do try this:
1. Open OctoPrint in the browser
2. Press Ctrl+F5 to refresh the page and all the JavaScript
3. Press F12 to open the JavaScript console
4. Toggle the checkbox and save the setting and see if you get any errors in the console

cociweb, 2021-01-18:
Hello,
Maybe I've catch it:
It doesn't really matter what I change....
If I click on the save there is message immediatelly with "ConfiguredLoggers has not changed. Not saving." by packed_core.js after that immediatelly, it sends a POST request (/api/settings) after that there are some hassle with the samesite cookie attribute. After that,  a few seconds(~5) later, everything is re-requested by octoprint (GET): /api/settings has the same json content in the response, but they differ from the request.

If I wait till the GET request, everything is ok. If I view the plugin settings between the POST and the GET request, there is the anomaly in FF & Edge as well.

[image]

diff of the POST/GET request/response:
[image]

From my readings it seems it's an issue of the core octoprint engine. - but I'm not sure.
BTW: I have no info about the behavior of the stable release, I don't want to loose the config while I'm reverting back.

benlye, 2021-01-18:
Ahh, OK.  I can reproduce the same thing if I go back into the plugin settings really quickly.

It seems like the main OctoPrint settings dialog solves this by delaying the modal closing until the settings really are saved - you see a spinner on the Save button while the settings are saved in the background.

I'll see if I can implement the same kind of thing.

cociweb, 2021-01-18:
Great!
Anyway, I don't insist to this improvement. If it is not a blocker to anyone else - I hope, not. :) (I can leave with the spinner/5 sec delay.)
May it's another issue, and I think it's not really stick with the original feature request.
What do you think?
Can/should we close this ticket/new feature?

benlye, 2021-01-18:
It's an annoying UX issue that I'm glad you spotted, and I thought it was worth fixing.  1.7.7b4 has the fix, if you'd like to try it.

I'll probably do the new release tomorrow.

cociweb, 2021-01-19:
What a quick fix!
1.7.7b4 is tested, it works as expected: the settings box does not disappear, until the settings are not saved. The spinner is spinning from the post request untill the whole get sequence. So there is no possibility, to reopen the window again, before the saving is not finnished. Stupid-sure release: that's what I need for me. :D
Thank you for your work!

benlye, 2021-01-21:
v1.7.7 released with this feature.

closed 2021-01-21 by benlye