OctoPrint/OctoPrint-FirmwareUpdater #177: Avrdude says: 'not in sync: resp=0x65

screechyboy79, 2021-01-06:
Stock Creality Ender 5 with 1.1.4 board

Whenever I attempt to install firmware Marlin or Creality officially I get Avrdude says: 'not in sync: resp=0x65

[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5776810/plugin_firmwareupdater_console.log)
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/5776811/octoprint.log)

benlye, 2021-01-06:
A quick Google search says that you need to install a bootloader on the board.  That's not something I can help with.

closed 2021-01-06 by benlye

screechyboy79, 2021-01-06:
I have a bootloader installed my BL Touch came with a USB ISP and I used ProgISP to install the bootloader following this guide https://www.reddit.com/r/ender3/comments/cfmbdy/howto_installing_a_bootloader_to_an_ender_3_pro/ although mine is the ATMEGA2560 and an Ender 5. This all runs smoothly as far as I can tell but I cannot get passed this error

benlye, 2021-01-06:
If you have an Atmega2560 then you need to configure the plugin correctly for that board.  Your logs show that it is configured for an Atmega1284p, which is what Google says the MCU on the 1.1.4 board is.

Basically, make sure your settings match your board and it will work.