OctoPrint/OctoPrint-FirmwareUpdater #180: Prusa MK3S - Flashing Failed, Error opening serial device

wasyl00, 2021-01-13:
**Hardware Setup**
Prusa MK3S connected to Octoprint on Rpi3 directly to Einsy Rambo via J19 pins. 

**Describe the problem**
I have a problem with using this plugin when trying to flash firmware I get error "Flashing failed Error opening serial device"
Octoprint works just fine, just the flashing fails. I tried two available serial port in options (see errors in the log files). Any suggestion?

Below are my settings: 
https://i.imgur.com/vJnMzB4.png

**Log Files**
**Firmware updater log:**
`2021-01-13 13:03:34,944 /usr/bin/avrdude -v -q -p m2560 -c arduino -P dev/ttyAMA0 -D -U flash:w:/tmp/tmpHID1Xr:i
2021-01-13 13:03:35,694 > 
2021-01-13 13:03:35,696 > avrdude: Version 6.3-20171130
2021-01-13 13:03:35,697 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2021-01-13 13:03:35,699 >          Copyright (c) 2007-2014 Joerg Wunsch
2021-01-13 13:03:35,700 > 
2021-01-13 13:03:35,701 >          System wide configuration file is "/etc/avrdude.conf"
2021-01-13 13:03:35,706 >          User configuration file is "/home/pi/.avrduderc"
2021-01-13 13:03:35,707 >          User configuration file does not exist or is not a regular file, skipping
2021-01-13 13:03:35,709 > 
2021-01-13 13:03:35,710 >          Using Port                    : dev/ttyAMA0
2021-01-13 13:03:35,712 >          Using Programmer              : arduino
2021-01-13 13:03:35,713 > avrdude: ser_open(): can't open device "dev/ttyAMA0": No such file or directory
2021-01-13 13:03:35,727 > 
2021-01-13 13:03:35,730 > avrdude done.  Thank you.
2021-01-13 13:03:35,733 > 
2021-01-13 13:03:35,735 > 
2021-01-13 13:03:53,280 
2021-01-13 13:03:53,311 /usr/bin/avrdude -v -q -p m2560 -c wiring -P dev/ttyAMA0 -D -U flash:w:/tmp/tmpww6N9Y:i
2021-01-13 13:03:54,092 > 
2021-01-13 13:03:54,094 > avrdude: Version 6.3-20171130
2021-01-13 13:03:54,096 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2021-01-13 13:03:54,097 >          Copyright (c) 2007-2014 Joerg Wunsch
2021-01-13 13:03:54,099 > 
2021-01-13 13:03:54,101 >          System wide configuration file is "/etc/avrdude.conf"
2021-01-13 13:03:54,103 >          User configuration file is "/home/pi/.avrduderc"
2021-01-13 13:03:54,104 >          User configuration file does not exist or is not a regular file, skipping
2021-01-13 13:03:54,106 > 
2021-01-13 13:03:54,107 >          Using Port                    : dev/ttyAMA0
2021-01-13 13:03:54,109 >          Using Programmer              : wiring
2021-01-13 13:03:54,110 > avrdude: ser_open(): can't open device "dev/ttyAMA0": No such file or directory
2021-01-13 13:03:54,111 > ioctl("TIOCMGET"): Inappropriate ioctl for device
2021-01-13 13:03:54,112 > ioctl("TIOCMGET"): Inappropriate ioctl for device
2021-01-13 13:03:54,113 > 
2021-01-13 13:14:09,500 
2021-01-13 13:14:09,506 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyS0 -D -U flash:w:/tmp/tmppP4ACS:i
2021-01-13 13:14:10,259 > 
2021-01-13 13:14:10,272 > avrdude: Version 6.3-20171130
2021-01-13 13:14:10,292 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2021-01-13 13:14:10,304 >          Copyright (c) 2007-2014 Joerg Wunsch
2021-01-13 13:14:10,312 > 
2021-01-13 13:14:10,320 >          System wide configuration file is "/etc/avrdude.conf"
2021-01-13 13:14:10,329 >          User configuration file is "/home/pi/.avrduderc"
2021-01-13 13:14:10,332 >          User configuration file does not exist or is not a regular file, skipping
2021-01-13 13:14:10,335 > 
2021-01-13 13:14:10,337 >          Using Port                    : /dev/ttyS0
2021-01-13 13:14:10,340 >          Using Programmer              : wiring
2021-01-13 13:14:10,340 > avrdude: ser_open(): can't set attributes for device "/dev/ttyS0": Inappropriate ioctl for device
2021-01-13 13:14:10,341 > ioctl("TIOCMGET"): Bad file descriptor
2021-01-13 13:14:10,341 > ioctl("TIOCMGET"): Bad file descriptor
2021-01-13 13:14:10,342 > avrdude: ser_drain(): select(): Bad file descriptor
2021-01-13 13:14:10,343 > avrdude: ser_send(): write error: Bad file descriptor
2021-01-13 13:14:10,343 > avrdude: stk500_send(): failed to send command to serial port
2021-01-13 13:14:10,344 > avrdude: ser_recv(): select(): Bad file descriptor
2021-01-13 13:14:10,344 > avrdude: stk500v2_ReceiveMessage(): timeout
2021-01-13 13:14:10,345 > avrdude: ser_send(): write error: Bad file descriptor
2021-01-13 13:14:10,346 > avrdude: stk500_send(): failed to send command to serial port
2021-01-13 13:14:10,349 > avrdude: ser_recv(): select(): Bad file descriptor
[... 60 lines cut ...]
2021-01-13 13:14:09,451 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2021-01-13 13:14:09,473 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2021-01-13 13:14:09,491 - octoprint.plugins.firmwareupdater - INFO - Running '/usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyS0 -D -U flash:w:/tmp/tmppP4ACS:i' in /usr/bin
2021-01-13 13:14:10,365 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Timeout communicating with programmer.
2021-01-13 13:14:10,370 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=dev/ttyAMA0, baudrate=115200, profile={'heatedChamber': False, 'name': 'Prusa i3 MK3', 'color': 'default', 'axes': {'y': {'speed': 6000, 'inverted': False}, 'x': {'speed': 6000, 'inverted': False}, 'z': {'speed': 200, 'inverted': False}, 'e': {'speed': 300, 'inverted': False}}, 'heatedBed': True, 'volume': {'origin': 'lowerleft', 'formFactor': 'rectangular', 'depth': 210.0, 'width': 250.0, 'custom_box': {'z_min': 0.0, 'y_min': -4.0, 'x_max': 250.0, 'x_min': 0.0, 'y_max': 210.0, 'z_max': 200.0}, 'height': 200.0}, 'model': 'Prusa i3 MK3', 'id': '_default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}}
2021-01-13 13:14:10,482 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"
2021-01-13 13:14:10,498 - octoprint.util.comm - INFO - Connecting to port dev/ttyAMA0, baudrate 115200
2021-01-13 13:14:10,523 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial connection" to "Error: Connection error, see Terminal tab"
2021-01-13 13:14:10,535 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=Error
2021-01-13 13:14:10,541 - octoprint.util.comm - INFO - Changing monitoring state from "Error: Connection error, see Terminal tab" to "Offline (Error: Connection error, see Terminal tab)"
2021-01-13 13:14:10,554 - octoprint.util.comm - ERROR - Unexpected error while connecting to serial port dev/ttyAMA0, baudrate 115200 from hook default: SerialException: '[Errno 11] Could not exclusively lock port dev/ttyAMA0: [Errno 11] Resource temporarily unavailable' @ comm.py:_open_serial:3670
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3670, in _open_serial
    settings().getFloat(["serial", "timeout", "connection"]),
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3644, in default
    serial_obj.open()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/serial/serialposix.py", line 312, in _reconfigure_port
    raise SerialException(msg.errno, "Could not exclusively lock port {}: {}".format(self._port, msg))
SerialException: [Errno 11] Could not exclusively lock port dev/ttyAMA0: [Errno 11] Resource temporarily unavailable
2021-01-13 13:14:10,558 - octoprint.plugins.action_command_notification - INFO - Notifications cleared`

**Additional context**
N/A

benlye, 2021-01-13:
I've never tried this, but I've looked at my Pi3 and done some googling.

It looks like you need to use `/dev/ttyS0` (NOT `/dev/ttyAMA0`, which is mapped to Bluetooth) in order to use the GPIO serial ports.  Your logs show that you don't have a `/dev/ttyS0` device, and neither do I on my Pi3.  It seems that this is because in the latest Pi releases ("Jessie / Stretch / Buster releases (as of August 2019)") it needs to be manually enabled.

Take a look at this page to get more information and see how to enable it:
https://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3-4/

I did the steps to enable the GPIO port and now I have `/dev/ttyS0` on my Pi3.  It was not there before.

wasyl00, 2021-01-13:
My /boot/config.txt file shows that I have enable_uart=1 

Also when I run
$ ls -l /dev
lrwxrwxrwx 1 root root           7 Jan 13 13:00 serial0 -> ttyAMA0
lrwxrwxrwx 1 root root           5 Jan 13 13:00 serial1 -> ttyS0

Does that mean that ports supposed to work fine?

benlye, 2021-01-13:
Hmm, maybe you just don't have permission to access `/dev/S0`?  I don't know what this message means:

```
2021-01-13 13:14:10,340 > avrdude: ser_open(): can't set attributes for device "/dev/ttyS0": Inappropriate ioctl for device
2021-01-13 13:14:10,341 > ioctl("TIOCMGET"): Bad file descriptor
2021-01-13 13:14:10,341 > ioctl("TIOCMGET"): Bad file descriptor
```

Seems like AVRDUDE doesn't like that port for some reason.  

This isn't a problem with the plugin, it's something in your hardware/device configuration which you'll have to resolve.

dpackham, 2021-02-07:
Did you resolve this? I am getting the same errors

wasyl00, 2021-02-07:
Sadly no.

> Did you resolve this? I am getting the same errors

Nope, still looking.

benlye, 2021-02-07:
What option(s) are you using in `/boot/config.txt` to enable the GPIO serial port?

wasyl00, 2021-02-07:
This is my whole config.txt

`
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128

enable_uart=1
dtoverlay=pi3-disable-bt

`

benlye, 2021-02-07:
What about `/boot/cmdline.txt`?

You also need to disable the serial console on the default serial device.  You can use `raspi-config` to do this.

I have just confirmed that I can use the GPIO header pins on my Pi3 to program a Mega2560.  The only fiddly part was resetting the board at the right time.

In `/boot/config.txt` I have this: at the end:
```
enable_uart=1
#dtoverlay=pi3-disable-bt
```

Note the last line is disabled.

Then in cmdline.txt:
```
console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

Note the `console=` part.

And then (with some practice getting the reset timing right) ....
```
pi@octopi:~ $ sudo avrdude -p atmega2560 -P /dev/ttyS0 -c wiring -v

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyS0
         Using Programmer              : wiring
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD

avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D8, L:FF)

avrdude done.  Thank you.

pi@octopi:~ $
```

benlye, 2021-02-07:
If you use the disable-bluetooth overlay by adding `dtoverlay=pi3-disable-bt` in `/boot/config.txt`, you must use `/dev/ttyAMA0` instead of `/dev/ttyS0` :

```
pi@octopi:~ $ sudo avrdude -p atmega2560 -P /dev/ttyAMA0 -c wiring -v

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyAMA0
         Using Programmer              : wiring
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD

avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D8, L:FF)

avrdude done.  Thank you.

pi@octopi:~ $

```

But you still need to disable the serial console, which is what is blocking the device and causing this error:
```
avrdude: ser_open(): can't set attributes for device "/dev/ttyS0": Inappropriate ioctl for device
ioctl("TIOCMGET"): Bad file descriptor
```

wasyl00, 2021-02-07:
Many thanks. I'll try this and report back after I finish my several hour print.

benlye, 2021-02-07:
The other thing you can do, if your board exposes the ICSP pins, is use `linuxgpio` with avrdude.

This is a good tutorial:
https://ozzmaker.com/program-avr-using-raspberry-pi-gpio/

I got it working in a few minutes.  You don't have to mess with serial ports, and it takes care of resetting the board. You need a dew extra wires though.

benlye, 2021-02-17:
I'm going to assume that you got this working.

closed 2021-02-17 by benlye

dpackham, 2021-02-17:
I did not.  I had to get a USB cable and my laptop to do the update.  My Pi
running octo is directly connected to the board via the pins and does not
work

benlye, 2021-02-17:
> I did not. I had to get a USB cable and my laptop to do the update. My Pi running octo is directly connected to the board via the pins and does not work

Must be something strange going on with your setup then because I had no problems doing it as described above.  Shame it didn't work for you.