OctoPrint/OctoPrint-FirmwareUpdater #184: SKR 1.4 Turbo

z0n1PS, 2021-01-27:
mount: /mnt/usb: can't find in /etc/fstab.

followed the guide but can not mount USB or USB0 ...
how to fix this

benlye, 2021-01-27:
I don't think you have followed the guide - /mnt and fstab are never mentioned.

You need to install usbmount and have that handle mounting the device for you.  If it doesn't work, you have some problem with your setup that you need to fix.

z0n1PS, 2021-01-29:
all is installed, as in the documentation, but when I try to mount media/usb, it gives the error, and not possible to flash the firmware since it can not mount the SD card to upload the firmware

benlye, 2021-01-29:
Nowhere in the instructions are you told to mount /media/usb manually.  usbmount handles the mounting automatically for you.

If that is not working then either:
1. Your usbmount installation is broken
2. Your printer firmware was not compiled with the correct SD support options and is not exposing
3. Something else in your system/setup is stopping it working

Unfortunately, despite me asking several times, no one with one of these boards has contributed to the documentation explaining precisely what firmware options make this board work correctly, so you you will need to read through the other issues for the board and figure out what other people have tried.  [This](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/149#issuecomment-737510778) seems like a good start.  I do not have the board so cannot help.

Bottom line, this is not a bug in the plugin.  It will work if you set your system and firmware up correctly.

z0n1PS, 2021-01-29:
Thanks for the info

benlye, 2021-01-29:
If you get it to work please come back and tell me how.  I would really like to get the documentation updated so that people don't have issues with this board any more.

benlye, 2021-01-30:
I just received my BTT SKR 1.4 Turbo.

I plugged it in and it immediately mounted to /media/usb (I already had usbmount installed and configured per the instructions).  I was able to use the FirmwareUpdater plugin to flash new firmware to it straight away, just by configuring the LPC1768 method as per the instructions.

The firmware I flashed had the following options enabled:
```
#define SDSUPPORT
#define SD_IGNORE_AT_STARTUP 
#define SDCARD_CONNECTION ONBOARD
```

I have since flashed it several times (including back to stock FW) and it has succeeded every time.  I will add these FW options to the documentation, then I'm done with this board.

benlye, 2021-01-31:
I decided to upgrade the RAMPS setup on my trusty old A8 with this board, so it's now my daily driver.

I can say that if all the instructions are followed correctly, it 100% works with the plugin - I've been able to flash my board a dozen or more times today as I've dialed my config and settings in.

I worked out that `SD_IGNORE_AT_STARTUP` is optional.  You can leave that out if you want to use the SD card for Marlin (e.g. for EEPROM emulation), it just means flashing is a little slower while the SD card is released by Marlin then mounted by the OctoPrint host.  I've made one change to the plugin to put a very short delay in the sequence after the `M22` command is sent, which makes flashing more reliable if Marlin has the board mounted.  That change will be released soon in v1.8.0 (an RC can be installed now from the RC or Development branch).

The RC branch currently has all the documentation updates which I'll be pushing to the main branch tomorrow.

benlye, 2021-02-07:
If you're feeling brave, support for flashing using Binary File Transfer has been added in the latest beta version, available on the Development release channel:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/releases/tag/1.8.1b0

More information in issue #161.

I'm going to close this issue as SKR 1.4 Turbo flashing definitely works if the instructions to set up usbmount are followed correctly, and in the new beta it works really well with binary file transfer.

closed 2021-02-07 by benlye