OctoPrint/OctoPrint-FirmwareUpdater #190: Request support for flashing Ender 3 V2 firmware

k7beb, 2021-02-02:
**Hardware Setup**
Creality 4.2.2 32-bit board
Ender 3 V2 printer

**Describe the problem**
Cannot find information on how to flash the Creality 4.2.2 board using FirmwareUpdater. (Note: similar 4.2.7 board should also be supported).

**Log Files**
N/A

**Additional context**
This is the stock board in the Ender 3 V2 printer and is shipped with a bootloader that will load a firmware update from SD card. If updating through FirmwareUpdater is supported I haven't been able to find instructions for doing so. 

This is a large and growing segment of the user base. I'm sure the ability to update firmware would be very popular.

benlye, 2021-02-02:
I'm in the middle of updating and reorganizing the documentation.  From what I can find the board has an STM32 MCU and should work like an SKR Pro.

Try the instructions here and see if they work for you:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/rc/doc/lpc176x.md

k7beb, 2021-02-02:
Unfortunately using usbmount doesn't seem to work. It may be the firmware - I'm using Smiths3D firmware and do not compile it myself so I'm not able to add #define SDCARD_CONNECTION ONBOARD

I appear to have successfully installed and configured per the directions but ls /media/usb returns nothing, and any attempt to mount /media/usb informs me that it is not in /etc/fstab. 

I'm no great shakes at linux so I may have done something wrong, of course.

benlye, 2021-02-03:
Does it show up as a storage device if you plug it into Windows?  Do you have an option on the printer to unmount/release the SD card?  If you do, does it show up as a storage device then?

If you can't make the board present a storage device via USB then there's pretty much no way to get the plugin to work.

(The only other option would be the Marlin Binary File Transfer protocol, but I haven't added that feature yet, and you would still need to compile your own firmware with it enabled.)

k7beb, 2021-02-03:
Unfortunately I don't have any practical way to connect a Windows PC to my printer to check. I'm going to assume it's because the Smith3D firmware is not configured to allow access to the SD Card over USB so this can't be done.

benlye, 2021-02-03:
OK.  I'll try to remember to come back and update this issue when #161 is implemented.

k7beb, 2021-02-03:
I'll watch that issue as well, thanks.

LekeFly, 2021-02-16:
Been trying to get this to work for a hours now. Got the 4.2.2 board. I see there's also multiple people looking for a solution. Does anyone have this running on a Ender 3 V2 yet? Please give us some hints.

https://www.reddit.com/r/ender3v2/comments/l5hp13/has_anyone_gotten_the_octoprint_firmwareupdater/

benlye, 2021-02-16:
> Been trying to get this to work for a hours now. Got the 4.2.2 board. I see there's also multiple people looking for a solution. Does anyone have this running on a Ender 3 V2 yet? Please give us [...]

Have you tried using binary file transfer from the devel branch? (see https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/161#issuecomment-774475946)

LekeFly, 2021-02-16:
Thank you @benlye. I had a hard time understanding i had to run "~/oprint/bin/pip install marlin-binary-protocol" but i got both checkmarks, but when i run this happens:

```2021-02-16 14:33:45,534 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
2021-02-16 14:33:45,534 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
2021-02-16 14:34:27,431 - octoprint.plugins.firmwareupdater - INFO - Python package 'marlin-binary-protocol' is installed: True
2021-02-16 14:34:27,433 - octoprint.plugins.firmwareupdater - INFO - Marlin BINARY_FILE_TRANSFER capability is enabled: True
2021-02-16 14:34:27,439 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2021-02-16 14:34:27,441 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2021-02-16 14:34:27,506 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2021-02-16 14:34:27,513 - octoprint.plugins.firmwareupdater - INFO - Current Baud: 115200
2021-02-16 14:34:27,519 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2021-02-16 14:34:27,521 - octoprint.plugins.firmwareupdater - INFO - Initializing Marlin BFT protocol
2021-02-16 14:34:27,538 - octoprint.plugins.firmwareupdater - INFO - pySerial Version: 3.5
2021-02-16 14:34:27,568 - octoprint.plugins.firmwareupdater - INFO - Connecting to printer at '/dev/ttyUSB0' using Marlin BFT protocol
2021-02-16 14:34:27,571 - octoprint.plugins.firmwareupdater - INFO - Connecting: Switching Marlin to binary protocol...
2021-02-16 14:34:27,702 - octoprint.plugins.firmwareupdater - INFO - Connection synced [0], binary protocol version 0.1.0, 96 byte payload buffer
2021-02-16 14:34:27,702 - octoprint.plugins.firmwareupdater - INFO - Transfering file to printer using Marlin BFT '/tmp/tmplfqvyjrf' -> /firmware.bin
2021-02-16 14:34:47,920 - octoprint.plugins.firmwareupdater - INFO - Binary file transfer complete
2021-02-16 14:34:48,937 - octoprint.plugins.firmwareupdater - INFO - Firmware update reset: attempting to reset the board
2021-02-16 14:34:48,938 - octoprint.plugins.firmwareupdater - INFO - Resetting printer at '/dev/ttyUSB0'
2021-02-16 14:34:49,021 - octoprint.plugins.firmwareupdater - INFO - Waiting for printer at '/dev/ttyUSB0' to reset
2021-02-16 14:34:59,199 - octoprint.plugins.firmwareupdater - ERROR - Timeout waiting for board reset to init
2021-02-16 14:34:59,200 - octoprint.plugins.firmwareupdater - ERROR - Board reset failed
2021-02-16 14:34:59,204 - octoprint.plugins.firmwareupdater - ERROR - Reset failed```

And "Flashing failed - Board reset failed" Thanks for the quick reply

Verbose:
```
2021-02-16 14:46:18,267 - octoprint.plugins.firmwareupdater - INFO - Current Baud: 115200
2021-02-16 14:46:18,269 - octoprint.plugins.firmwareupdater - INFO - Initializing Marlin BFT protocol
2021-02-16 14:46:18,276 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2021-02-16 14:46:18,280 - octoprint.plugins.firmwareupdater - INFO - pySerial Version: 3.5
2021-02-16 14:46:18,305 - octoprint.plugins.firmwareupdater - INFO - waiting 5s after protocol connect
2021-02-16 14:46:23,312 - octoprint.plugins.firmwareupdater - INFO - Connecting to printer at '/dev/ttyUSB0' using Marlin BFT protocol
2021-02-16 14:46:23,316 - octoprint.plugins.firmwareupdater - INFO - Connecting: Switching Marlin to binary protocol...
2021-02-16 14:46:23,337 - octoprint.plugins.firmwareupdater - INFO - Connection synced [0], binary protocol version 0.1.0, 96 byte payload buffer
2021-02-16 14:46:23,337 - octoprint.plugins.firmwareupdater - INFO - Transfering file to printer using Marlin BFT '/tmp/tmp8squfmly' -> /firmware.bin
2021-02-16 14:46:23,356 - octoprint.plugins.firmwareupdater - INFO - File Transfer version: 0.1.0, compression: heatshrink
2021-02-16 14:46:23,401 - octoprint.plugins.firmwareupdater - INFO - Opened file: firmware.bin
2021-02-16 14:46:23,487 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.00% 6.28KiB/s [7.52KiB/s] Errors: 0
2021-02-16 14:46:23,501 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.07% 6.58KiB/s [7.88KiB/s] Errors: 0
2021-02-16 14:46:23,515 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 0.14% 6.64KiB/s [7.95KiB/s] Errors: 0
...
2021-02-16 14:46:43,938 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 99.86% 6.66KiB/s [7.98KiB/s] Errors: 0
2021-02-16 14:46:43,946 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 99.93% 6.66KiB/s [7.98KiB/s] Errors: 0
2021-02-16 14:46:43,946 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 100.00% 6.66KiB/s [7.98KiB/s] Errors: 0
2021-02-16 14:46:43,969 - octoprint.plugins.firmwareupdater - INFO - File closed
2021-02-16 14:46:43,969 - octoprint.plugins.firmwareupdater - INFO - Transfer complete
2021-02-16 14:46:43,969 - octoprint.plugins.firmwareupdater - INFO - Binary file transfer complete
2021-02-16 14:46:44,989 - octoprint.plugins.firmwareupdater - INFO - Firmware update reset: attempting to reset the board
2021-02-16 14:46:44,991 - octoprint.plugins.firmwareupdater - INFO - Resetting printer at '/dev/ttyUSB0'
2021-02-16 14:46:45,084 - octoprint.plugins.firmwareupdater - INFO - Waiting for printer at '/dev/ttyUSB0' to reset
2021-02-16 14:46:55,308 - octoprint.plugins.firmwareupdater - ERROR - Timeout waiting for board reset to init
2021-02-16 14:46:55,309 - octoprint.plugins.firmwareupdater - ERROR - Board reset failed
[... cut ...]
2021-02-16 14:46:55,320 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyUSB0, baudrate=115200, profile={'axes': {'e': {'inverted': False, 'speed': 300}, 'x': {'inverted': False, 'speed': 6000}, 'y': {'inverted': False, 'speed': 6000}, 'z': {'inverted': False, 'speed': 200}}, 'color': 'default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}, 'heatedBed': True, 'heatedChamber': False, 'id': '_default', 'model': 'Ender-3 V2', 'name': 'Ender-3', 'volume': {'custom_box': False, 'depth': 220.0, 'formFactor': 'rectangular', 'height': 250.0, 'origin': 'lowerleft', 'width': 220.0}}
```

benlye, 2021-02-16:
It looks like the `M997` command doesn't reset the board.  Presumably if you reset it manually the firmware would get updated?

You could try sending `M997` to the printer from your OctoPrint terminal to see if the board responds to it.

benlye, 2021-02-16:
@LekeFly I've just seen [your post on Discord](https://discord.com/channels/704958479194128507/705046975669600298/811236138794483763).  Copying and pasting the M997 output here:
```
Send: M997
Recv: start
Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Send: N0 M110 N0*125
Recv: PowerUp
Recv: Marlin bugfix-2.0.x
Recv: 
Recv: echo: Last Updated: 2021-02-16 | Author: (none, default config)
Recv: echo:Compiled: Feb 16 2021
Recv: echo: Free Memory: 49079  PlannerBufferBytes: 1344
Recv: 
Recv: DWIN handshake error.
Recv: BL24CXX Check succeeded!
Recv: echo:SD card ok
Recv: ok
Send: N0 M110 N0*125
Recv: ok
Send: M155 S2
Recv: ok
```

I need to know if the board is actually resetting when it receives the `M997` command.  It looks like it might do some kind of soft reset.  If you use the firmware updater plugin, does it actually succeed in updating the firmware when this happens?  The timestamp in the snippet you pasted was from today, so maybe it did??

From what I can tell:
1 .The firmware.bin file is getting copied to the SD card
1. `M997` is getting sent
1. The printer _might_ be soft-resetting
1. The plugin reports failure because it doesn't detect the serial port going away and coming back

Can you confirm whether or not the firmware has been updated at this point?  If not, is it updated if you hard-reset the board now?  The firmware.bin file should be in place, so I would expect it to be updated....

I'm thinking two things:
* If `M997` is resetting the board and the firmware is updated I need to update the plugin to correctly detect the reset (I'm hoping I can hook into the `start` message)
* If the `M997` command isn't enough to actually reset the board (and get it to update the firmware) I don't know what I'll be able to do for you - what other methods do you have to reset the board?

Sennevds, 2021-02-21:
@benlye I've just tried to flash my ender 3 V2 with the new beta marlinbft method and it works! (I've recompiled the firmware where I changed the build number and after flashing and checking the info screen it was updated!) I just receive flashing failed error in octoprint with following error: Board reset failed but the reset was successful!

benlye, 2021-02-21:
> @benlye I've just tried to flash my ender 3 V2 with the new beta marlinbft method and it works! (I've recompiled the firmware where I changed the build number and after flashing and checking the [...]

That's great! I'll work on adding detection for the `start` message the boards sends when it resets, but for now you can use the 'Don't wait for reset' option to avoid getting the error message.

Sennevds, 2021-02-21:
Perfect. If you need anymore info or debugging just let me know.

jjfs127, 2021-02-21:
Is the only way to test is by enabling marlinbft in the configuration? I noticed that compiling fails if using meatpack - it seems either one or the other.

benlye, 2021-02-21:
> Is the only way to test is by enabling marlinbft in the configuration? I noticed that compiling fails if using meatpack - it seems either one or the other.

Yep, looks like it's one or the other (and yes, it's needed in the configuration if you want to use the marlinbft method).

I didn't know that, and I don't know why it's that way.  Hopefully the final version of binary file transfer doesn't have that limitation.

benlye, 2021-02-21:
It turns out that I can't detect the `start` message that the Ender board sends - at the time it is sent the printer is still disconnected, so OctoPrint doesn't see it and the plugin can't detect it.

It seems like for an Ender 3 V2, the best option is to use a combination of settings:
* Enable '**Don't wait for reset**' in the Advanced Flash Method Settings
   This will prevent the plugin from waiting for the serial port to disappear after sending M997 to the board and will avoid the "Board reset failed" error

Optionally:
* Add a '**Post flash delay**' time in the Pre and Post Flash Settings (set to the time it takes for the board to reset - e.g. 10s)
   This will make the plugin wait for the specified number of seconds before allowing OctoPrint to reconnect.  This should prevent OctoPrint trying to reconnect before or during the printer reset.

I don't know if the post flash delay is really necessary or not.

Hypertoken, 2021-02-22:
Can confirm the marlin-binary-protocol method works. Post flash delay unnecessary. However it was a 14min file transfer. Maybe I need to up the baud rate from 115200?

benlye, 2021-02-22:
> Can confirm the marlin-binary-protocol method works. Post flash delay unnecessary. However it was a 14min file transfer. Maybe I need to up the baud rate from 115200?

14 min!  That's crazy.  My board transfers the ~192KB firmware file in a few seconds at 115200.  It would be interesting to enable verbose progress logging and see the log, but to be honest there's probably not a lot I can do about it as the transfer protocol is completely outside my control and the developer of it is currently working on the release version, which is a big re-write of the current version.

Maybe try a different USB cable?

Sennevds, 2021-03-05:
Okay noticed one thing. The Ender 3 V2 needs a different filename every time you flash. The compile script creates a random filename but when I upload with your plugin it renames it to firmware.bin so this only works the first time but the second time the printer won't update because it's the same name as the previous firmware. Maybe an advanced option to keep the name?

jimcarreer, 2021-03-05:
I've seen some back and forth here, is there documentation on the exact process for this now or is that still TBD? I don't mind piecing the process together here based on this thread, but if someone's gone to the trouble to write it up concisely somewhere I'd rather use that.

benlye, 2021-03-05:
> I've seen some back and forth here, is there documentation on the exact process for this now or is that still TBD? I don't mind piecing the process together here based on this thread, but if someone's gone to the trouble to write it up concisely somewhere I'd rather use that.

I believe that the process is:
* Install a Marlin FW with `BINARY_FILE_TRANSFER` enabled
* Configure the plugin to use the marlinbft method
   * Make sure you install the pre-requisite Python package
* Enable 'Don't wait for reset' in the Advanced Flash Method Settings

If someone who has this board can verify that, and/or fill in any gaps, I'd be more than happy to add it to the docs.

Sennevds, 2021-03-05:
> > I've seen some back and forth here, is there documentation on the exact process for this now or is that still TBD? I don't mind piecing the process together here based on this thread, but if someone's gone to the trouble to write it up concisely somewhere I'd rather use that.
> 
> I believe that the process is:
> 
> * Install a Marlin FW with `BINARY_FILE_TRANSFER` enabled
> * Configure the plugin to use the marlinbft method
>   
>   * Make sure you install the pre-requisite Python package
> * Enable 'Don't wait for reset' in the Advanced Flash Method Settings
> 
> When you flash firmware using the plugin you will see the 'flash succeeded' message before the printer has actually restarted, but seems to be nothing we can do about that.
> 
> If someone who has this board can verify that, and/or fill in any gaps, I'd be more than happy to add it to the docs.

Looks okay! The only problem you got is with the filenames like I posted here. I'm trying to make the filename random but for some reason I can only push firmware.bin as filename when I try to change the filename to something like filenamea.bin I get following error: `Cannot open file on client`

jimcarreer, 2021-03-05:
Oh, I'm a bit new to this for:
> Install a Marlin FW with BINARY_FILE_TRANSFER enabled
Does this mean I need to install firmware, different from the stock firmware with my Ender V2, before I can use this plugin?

LekeFly, 2021-03-05:
Can confirm that the filename needs to be unique on the Ender 3 V2.

Sennevds, 2021-03-05:
> Oh, I'm a bit new to this, for:
> 
> > Install a Marlin FW with BINARY_FILE_TRANSFER enabled
> 
> Does this mean I need to install firmware, different from the stock firmware with my Ender V2, before I can use this plugin?

Yes you need custom firmware. If you have never done this it's best you check some videos on how to do this.
If you want a firmware version with a lot of extra features for the v2 checkout https://github.com/Jyers/Marlin/

benlye, 2021-03-05:
> Can confirm that the filename needs to be unique on the Ender 3 V2.

Can you explain what that means?  

The plugin will always use firmware.bin as the destination file name no matter what you provide as input.  Do I need to have an option to change/randomize the destination name?

Sennevds, 2021-03-05:
> > Can confirm that the filename needs to be unique on the Ender 3 V2.
> 
> Can you explain what that means?
> 
> The plugin will always use firmware.bin as the destination file name no matter what you provide as input. Do I need to have an option to change/randomize the destination name? If so, what works?

Yeah for the v2 the firmware file needs to be different every time. It won't flash the same name twice. I was experimenting with the binary file protocol but can't get it to upload a different filename. If I use a different name then firmware.bin I get the error `Cannot open file on client`

t3chguy, 2021-03-05:
>  If so, what works?

As long as the name of the new firmware isn't the same as that which was used to install the current one it'll work.
So even just flipping between firmware1.bin and firmware2.bin should work, but using the timestamp would be sane

benlye, 2021-03-05:
> Yeah for the v2 the firmware file needs to be different every time. It won't flash the same name twice. I was experimenting with the binary file protocol but can't get it to upload a different [...]

What is generating that error?  Where are you seeing it?  Are you using the marlinbft plugin?

> As long as the name of the new firmware isn't the same as that which was used to install the current one it'll work.
So even just flipping between firmware1.bin and firmware2.bin should work, but using the timestamp would be sane

OK, I can easily add an option in the plugin to use a timestamp as the destination filename - something like `fwHHMMSS.bin`.

The current development branch has the new printer profiles feature so this will be built on top of that.

Sennevds, 2021-03-05:
> What is generating that error? Where are you seeing it? Are you using the marlinbft plugin?
> 
> >

I'm using the binproto2 library directly to test out if it would work just to test it out if it would work so this is output of that library and not your plugin. I know your plugin always use firmware.bin as filename

benlye, 2021-03-05:
You can try v1.10.2 from the devel branch - it has an option to use timestamp-based filenames.

[image]

Sennevds, 2021-03-05:
Did it work with binproto2 to use a different filename? Just out of curiosity

benlye, 2021-03-05:
> Did it work with binproto2 to use a different filename? Just out of curiosity

It does for me - I verified that a timestamp-named file was transferred to my board.

```
2021-03-05 15:51:04,389 - octoprint.plugins.firmwareupdater - INFO - Connecting to printer at '/dev/ttyACM0' using Marlin BFT protocol
2021-03-05 15:51:04,391 - octoprint.plugins.firmwareupdater - INFO - Connecting: Switching Marlin to binary protocol...
2021-03-05 15:51:04,451 - octoprint.plugins.firmwareupdater - INFO - Connection synced [0], binary protocol version 0.1.0, 96 byte payload buffer
2021-03-05 15:51:04,452 - octoprint.plugins.firmwareupdater - INFO - Transfering file to printer using Marlin BFT '/tmp/tmpm2ojmil_' -> /fw155104.bin
2021-03-05 15:51:09,709 - octoprint.plugins.firmwareupdater - INFO - Binary file transfer complete
```

My board isn't an Ender 3 V2 so it ignored the new file, but it was definitely transferred.

This is where I'm doing the copy.  You can see the lines above which set the name.
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/devel/octoprint_firmwareupdater/methods/marlinbft.py#L87

Sennevds, 2021-03-05:
Hmm maybe it's something with filename length. I did almost the same as you (without the ui changes) but it always failed. I can't test right now because I'm printing at the moment I will try tonight or tomorrow.

benlye, 2021-03-07:
Did anyone try the development branch version with the timestamp filename yet?

Systropia, 2021-03-07:
I tried it. Timestamping works, the firmware file is uploaded to the board SD-card, and the firmware updates. However, the firmware was not updated the second time I tried. The problem (or my problem) seems to be that since the current firmware file is left on the SD-card, the board will not update the firmware as long as the current firmware file is present on the SD-card. When I manually deleted the current firmware file from the SD-card, I was able to update the firmware through OctoPrint again. I even had other “non-current” firmware files on the SD-card, but the firmware was updated to the one I updated through OctoPrint.

I tested this also with the manual/intended flashing method with three files on the SD-card: current and two new ones. The firmware was not updated as long as the current firmware was on the SD-card. However, once the current firmware was deleted, the firmware was updated to the one on top of the directory list.

So after the update or before next update the current firmware file should be deleted from the SD-card.

benlye, 2021-03-07:
OK I'll have to see if I can delete files using BFT. If I can, implementing this should be quite easy.

benlye, 2021-03-07:
Please give v1.10.3 a try.

Marlin BFT does not have the ability to delete files, so I'm using an `M30` command to delete the last file from the SD card.  This is only attempted if 'Use timestamp filenames' is enabled.

I ran into an edge case on my setup where doing this crashes the BFT protocol and the printer needs to be reset.  This only happens if the board's SD card is also mounted on the host at the time the BFT upload is attempted.  

I have my SD card mounted using usbmount so that I can also use/test the LPC176x flash method.  I was able to get around this by adding `sudo umount /media/usb` as a pre-flash system command.  I don't expect that most people will be trying to use usbmount AND binary file transfer, but if they are the workaround is very simple.

benlye, 2021-03-08:
1.10.4 is the latest version in the development branch now.  It adds the timestamp filename option to the LPC176x method for anyone who wants to use that with an Ender 3 V2.

Sennevds, 2021-03-08:
I've just tested the Marlin BFT and that works! (and it was a problem with the amount of characters in the name)

benlye, 2021-03-08:
> 1.10.4 is the latest version in the development branch now. It adds the timestamp filename option to the LPC176x method for anyone who wants to use that with an Ender 3 V2.

1.10.4 was missing a commit due to me not pressing the 'push' button.  It is replaced with 1.10.5, which has the option for lpc176x.

benlye, 2021-03-08:
> I've just tested the Marlin BFT and that works! (and it was a problem with the amount of characters in the name)

But you're going to use the Firmware Updater plugin anyway, right ;-)

Sennevds, 2021-03-08:
Of course just interested why it didn't worked ;)

Systropia, 2021-03-10:
The new version seems to work well now.

benlye, 2021-03-17:
v1.10.9 released with support for Ender 3 V2 using either Marlin BFT or lpc1769 flash methods.

closed 2021-03-17 by benlye