OctoPrint/OctoPrint-FirmwareUpdater #194: Debian buester SKRV1.3  cannot install usbmount

dgpower, 2021-02-04:
Debian buester SKRV1.3 不能安装usbmount
我用的是全志的电视盒来安装Octoprint，在安装usbmount组件的时候，报错：无法定位软件包 usbmount
-----------------------------------------------------------------------------------------------------------
Debian buester SKRV1.3  cannot install usbmount

I used the TV box of Allwinner to install Octoprint. When installing the usbmount component, an error was reported: Unable to locate the software package usbmount

benlye, 2021-02-04:
You should open an issue on the usbmount Github repo or consult documentation for your Debian distribution.

closed 2021-02-04 by benlye

dgpower, 2021-02-06:
电视盒子不能使用固件升级插件的问题已经解决。
1、电视盒子自带了存储，所以linux系统可以不需要安装usbmount组件。
2、只需要找一个合适的目录，保存上传的固件文件，我选择的是/home/pi
3、相应地，在高级设置里面的 unmount 指令 也不需要设置
-----------------------------------------------------------------------------------------------
The problem that the TV box cannot use this plug-in has been resolved.
1. The TV box comes with storage, so there is no need to install usbmount components.
2. Just find a suitable directory and save the uploaded firmware file, I choose /home/pi
3. Correspondingly, the unmount command in the Advanced FlashMethod Settings does not need to be set

benlye, 2021-02-06:
And how are you transferring the firmware file from /home/pi to the printer??

Give me a few more days and you'll be able to use Marlin's Binary File Transfer protocol - #161.