OctoPrint/OctoPrint-FirmwareUpdater #196: Consider removing "accept" attribute from file input

amdprophet, 2021-02-08:
**Hardware Setup**
Prusa i3 MK3S

**Describe the problem**
Mobile phones (iOS & Android) do not [fully support](https://caniuse.com/input-file-accept) the `accept` attribute on file inputs. As a result, firmwares cannot be uploaded from mobile phones. 

**Log Files**
N/A

**Additional context**
I've tried filtering files based on the `text/plain` mime type, instead of checking file extensions, but it didn't seem to allow the Prusa fw to be selected

benlye, 2021-02-08:
I'll make it a configurable option.

benlye, 2021-02-08:
Please try [v1.8.1b1](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/releases/tag/1.8.1b1) from the Development branch (switch the branch for this plugin in the OctoPrint Software Update plugin) and see if the new option solves your problem.

[image]

amdprophet, 2021-02-09:
Unfortunately I'm running into an error with one of the new dependencies:

```
Building wheels for collected packages: OctoPrint-FirmwareUpdater, heatshrink
  Building wheel for OctoPrint-FirmwareUpdater (setup.py) ... done
  Created wheel for OctoPrint-FirmwareUpdater: filename=OctoPrint_FirmwareUpdater-1.8.1b1-py3-none-any.whl size=29374 sha256=e69aa38e4f3e41e1de7a26b25a04c11bd48b17105a2229a7abceac39de6f8610
  Stored in directory: /tmp/pip-ephem-wheel-cache-8uq1hdto/wheels/db/48/f9/d402162b3603aa5cc2fc9cd4d9888d1aec57980529beb739a8
  Building wheel for heatshrink (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/oprint/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-vqtv_0av/heatshrink_d6ba784e0f344cc3a49deaebdbade2eb/setup.py'"'"'; __file__='"'"'/tmp/pip-install-vqtv_0av/heatshrink_d6ba784e0f344cc3a49deaebdbade2eb/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-yyribdbe
       cwd: /tmp/pip-install-vqtv_0av/heatshrink_d6ba784e0f344cc3a49deaebdbade2eb/
  Complete output (138 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-aarch64-3.8
  creating build/lib.linux-aarch64-3.8/heatshrink
  copying heatshrink/streams.py -> build/lib.linux-aarch64-3.8/heatshrink
  copying heatshrink/__init__.py -> build/lib.linux-aarch64-3.8/heatshrink
  running build_ext
  building 'heatshrink.core' extension
  creating build/temp.linux-aarch64-3.8
  creating build/temp.linux-aarch64-3.8/heatshrink
  creating build/temp.linux-aarch64-3.8/heatshrink/_heatshrink
  aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I. -I./heatshrink/_heatshrink -I/home/pi/oprint/include -I/usr/include/python3.8 -c heatshrink/core.c -o build/temp.linux-aarch64-3.8/heatshrink/core.o -std=c99
  heatshrink/core.c: In function ‘__pyx_pf_10heatshrink_4core_7Encoder_7fill’:
  heatshrink/core.c:4508:38: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘Py_ssize_t’ {aka ‘long int’} [-Wsign-compare]
   4508 |     __pyx_t_5 = ((__pyx_v_total_sunk < __pyx_t_9) != 0);
        |                                      ^
  heatshrink/core.c: In function ‘PyInit_core’:
  heatshrink/core.c:7279:3: warning: ‘tp_print’ is deprecated [-Wdeprecated-declarations]
   7279 |   __pyx_type_10heatshrink_4core_Writer.tp_print = 0;
        |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  In file included from /usr/include/python3.8/object.h:746,
                   from /usr/include/python3.8/pytime.h:6,
                   from /usr/include/python3.8/Python.h:85,
                   from heatshrink/core.c:24:
  /usr/include/python3.8/cpython/object.h:260:30: note: declared here
    260 |     Py_DEPRECATED(3.8) int (*tp_print)(PyObject *, FILE *, int);
        |                              ^~~~~~~~
  heatshrink/core.c:7283:3: warning: ‘tp_print’ is deprecated [-Wdeprecated-declarations]
   7283 |   __pyx_type_10heatshrink_4core_Reader.tp_print = 0;
        |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  In file included from /usr/include/python3.8/object.h:746,
                   from /usr/include/python3.8/pytime.h:6,
                   from /usr/include/python3.8/Python.h:85,
                   from heatshrink/core.c:24:
  /usr/include/python3.8/cpython/object.h:260:30: note: declared here
    260 |     Py_DEPRECATED(3.8) int (*tp_print)(PyObject *, FILE *, int);
        |                              ^~~~~~~~
  heatshrink/core.c:7287:3: warning: ‘tp_print’ is deprecated [-Wdeprecated-declarations]
   7287 |   __pyx_type_10heatshrink_4core___pyx_scope_struct___drain.tp_print = 0;
        |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  In file included from /usr/include/python3.8/object.h:746,
                   from /usr/include/python3.8/pytime.h:6,
                   from /usr/include/python3.8/Python.h:85,
                   from heatshrink/core.c:24:
  /usr/include/python3.8/cpython/object.h:260:30: note: declared here
    260 |     Py_DEPRECATED(3.8) int (*tp_print)(PyObject *, FILE *, int);
        |                              ^~~~~~~~
  heatshrink/core.c: In function ‘__Pyx__ExceptionSave’:
  heatshrink/core.c:8219:21: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_type’; did you mean ‘curexc_type’?
   8219 |     *type = tstate->exc_type;
        |                     ^~~~~~~~
        |                     curexc_type
  heatshrink/core.c:8220:22: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_value’; did you mean ‘curexc_value’?
   8220 |     *value = tstate->exc_value;
        |                      ^~~~~~~~~
        |                      curexc_value
[... 58 lines cut ...]
        |                        curexc_type
  heatshrink/core.c:11166:25: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_value’; did you mean ‘curexc_value’?
  11166 |     tmp_value = tstate->exc_value;
        |                         ^~~~~~~~~
        |                         curexc_value
[... cut ...]
  11169 |     tstate->exc_value = *value;
        |             ^~~~~~~~~
        |             curexc_value
  heatshrink/core.c:11170:13: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
  11170 |     tstate->exc_traceback = *tb;
        |             ^~~~~~~~~~~~~
        |             curexc_traceback
  error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for heatshrink
```

benlye, 2021-02-09:
What are you running OctoPrint on?  What version of Python are you using?

benlye, 2021-02-09:
Looks like I'll probably have to pull Marlin binary file transfer support out, at least in its current form - the library it uses relies on heatshrink, which has weird python2 and python3 dependencies.

It's the same issue as https://github.com/CharlesWillis3/OctoPrint-MarlinBft/issues/8

I'll need a few days to unwind the BFT changes.

amdprophet, 2021-02-09:
I'm using the OctoPi nightly for arm64 with Python 3.8.6. It seems like heatshrink installation will fail on systems without Python 2: https://github.com/CharlesWillis3/OctoPrint-MarlinBft/issues/8.

benlye, 2021-02-09:
I think what I'll have to do is remove the dependency on the marlin-binary-protocol package and let people who want to use it install that package and its dependencies manually.

benlye, 2021-02-10:
Please try 1.8.1b2 - I've removed the installer dependency on marlin-binary-protocol (users who want to use BFT will have to install it themselves).
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/releases/tag/1.8.1b2

benlye, 2021-02-17:
Implemented in 1.8.1, due to be released shortly.

closed 2021-02-17 by benlye