OctoPrint/OctoPrint-FirmwareUpdater #212: Help by using this Plugin (lpc1768)

BlackMelt, 2021-03-02:
Hi there

I am trying to flash an SKR V1.3 with the latest Marlin 2.0.x Bugfix. 
SD-Card is Verified as Writeable under Settings. 
I select the Marlin firmware.bin as Flash Medium, but I can't see any differences in the Firmware after "Flashing"
I also tried the m997 Command witch doesn't change anything. 

Thanks in advance :D

benlye, 2021-03-02:
Attach your log files.

BlackMelt, 2021-03-02:
https://pastebin.com/442pMUYF

Here i have "Reset before Flashing" under the Advanced Flash Method Setting enabled, aswell as "sudo umount {mountpoint}"

benlye, 2021-03-02:
Not terminal logs... the octoprint.log file.

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#troubleshooting

BlackMelt, 2021-03-02:
Sorry,
here is the log file from Octoprint

https://pastebin.com/TTs7GTPP

benlye, 2021-03-02:
There are a lot of errors in that log - you have a lot of broken plugins.

For the firmware updater... it seems like your board is not really mounted.  Did you follow all the instructions to install and configure usbmount?  At any point did you change the permissions on /media/usb?

This line is very suspicious because that should take _some_ time.
`2021-03-02 19:51:22,189 - octoprint.plugins.firmwareupdater - INFO - Firmware update folder '/media/usb' available for writing after 0.0 seconds`

The fact that the plugin thought that the board was mounted after 0.0 seconds suggests that the permissions on /media/usb have been modified.

Most likely the plugin is just copying the firmware to an empty folder on the host (because the board isn't really mounted) and then resetting the board.  Because the firmware was never really copied to the board, it never gets updated.

BlackMelt, 2021-03-02:
I used the Manuel mostly completly as it is for the LPC1768.
But as i run in to Problems i looked arround the Internet an found "sudo chmod 777 /media/usb0", after this i was able to pass the SD Path verification with no errors.

So that's my usbmount permission line:

FS_MOUNTOPTIONS="-fstype=vfat,gid=users,dmask=0007,fmask=0111"

And the systemd-udevd is

[Service]
PrivateMounts=no
MountFlags=shared

BlackMelt, 2021-03-02:
But now i will make a Backup from my current Octoprint install, and will try to reconfig everything from scratch, let's see.

I will report on Progress, if i have no succsess, otherwise, i will close the issue if i can.

benlye, 2021-03-02:
> I used the Manuel mostly completly as it is for the LPC1768.
> But as i run in to Problems i looked arround the Internet an found "sudo chmod 777 /media/usb0", after this i was able to pass the SD Path verification with no errors.

The instructions specifically tell you NOT to do that.  It's even in bold.
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/lpc176x.md#file-system-permissions

BlackMelt, 2021-03-02:
Sorry for being so stupid. 
Works now perfectly fine.

closed 2021-03-02 by BlackMelt