OctoPrint/OctoPrint-FirmwareUpdater #216: Prerequisites met (per setup instructions) but not showing as installed

tpsherrill, 2021-03-06:
**Hardware Setup**
EZBoard Lite on Ender 3 Pro

**Describe the problem**
Wanting to flash an updated Marlin image. Installed FirmwareUpdater to do so. SSH'd to the Pi to add the packages needed using pip3. Reboot and plugin still shows the prerequisites aren't being met.

**Log Files**
[octoprint (1).log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6094490/octoprint.1.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6094491/plugin_firmwareupdater_console.log)

**Additional context**
I solved my problem by getting off my butt and putting firmware.bin on the SD card and I'm OK for now. Thanks for your work on the plugin.

benlye, 2021-03-06:
The logs don't show anything useful - it looks like it was rolled over.
* What environment are you running OctoPrint in/on?  Is it OctoPi?  Did you run the pip command in the appropriate Python venv?
* Did you restart OctoPrint after installing the Python package?
* Did you reload the web interface?
* Was the printer reconnected after flashing FW with binary file transfer included?

If you've done all those things and they still don't show as installed, please restart OctoPrint and then grab the log.  After all the plugins are loaded you should see a line like this:

```
2021-03-05 15:54:47,284 - octoprint.plugins.firmwareupdater - INFO - Setting BINARY_FILE_TRANSFER capability to True
```
It would also be helpful to see the capabilities output as reported in the OctoPrint terminal when the printer is first connected, or by sending `M115`.

benlye, 2021-03-06:
I just did a fresh install of OctoPi and confirmed that following the instructions the Python package is installed and properly detected.

I'm assuming that you just ran pip3, and not the correct OctoPi venv pip commands, which would be:
```
~/oprint/bin/pip install https://github.com/p3p/pyheatshrink/releases/download/0.3.3/pyheatshrink-pip.zip
~/oprint/bin/pip install marlin-binary-protocol
```

I'm also guessing that you hadn't previously flashed a FW image which had  `#define BINARY_FILE_TRANSFER` uncommented.

DakotaHoosier, 2021-03-06:
Old FW did NOT have #define BINARY_FILE_TRANSFER uncommented.

I'm good with Python language spec but suck at things like virtual environments... 

I need to use pip even for Python3? I'll run those commands as-is and let you know (may be a few days). I had copied/pasted the commands with pip3 from the setup guide.

benlye, 2021-03-06:
Right, so you hadn't actually met either requirement.  I updated the documentation to try tp make it clearer.

On OctoPi it seems like pip/pip3 are the same.  I ran the commands exactly as I pasted above and it worked for me.

DakotaHoosier, 2021-03-06:
Thanks, I did change the pragma on the new firmware so I thought maybe you were scanning the uploaded file. Silly of me in retrospect...

Appreciate the coaching on the virtual envs... Thanks for the time on this plugin! My goal was to be able to change some non-printinf FW parameters and see what worked without having to walk from the room with my dev setup (good screens, comfortable seating, music) to where the printer is (not far, but two doors and not a great place to fiddle and research Marlin!)

closed 2021-03-10 by benlye