OctoPrint/OctoPrint-FirmwareUpdater #219: Marlin BINARY_FILE_TRANSFER works only once than it is not detected

Sekisback, 2021-03-14:
**Hardware Setup**
SKR Mini E3 v2 on an   Ender 3 
Raspberry Pi 3 Model B Rev 1.2
octopi_version: 0.17.0
octoprint.version: 1.5.3
printer.firmware: Marlin 2.0.7.2
OctoPrint-FirmwareUpdater-1.10.7

**Describe the problem**
I use the BINARY_FILE_TRANSFER and it works as expected but only once.
after flashing a new firmware and before I try to flash another firmware version a message pop up that the BINARY_FILE_TRANSFER can not be detected ore something like that. According to the Configuration the Marlin BINARY_FILE_TRANSFER firmware feature is NOT detected.

I have to reboot the Raspberry to get this working again

[image]

**Log Files**
2021-03-13 11:44:08,177 Flashing successful.
2021-03-14 07:07:12,518 Flashing successful.

Send: M115
Recv: FIRMWARE_NAME:Marlin 2.0.7.2 (Mar 13 2021 11:42:43)
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:1

benlye, 2021-03-14:
Please attach the log files.

Sekisback, 2021-03-14:
Hey Ben Lye, i attached the log file. Ill check if disconnect/reconnect the printer after the first flash will work

benlye, 2021-03-14:
This is the relevant log content:
```
  128 2021-03-14 06:55:11,793 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
  129 2021-03-14 06:55:11,796 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
  153 2021-03-14 06:55:11,984 - octoprint.plugins.firmwareupdater - INFO - Setting BINARY_FILE_TRANSFER capability to True
 1127 2021-03-14 07:00:33,816 - octoprint.plugins.firmwareupdater - INFO - Python binproto2 package installed: True
 2451 2021-03-14 07:02:01,264 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
 2452 2021-03-14 07:02:01,268 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
 2458 2021-03-14 07:02:01,293 - octoprint.plugins.firmwareupdater - INFO - Setting BINARY_FILE_TRANSFER capability to True
 2642 2021-03-14 07:02:34,802 - octoprint.plugins.firmwareupdater - INFO - Setting BINARY_FILE_TRANSFER capability to True
 2998 2021-03-14 07:04:38,140 - octoprint.plugins.firmwareupdater - INFO - Python binproto2 package installed: True
 3861 2021-03-14 07:05:31,296 - octoprint.plugins.firmwareupdater - INFO - Setting BINARY_FILE_TRANSFER capability to True
 3873 2021-03-14 07:05:31,362 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
 3874 2021-03-14 07:05:31,373 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
 4181 2021-03-14 07:06:48,250 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile index: 0
 4182 2021-03-14 07:06:48,254 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile name: Default
 4183 2021-03-14 07:06:48,255 - octoprint.plugins.firmwareupdater - INFO - Printer port: /dev/ttyACM0
 4184 2021-03-14 07:06:48,269 - octoprint.plugins.firmwareupdater - INFO - Flash method: marlinbft
 4185 2021-03-14 07:06:48,272 - octoprint.plugins.firmwareupdater - INFO - Python package 'marlin-binary-protocol' is installed: True
 4186 2021-03-14 07:06:48,273 - octoprint.plugins.firmwareupdater - INFO - Marlin BINARY_FILE_TRANSFER capability is enabled: True
 4187 2021-03-14 07:06:48,288 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
 4188 2021-03-14 07:06:48,289 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
 4213 2021-03-14 07:06:48,422 - octoprint.plugins.firmwareupdater - INFO - Current Baud: 115200
 4214 2021-03-14 07:06:48,425 - octoprint.plugins.firmwareupdater - INFO - Initializing Marlin BFT protocol
 4221 2021-03-14 07:06:48,431 - octoprint.plugins.firmwareupdater - INFO - pySerial Version: 3.5
 4230 2021-03-14 07:06:48,469 - octoprint.plugins.firmwareupdater - INFO - Connecting to printer at '/dev/ttyACM0' using Marlin BFT protocol
 4233 2021-03-14 07:06:48,473 - octoprint.plugins.firmwareupdater - INFO - Connecting: Switching Marlin to binary protocol...
 4258 2021-03-14 07:06:48,534 - octoprint.plugins.firmwareupdater - INFO - Connection synced [0], binary protocol version 0.1.0, 96 byte payload buffer
 4259 2021-03-14 07:06:48,544 - octoprint.plugins.firmwareupdater - INFO - Transfering file to printer using Marlin BFT '/tmp/tmp45axmmeu' -> /firmware.bin
 8652 2021-03-14 07:06:58,256 - octoprint.plugins.firmwareupdater - INFO - Binary file transfer complete
 8658 2021-03-14 07:06:59,274 - octoprint.plugins.firmwareupdater - INFO - Firmware update reset: attempting to reset the board
 8659 2021-03-14 07:06:59,276 - octoprint.plugins.firmwareupdater - INFO - Resetting printer at '/dev/ttyACM0'
 8660 2021-03-14 07:06:59,345 - octoprint.plugins.firmwareupdater - INFO - Waiting for printer at '/dev/ttyACM0' to reset
 8664 2021-03-14 07:07:00,039 - octoprint.plugins.firmwareupdater - INFO - Printer at '/dev/ttyACM0' is resetting
 8710 2021-03-14 07:07:12,514 - octoprint.plugins.firmwareupdater - INFO - Printer at '/dev/ttyACM0' reset in 13.17 seconds
 8711 2021-03-14 07:07:12,518 - octoprint.plugins.firmwareupdater - INFO - Flashing successful.
 8712 2021-03-14 07:07:12,530 - octoprint.plugins.firmwareupdater - INFO - No postflash gcode or postflash is disabled, setting run_postflash_gcode to false
 8713 2021-03-14 07:07:12,538 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyACM0, baudrate=115200, profile={'axes': {'e': {'inverted': False, 'speed': 300}, 'x': {'inverted': False, 'speed': 6000}, 'y': {'inverted': False, 'speed': 6000}, 'z': {'inverted': False, 'speed': 200}}, 'color': 'default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}, 'heatedBed': True, 'heatedChamber': False, 'id': '_default', 'model': 'Ender3', 'name': 'Ender3', 'volume': {'custom_box': False, 'depth': 235.0, 'formFactor': 'rectangular', 'height': 250.0, 'origin': 'lowerleft', 'width': 235.0}}
 8864 2021-03-14 07:07:20,033 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
 8865 2021-03-14 07:07:20,036 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
 8874 2021-03-14 07:07:20,260 - octoprint.plugins.firmwareupdater - INFO - Setting BINARY_FILE_TRANSFER capability to True
```

The final line says that the BFT capability is detected, and at every step the feature is also detected as enabled.  When the feature is detected this is stored in the `config.yaml` config file and a notification is sent to the front-end.  It seems like somehow that notification is not getting through.

What happens if you reload the OctoPrint UI with Ctrl-F5?

Sekisback, 2021-03-14:
After reloading the UI with Ctrl-F5 it show still the same "no detection" the config.yaml looks like this
[image]

benlye, 2021-03-14:
I can't reproduce that behavior here at all.  When the config file has the capability set to true, the front end reads that and sets the status accordingly when the UI is loaded.

What browser are you using?  Does your browser have a Javascript console and does it show any errors?

In Chrome or Edge right-click the page and choose 'Inspect' then at the top of the side-panel click 'Console' to see the Javascript console window.  There should be no errors.

Can you try another browser?

benlye, 2021-03-14:
Just saw your last update with 'Remember URL' - looking at that now.

benlye, 2021-03-14:
Yep, you found a bug - will have a new version out shortly.

Sekisback, 2021-03-14:
I am glad that you found something

benlye, 2021-03-14:
1.10.8 released in the RC branch - it should fix this.  I had missed a couple of changes needed for 'Save last URL' when I implemented the printer profiles feature.

Last URL will now be correctly saved and read per profile (there is no functional change if you don't enable/use profiles).

Sekisback, 2021-03-14:
awesome, ill check that later the day

benlye, 2021-03-14:
The change is in the Javascript code, so please make sure you press Ctrl-F5 to reload everything after you install the update and the web UI has reloaded.

Sekisback, 2021-03-14:
tested, work like expected. Thx for fast solution

PR #220 (2021-03-14): Fix last URL issue
Fixes #219
closed 2021-03-14 by Sekisback