OctoPrint/OctoPrint-FirmwareUpdater #224: New version of Firmware Updater doesn't show up correctly and produces errors in the JavaScript console.

ManuelMcLure, 2021-03-18:
**Hardware Setup**
Custom printer with Re-ARM/RAMPS motherboard.

**Describe the problem**
Version 1.10.9 does not display properly, buttons do not work. Javascript console shows the following:
```
main.js:850 Could not bind view model FirmwareUpdaterViewModel to target #settings_plugin_firmwareupdater : ReferenceError: Unable to process binding "visible: function(){return configProfilesVisible }"
Message: configProfilesVisible is not defined
    at visible (eval at parseBindingsString (http://oprint.internal.mclure.org/static/js/lib/knockout.js:74:172), <anonymous>:3:60)
    at update (http://oprint.internal.mclure.org/static/js/lib/knockout.js:117:187)
    at a.$.l (http://oprint.internal.mclure.org/static/js/lib/knockout.js:79:379)
    at Function.yd (http://oprint.internal.mclure.org/static/js/lib/knockout.js:55:359)
    at Function.zd (http://oprint.internal.mclure.org/static/js/lib/knockout.js:55:45)
    at Function.ha (http://oprint.internal.mclure.org/static/js/lib/knockout.js:54:384)
    at Object.a.o.a.$ (http://oprint.internal.mclure.org/static/js/lib/knockout.js:52:147)
    at http://oprint.internal.mclure.org/static/js/lib/knockout.js:79:366
    at Object.D (http://oprint.internal.mclure.org/static/js/lib/knockout.js:11:52)
    at p (http://oprint.internal.mclure.org/static/js/lib/knockout.js:78:408)
    at k (http://oprint.internal.mclure.org/static/js/lib/knockout.js:76:170)
    at m (http://oprint.internal.mclure.org/static/js/lib/knockout.js:76:52)
    at k (http://oprint.internal.mclure.org/static/js/lib/knockout.js:76:228)
    at m (http://oprint.internal.mclure.org/static/js/lib/knockout.js:76:52)
    at k (http://oprint.internal.mclure.org/static/js/lib/knockout.js:76:228)
    at m (http://oprint.internal.mclure.org/static/js/lib/knockout.js:76:52)
    at k (http://oprint.internal.mclure.org/static/js/lib/knockout.js:76:228)
    at Object.a.vc (http://oprint.internal.mclure.org/static/js/lib/knockout.js:85:159)
    at http://oprint.internal.mclure.org/static/js/app/main.js:830:36
    at Pn (http://oprint.internal.mclure.org/static/js/lib/lodash.min.js:9:530)
    at Function.<anonymous> (http://oprint.internal.mclure.org/static/js/lib/lodash.min.js:30:66)
    at http://oprint.internal.mclure.org/static/js/app/main.js:775:27
    at Pn (http://oprint.internal.mclure.org/static/js/lib/lodash.min.js:9:530)
    at Function.<anonymous> (http://oprint.internal.mclure.org/static/js/lib/lodash.min.js:30:66)
    at bindViewModels (http://oprint.internal.mclure.org/static/js/app/main.js:716:15)
    at sentryWrapped (http://oprint.internal.mclure.org/plugin/errortracking/static/js/sentry.min.js:2:45619)
```
```
helpers.js:1455 Error calling onAllBound on view model FirmwareUpdaterViewModel : TypeError: self.settingsViewModel.settings.plugins.firmwareupdater.marlinbft_hascapability is not a function
    at FirmwareUpdaterViewModel.self.onAllBound (http://oprint.internal.mclure.org/plugin/firmwareupdater/static/js/firmwareupdater.js:168:97)
    at callViewModelIf (http://oprint.internal.mclure.org/static/js/app/helpers.js:1439:35)
    at http://oprint.internal.mclure.org/static/js/app/helpers.js:1366:13
    at Pn (http://oprint.internal.mclure.org/static/js/lib/lodash.min.js:9:530)
    at Function.<anonymous> (http://oprint.internal.mclure.org/static/js/lib/lodash.min.js:30:66)
    at callViewModelsIf (http://oprint.internal.mclure.org/static/js/app/helpers.js:1364:7)
    at callViewModels (http://oprint.internal.mclure.org/static/js/app/helpers.js:1358:5)
    at bindViewModels (http://oprint.internal.mclure.org/static/js/app/main.js:885:13)
    at sentryWrapped (http://oprint.internal.mclure.org/plugin/errortracking/static/js/sentry.min.js:2:45619)
```

**Log Files**
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6161117/octoprint.log)
`plugin_firmwareupdater_console.log` is empty even though I enabled debug logging for Firmware Updater

**Additional context**
This is how the Firmware Updater settings show up. None of the buttons work.
[image]

ManuelMcLure, 2021-03-18:
Seems to have been some issue with my browser. It seems to be working now.

closed 2021-03-18 by ManuelMcLure

benlye, 2021-03-18:
Browser was getting an old JS file from cache rather than loading the new one so the JS elements that the UI needed weren't available. 

Ctrl-F5 would have reloaded everything and fixed it.