OctoPrint/OctoPrint-FirmwareUpdater #226: Since last update the new firmware seems to be flashed but isn't

FunnyGermanGuy, 2021-03-21:
So first my setup. I've got a BTT SKR 1.4 Turbo Board and a RPi 3b+ connected via USB running latest OctoPrint, FirmwareUpdater ist configured to upload via Serial Transfer Protocol. I've updated the FirmwareUpdater I think 2 day ago. Since then I often was wondering why some changes I made were not showing up in my Settings on the printer. Since the Updater always said that flashing went finde and I saw the board restarting I thought I forgot to save my changes in the Marlin Firmware. But weird things still happened. So I tried to change my Z-Probe-Offset in the Firmware in VSCode. Exported the new firmware and uploaded via Octoprint. Initialized the EEProm but the old Offset was still present. Plugged the micro SDCard from my Printerboard into my Computer Uploaded the same Firmware via VSCode, back to the board, initialized EEProm, new Z-Offset was showing. Did something change with the update which I have to change on my side too? Thanks for your great work

benlye, 2021-03-21:
The Marlin transfer protocol is working fine for me, I use it to update my SKR 1.4 Turbo nearly every day. 

Please try to update your firmware then collect and post the octoprint.log log file.

FunnyGermanGuy, 2021-03-23:
Hm I updated Marlin to the latest bugfix. To test things out I changed the z-Offset in the firmware to 0.5 from before 0.25. Reading the log it seems to be working fine but the changes do not appear on the printer. Tested to change one endstop flag from true to false, that didn't show on the printer either. 

```
2021-03-23 17:58:25,847 - octoprint.plugins.eeprom_marlin - INFO - Finished data collection, updating UI
2021-03-23 17:58:26,739 - octoprint.plugins.eeprom_marlin - INFO - M503 detected, collecting data
2021-03-23 17:58:26,764 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-23 17:58:26,765 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M149 C ; Units in Celsius

2021-03-23 17:58:26,817 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-23 17:58:26,817 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M906 X450 Y750 Z450

2021-03-23 17:58:26,818 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-23 17:58:26,818 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M906 I1 Z450

2021-03-23 17:58:26,825 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-23 17:58:26,825 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M906 T0 E800

2021-03-23 17:58:26,829 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-23 17:58:26,832 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M914 X80 Y120

2021-03-23 17:58:26,833 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-23 17:58:26,834 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M569 S1 X Y Z

2021-03-23 17:58:26,835 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-23 17:58:26,835 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M569 S1 I1 Z

2021-03-23 17:58:26,838 - octoprint.plugins.eeprom_marlin - INFO - Finished data collection, updating UI
2021-03-23 17:58:33,192 - octoprint.plugins.eeprom_marlin - INFO - M503 detected, collecting data
2021-03-23 17:58:33,208 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-23 17:58:33,209 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M149 C ; Units in Celsius

2021-03-23 17:58:33,258 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-23 17:58:33,258 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M906 X450 Y750 Z450

2021-03-23 17:58:33,259 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-23 17:58:33,260 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M906 I1 Z450

2021-03-23 17:58:33,261 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-23 17:58:33,261 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M906 T0 E800

2021-03-23 17:58:33,263 - octoprint.plugins.eeprom_marlin - WARNING - Did not recognise EEPROM data, skipping line
2021-03-23 17:58:33,263 - octoprint.plugins.eeprom_marlin - WARNING - Line: echo:  M914 X80 Y120
```

benlye, 2021-03-23:
That log isn't helpful at all.  I meant for you to show the section of the log where the firmware update is occurring - lines which contain `octoprint.plugins.firmwareupdater`.

It should look something like this:
```
2021-03-22 14:43:27,083 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile index: 0
2021-03-22 14:43:27,088 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile name: SKR 1.4 Turbo (BFT)
2021-03-22 14:43:27,089 - octoprint.plugins.firmwareupdater - INFO - Printer port: /dev/ttyACM0
2021-03-22 14:43:27,095 - octoprint.plugins.firmwareupdater - INFO - Flash method: marlinbft
2021-03-22 14:43:27,097 - octoprint.plugins.firmwareupdater - INFO - Python package 'marlin-binary-protocol' is installed: True
2021-03-22 14:43:27,099 - octoprint.plugins.firmwareupdater - INFO - Marlin BINARY_FILE_TRANSFER capability is enabled: True
2021-03-22 14:43:27,116 - octoprint.plugins.firmwareupdater - INFO - Sending pre-flash gcode commands: M117 Firmware Updating
2021-03-22 14:43:27,125 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2021-03-22 14:43:27,125 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2021-03-22 14:43:27,162 - octoprint.plugins.firmwareupdater - INFO - Current Baud: 115200
2021-03-22 14:43:27,166 - octoprint.plugins.firmwareupdater - INFO - Initializing Marlin BFT protocol
2021-03-22 14:43:27,168 - octoprint.plugins.firmwareupdater - INFO - pySerial Version: 3.4
2021-03-22 14:43:27,173 - octoprint.plugins.firmwareupdater - INFO - Connecting to printer at '/dev/ttyACM0' using Marlin BFT protocol
2021-03-22 14:43:27,174 - octoprint.plugins.firmwareupdater - INFO - Connecting: Switching Marlin to binary protocol...
2021-03-22 14:43:27,178 - octoprint.plugins.firmwareupdater - INFO - Connection synced [0], binary protocol version 0.1.0, 96 byte payload buffer
2021-03-22 14:43:27,178 - octoprint.plugins.firmwareupdater - INFO - Transfering file to printer using Marlin BFT '/tmp/tmpag1np6_7' -> /firmware.bin
2021-03-22 14:43:30,367 - octoprint.plugins.firmwareupdater - INFO - Binary file transfer complete
2021-03-22 14:43:31,388 - octoprint.plugins.firmwareupdater - INFO - Firmware update reset: attempting to reset the board
2021-03-22 14:43:31,391 - octoprint.plugins.firmwareupdater - INFO - Resetting printer at '/dev/ttyACM0'
2021-03-22 14:43:31,500 - octoprint.plugins.firmwareupdater - INFO - Waiting for printer at '/dev/ttyACM0' to reset
2021-03-22 14:43:32,375 - octoprint.plugins.firmwareupdater - INFO - Printer at '/dev/ttyACM0' is resetting
2021-03-22 14:43:39,690 - octoprint.plugins.firmwareupdater - INFO - Printer at '/dev/ttyACM0' reset in 8.19 seconds
2021-03-22 14:43:39,693 - octoprint.plugins.firmwareupdater - INFO - Flashing successful.

```

FunnyGermanGuy, 2021-03-23:
Yeah, sorry. I saw that I coppied my terminal output instead of the log file. And while you were already typing I battled scp in terminal before I realised I just could download the log-File over the web interface. See changed comment above for the log

benlye, 2021-03-23:
I don't know what to tell you - the log looks absolutely fine.

Have you run `M502` after loading the new firmware so that your setting changes are loaded actually loaded?  Settings that are configured in the FW but stored in the EEPROM won't be updated unless you run `M502` to load them and `M503` to save them back to the EEPROM.

Otherwise, can you check something else, like the compile date/time?

E.g., when I connect OctoPrint the terminal tells me the compilation date and time, so it's really easy to tell that I'm running the new firmware:
```
Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x (Mar 22 2021 14:35:22) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
```

Can you check that output?

But like I said, this works 100% on my SKR Turbo 1.4 so I really don't know what could be stopping it working for you...

FunnyGermanGuy, 2021-03-23:
Okay, I think I am getting closer to the problem. Thanks for the tip with the terminal output regarding compilation time. I just recompiled the firmware with the timestamp 19:30. After flashing via the updater and reconnecting I still get:

`Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x (Mar 23 2021 17:13:00) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff`

17:13 might be around the time when I flashed the sdcard manually. Can it be that the firmware updater is flashing one location and while rebooting the printer is fetching the firmware from another aka the microsd card?

FunnyGermanGuy, 2021-03-23:
Just for fun I unplugged the sd card from the display, not the micro sd card. And it shows a firmware.bin with the date 1.1.2000 at 1:00. I think in my case since the last update, the updater flashes to the sdcard which is only for storing gcode files

benlye, 2021-03-23:
I was going to ask if you had two SD cards.  The plugin has no control over which card gets written to, Marlin controls that.

I'm pretty sure that it won't change anything, but you can try going back to v1.9.0 of the plugin by doing a manual installation from this URL:
`https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/refs/tags/1.9.0.zip`

Between v1.9.0 and the current v1.10 release there were no changes to the BFT protocol in the plugin, so I can only think that something has changed in Marlin or in your firmware configuration.

FunnyGermanGuy, 2021-03-23:
It's hard to tell when this happened exactly because I updated the firmware at the same time I switched from a Raspberry Pi Zero W connected by GPIO Pins to a RPi 3B+ via USB.

benlye, 2021-03-23:
I'm guessing you also updated the firmware as well at some point while making those other changes?  You can still try the older version of the plugin to prove one way or another if the behaviour changed in the plugin.

Also, how do you have the `SDCARD_CONNECTION` configured in your firmware?  I don't have a setup with an SD card in the LCD so I have it set to `ONBOARD`:
```
  /**
   * Set this option to one of the following (or the board's defaults apply):
   *
   *           LCD - Use the SD drive in the external LCD controller.
   *       ONBOARD - Use the SD drive on the control board.
   *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
   *
   * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
   */
  #define SDCARD_CONNECTION ONBOARD
```

Presumably you have it set to LCD?  

Binary File Transfer will only work for firmware updates if it is set to `ONBOARD`.

FunnyGermanGuy, 2021-03-23:
I haven't set either of these options, so in my case the board's defaults apply. I'm guessing it has something to do with the switch form GPIO Connection to USB. If I find the time I will investigate this issue further, for now I will set it to onboard and use the microsd card

benlye, 2021-03-23:
I'm pretty sure if you set it to `ONBOARD` it will just start to work.  I think too many other things changed and I'm also pretty sure that v1.9.0 of the plugin will behave exactly the same.

FunnyGermanGuy, 2021-03-23:
Now you made me curious. I will dig for the rpi zero w and will look if it still has its microsd card with the old system and connect it again via gpio without changing my firmware settings

FunnyGermanGuy, 2021-03-23:
Okay, found the old pi. But same behaviour as before. Tried the 1.9 and 1.10. Must have gotten lucky before that the printer chose to take the right sdcard for the updates. Will now switch to onboard in marlin settings. Thanks for your help and your great work. I'll close the issue since it was a fault on my side

closed 2021-03-23 by FunnyGermanGuy