OctoPrint/OctoPrint-FirmwareUpdater #227: Pre-flash system commands are not run before flashing

zjsdlr2, 2021-03-21:
**Hardware Setup**
Raspberry Pi 3+
Monoprice MakerSelect v2 (Melzi, sanguino1284p)

**Describe the problem**
Pre-flash system commands are not run before flashing with the new update (1.10.9).
G-code commands appear to still be working as expected.
My configuration within the plugin menu:
[image]
[image]
The system command was working with the last plugin version and works when entered in SSH.

**Log Files**
I included the log files:
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6178497/octoprint.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6178498/plugin_firmwareupdater_console.log)
I am not seeing it logged that that it executed the Pre-flash commandline. The post-flash command line is logged.

**Additional Information**
Tried it with and without a semi-colon on the end with no success (shouldn't matter as it is only a single line)

benlye, 2021-03-22:
This was caused by a copy and paste error.  It's fixed in v1.10.10b1, which you can install from the Development branch.  I'll do a new release branch release later on.

If you don't want to switch to the development branch just enable post-flash gcode and put a benign command in the box - some thing like `M117 Firmware updated`.

zjsdlr2, 2021-03-22:
Adding a post flash gcode fixes the issue. Thanks!

benlye, 2021-03-27:
Fixed in v1.10.10.

closed 2021-03-27 by benlye