OctoPrint/OctoPrint-FirmwareUpdater #229: avrdude path test does not work.

ephramcohen, 2021-03-23:
**Hardware Setup**
Prusa MK3S, raspberry pi 3B+, Firmware Updater plugin version 1.10.9

**Describe the problem**
I entered /usr/bin/avrdude (no spaces, I checked) into the path to avrdude entry widget, but the test fails with "Path Doesn't exist". usr/bin/avrdude, //usr/bin/avrdude and \/usr\/bin\/avrdude all fail too.

**Log Files**
2021-03-23 18:47:56,781 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile index: 0
2021-03-23 18:47:56,790 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile name: Default
2021-03-23 18:47:56,795 - octoprint.plugins.firmwareupdater - INFO - Printer port: /dev/ttyACM0
2021-03-23 18:47:56,809 - octoprint.plugins.firmwareupdater - INFO - Flash method: avrdude
2021-03-23 18:47:56,829 - octoprint.plugins.firmwareupdater - ERROR - Path to avrdude does not exist: /usr/bin/avrdude

the updater log was empty.

**Additional context**
in putty, from ~, /usr/bin/avrdude gets the usage info
pi was rebooted, and octoprint interface reloaded.

benlye, 2021-03-23:
What do `which avrdude` and `ls -al /usr/bin/avrdude` return?

ephramcohen, 2021-03-23:
:~ $ which avrdude
/usr/bin/avrdude

:~ $ ls -al /usr/bin/avrdude
-rwxr-xr-x 1 root root 383544 Feb 20  2018 /usr/bin/avrdude

benlye, 2021-03-23:
I just tested the current pluing version with avrdude (I also have it installed in /usr/bin/avrdude) and had no problems flashing a mega2560, so this doesn't appear to be a bug in the plugin - it must be something unusual on your system.

For some reason Python thinks that the path does not exist on your system.  The check is [here](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/octoprint_firmwareupdater/methods/avrdude.py#L50) in the code.

You'll have to figure out why OctoPrint can't see it.  My first thought is permissions.

ephramcohen, 2021-03-23:
so much smh.  I had putty'd into the wrong pi.  It was my pi-hole if you were at all curious.

Everything works now.  Thanks for your rapid response.

closed 2021-03-23 by ephramcohen

benlye, 2021-03-23:
Glad you got it sorted, and thanks for the laugh ;-)