OctoPrint/OctoPrint-FirmwareUpdater #230: BUG: TypeError: replace() argument 2 must be str, not int

Hypertoken, 2021-03-25:
**Hardware Setup**
Anet A8 Plus Stock board

**Describe the problem**
If you set a specific baudrate in the plugin it'll throw an error. You have to use "default" as the setting. 

**Log Files**

```
octoprint.plugins.firmwareupdater - ERROR - Error while attempting to flash
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_firmwareupdater/__init__.py", line 261, in _flash_worker
    if flash_callable(self, firmware=firmware, printer_port=printer_port):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_firmwareupdater/methods/avrdude.py", line 92, in _flash_avrdude
    avrdude_command = avrdude_command.replace("{baudrate}", avrdude_baudrate)
TypeError: replace() argument 2 must be str, not int
```

**Additional context**
This happened on all 3 of my printers.  At some point this has changed, because it used to work. I can't remember what version I was on last time I tried to run a FW update.

benlye, 2021-03-25:
Fallout from the profiles implementation and related settings changes.  Fixed in v1.10.10b2 (in the Devel branch).   I plan to release 1.10.10 on the Stable branch tomorrow.

benlye, 2021-03-27:
Fixed in v1.10.10.

closed 2021-03-27 by benlye