OctoPrint/OctoPrint-FirmwareUpdater #231: [Feature Request] OpenBLT support

davidtgbe, 2021-03-25:
@benlye Congratulations for your work.
I would like you to consider adding support for OpenBLT bootloader. It is an OpenSource bootloader suitable for the use with many microcontrollers, and, therefore, for the use with many 3D printers. I really think it is a powefull and customizable BL with very much potential and future.
I'm currently using it on Anet ET4/ET5 printers. 

OpenBLT updates could be performed through USB or SD CARD.
1. USB fw updates could be performed through a command line app: https://www.feaser.com/openblt/doku.php?id=manual:bootcommander and then a restart.
2. SD CARD only need a firmware.srec file placed on the root of the SD CARD and then, a printer restart (very similar to LPC)

This is the OpenBLT fork I use: https://github.com/davidtgbe/openblt

Feel free to close this FR if you think it does not fit your goals

benlye, 2021-03-25:
This seems feasible, and probably not a lot of work. I'm thinking I could do two things:

1. Make the destination filename configurable for the lpc176x and marlinbft flash methods
    The destination filename is hard-coded to `firmware.bin` (unless you use the Ender 3 V2 option to make it timestamp-based).  Letting the user change it to `firmware.srec` would make either of those methods work immediately.
    This is not a lot of work to implement.

2. Add a new flash method for bootcommander
    This would be slightly more work and I'd need to experiment with the flashing tool.  
    My main thought right now is whether or not bootcommander resets the device at the end of the update (like avrdude does) or if the user has to initiate the reset.  This is easy on bards which support `M997`, but harder on Atmega-based boards.

Is there anything else I should know before I look into it?

davidtgbe, 2021-03-25:
> Make the destination filename configurable for the lpc176x and marlinbft flash methods
The destination filename is hard-coded to firmware.bin (unless you use the Ender 3 V2 option to make it timestamp-based). Letting the user change it to firmware.srec would make either of those methods work immediately.
This is not a lot of work to implement.

Yes, I think would be enough.

> Add a new flash method for bootcommander
This would be slightly more work and I'd need to experiment with the flashing tool.
My main thought right now is whether or not bootcommander resets the device at the end of the update (like avrdude does) or if the user has to initiate the reset. This is easy on bards which support M997, but harder on Atmega-based board

Flashing tool is pretty simple. Only mandatory arguments are destination serial port and firmware file. Maybe baudrate, but, I think it is defaulted to 115200 which is fine.

> Is there anything else I should know before I look into it?

- While flashing through SD, progress info is sent by the BL through serial port.
- If the update completes without errors, BL transfers control and starts the FW. No reset is needed.

thisiskeithb, 2021-03-28:
> This is easy on bards which support M997

When I was testing various refactoring on the ET4 board support PR in the main Marlin repo, `M997` wasn't working for me.

The printer would reset, but the bootloader wouldn't pick up the new `firmware.srec` unless I did a full power cycle/cold boot.

davidtgbe, 2021-03-28:
> > This is easy on bards which support M997
> 
> When I was testing various refactoring on the ET4 board support PR in the main Marlin repo, `M997` wasn't working for me.
> 
> The printer would reset, but the bootloader wouldn't pick up the new `firmware.srec` unless I did a full power cycle/cold boot.

Correct. I wasn't aware of that behaviour. It's probably the vector table offset or something related. I'm curious... does M997 work fine for other solutions based on using bootoader + Marlin?

thisiskeithb, 2021-03-28:
> does M997 work fine for other solutions based on using bootloader + Marlin?

It works well on most the major LPC & STM32F1/F4-based boards from BigTreeTech & Makerbase.

benlye, 2021-03-31:
You can try 1.11.0b1 from the development release branch.  This version has the option to set a custom firmware filename for the LPC176x flash method, meaning you can upload the firmware as `firmware.srec`.

I tried to add the same option to the binary file transfer method but it seems BFT doesn't like the longer filename and chokes on it.  I don't know if that's a problem with four character extension, or it just doesn't like names that are longer than 12 characters in total.  I left most of the code there but disabled the UI settings to enable it in the hope that the next version of BFT might support long names.

I'll look at adding bootcommander support next.

benlye, 2021-03-31:
@davidtgbe could you post the normal command line output from BootCommander when uploading firmware?  I'll have a go at parsing it and updating the progress bar accordingly.

benlye, 2021-04-01:
I figured out how to make OpenBLT work on one of the STM32 development boards I have.  I should have a test release with support for BootCommander out at some point in the next few days.

davidtgbe, 2021-04-01:
That's great @benlye!
I see you have managed to set up a testing environment. I will test the 1.11.0b1 branch and give you feedback as soon as I have a while.

benlye, 2021-04-02:
v1.11.0b2 now has support for BootCommander.  It works for my STM32F1 dev board so hopefully it works for you.

I noticed some strange behavior a couple of times where the board didn't run the main application at the end of the flash, and wouldn't until I unplugged it and plugged it back in, but I don't know if that's something I'm doing (I was doing a lot of messing with baud rates), or a quirk of my board.

davidtgbe, 2021-04-02:
Ok, I have installed v1.11.0b2.
```
/usr/local/bin/python -m pip --disable-pip-version-check install https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/refs/tags/1.11.0b2.zip --no-cache-dir
```
Then, I have compiled BootCommander and its associated library and copied to /usr/local/bin and usr/local/lib. I have tested BootCommander and it seems to be working fine. 
```
root@92c31f2530bd:/tmp# BootCommander -d=/dev/ttyUSB0 -b=115200 firmware.srec
--------------------------------------------------------------------------
BootCommander version 1.04.02. Performs firmware updates on a micro-
controller based system that runs the OpenBLT bootloader.

Copyright (c) 2017 by Feaser  http://www.feaser.com
-------------------------------------------------------------------------
Processing command line parameters...[OK]
Detected firmware file: firmware.srec
Detected session protocol: XCP version 1.0
Using session protocol settings:
  -> Timeout T1: 1000 ms
  -> Timeout T3: 2000 ms
  -> Timeout T4: 10000 ms
  -> Timeout T5: 1000 ms
  -> Timeout T6: 50 ms
  -> Timeout T7: 2000 ms
  -> Seed/Key file: None
  -> Connection mode: 0
Detected transport layer: XCP on RS232
Using transport layer settings:
  -> Device: /dev/ttyUSB0
  -> Baudrate: 115200 bit/sec
Loading firmware data from file...[OK]
  -> Number of segments: 3
  -> Base memory address: 0x08010000
  -> Total data size: 549988 bytes
Connecting to target bootloader...[TIMEOUT]
Attempting backdoor entry (reset system if this takes too long)...[OK]
Erasing 392 bytes starting at 0x08010000...[OK]
Erasing 322716 bytes starting at 0x08010190...[OK]
Erasing 226880 bytes starting at 0x0805ee30...[OK]
Programming 392 bytes starting at 0x08010000...[OK]
Programming 322716 bytes starting at 0x08010190...[OK]
Programming 226880 bytes starting at 0x0805ee30...[OK]
Finishing programming session...[OK]
```
Next, I have tested the recent installed firmware updater on its version v1.11.0b2
I've attached some screenshots:
[image]
[image]
[image]

Progress bar never change its state. I've tried manually reboting printer, adding M997 (I think this is not necessary) as pre-flash gcode, modifying 30 seconds timeout etc. It seems like is not doing anything.

No data in plugin_firmwareupdater_console.log
```
root@92c31f2530bd:/octoprint/octoprint/logs# ls
octoprint.log                           plugin_cancelobject_cancelobject.log  plugin_softwareupdate_console.log
plugin_DisplayLayerProgress_events.log  plugin_firmwareupdater_console.log    serial.log
plugin_PrintTimeGenius_engine.log       plugin_octolapse.log                  serial.log.2021-04-02_22-38-35
plugin_arc_welder.log                   plugin_pluginmanager_console.log
root@92c31f2530bd:/octoprint/octoprint/logs# cat plugin_firmwareupdater_console.log
root@92c31f2530bd:/octoprint/octoprint/logs#
```
FYI, I have Octoprint installed as a docker container. May be it is not supported, but I think it shouldn't be a problem.
It is likely something I'm doing wrong. I will recheck it and tell you something.

thisiskeithb, 2021-04-02:
> FYI, I have Octoprint installed as a docker container. May be it is not supported, but I think it shouldn't be a problem.
It is likely something I'm doing wrong. I will recheck it and tell you something.

Shouldn't be, but I have an ET5X (modded with a silent board, flashed with OpenBLT) that I can hook up to a test pi/octoprint install if needed.

benlye, 2021-04-03:
> No data in plugin_firmwareupdater_console.log

What's in octoprint.log?

benlye, 2021-04-03:
I see from your log that you had to use the 'backdoor' reset to get the board flashed from the command line.  The plugin should show you 'Trying bootloader backdoor...' in the status bar while it is in that state (curious - do you have to reset your board manually at this point?).  BootCommander seems to wait in that state forever, but the plugin will only wait 30s (or the configured timeout) before killing the process.  I can make the status message more verbose if you think it's necessary.

I'm hoping that your octoprint.log file will shed some light on why the flash never seems to start.  I'm expecting an unhandled Python exception somewhere...

davidtgbe, 2021-04-03:
> Shouldn't be, but I have an ET5X (modded with a silent board, flashed with OpenBLT) that I can hook up to a test pi/octoprint install if needed.

@thisiskeithb ok, that would help, thanks.

> What's in octoprint.log?

Logged lines related to firmware_updater plugin.
```
2021-04-02 22:37:05,603 - octoprint.plugins.firmwareupdater - INFO - Python binproto2 package installed: False
|  Firmware Updater (1.11.0b2) = /octoprint/plugins/lib/python3.8/site-packages/octoprint_firmwareupdater
2021-04-02 22:37:10,228 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
2021-04-02 22:37:10,712 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
2021-04-02 22:37:12,108 - octoprint.plugins.firmwareupdater - INFO - Setting BINARY_FILE_TRANSFER capability to False
```
I have meatpack plugin enabled and `BINARY_FILE_TRANSFER` disabled on marlin. If I understand correctly, this has nothing to be with serial flashing method, but, with transferring the firmware to the SD through Marlin.

I have launched a `TOP` window and BootCommander never launches, so, I have to assume it is an issue from my side. Moreover, `plugin_firmwareupdater_console.log` is not populated with data. Weird...

On the other hand. Let me first clarify the update process I follow:
1. Usually, an already flashed Marlin FW is currently running on the printer.
2. I execute BootCommander or MicroBoot (Windows client), configure baud rate, serial port and .srec file. Then, this client waits for a specific answer from OpenBLT, which is sent through serial port. This way, OpenBLT reports it is ready to receive data from PC.
3. I manually restart de printer. OpenBLT detects an opened communication on the serial port, and answers it is ready to receive data.
4. Client acknowledges and flashing process starts.

Question is, are you manually rebooting your dev board, or have you added a `M997` to PRE-FLASH GCODES?

As @thisiskeithb has reported, there is a problem when Marlin restarts the printer with a `M997`. Printer restarts and OpenBLT takes control fine, but, SD updating process is not starting. It seems remounting SD after a soft reset is not working fine, resulting in OpenBLT not noticing there is a firmware.srec to be flashed on the SD. 
I think this doesn't happen if the update process through serial port (which is the case), but, anyway, I have to take a look at this.

benlye, 2021-04-03:
What's in the octoprint.log file immediately after you try to flash using the BootCommander method? There should be more than the snippet you posted above.

My dev board isn't running Marlin, it doesn't have the flash space, so I'm just flashing an arbitrary 'blink' program to it. BootCommander resets it automatically when the flashing is done. I don't use M997 or any pre or post flash commands. 

You're correct that Binary File Transfer has nothing to do with this and isn't needed. I don't know what MeatPack might do. It might be worth trying with that disabled.

benlye, 2021-04-03:
I was thinking about this a little more...

I assume that your Marlin firmware isn't 'OpenBLT aware' - it hasn't been compiled with the OpenBLT code which enables the application to activate the bootloader - so it will always be necessary to use the 'backdoor' reset to activate the bootloader so BootCommander can connect.

To achieve this, I could try sending an `M997` to the serial port either immediately before (or possibly just after, but I doubt that would work) initiating the BootCommander command.  Otherwise you will have to reset your board manually once the firmware update status bar says 'Trying bootloader backdoor'.  I should change that message to something like 'Trying bootloader backdoor - reset your board now'.

In any case, none of this matters until we figure out why the BootCommander flash command isn't even starting on your system.

davidtgbe, 2021-04-03:
That's right @benlye. Marlin is not OpenBLT aware. Only option is to:
- First, start Boot Commander and leave it in "waiting for OpenBLT to be ready" state.
- Send a M997 to Marlin to perform a soft reboot, thus, passing control to OpenBLT. Or, manually reboot the printer. In each case, serial port must be freed, just after the M997, or just after the BootCommander is in waiting state.

Don't worry about my particular problem. I will dig into it next week. I have not much time right now, but it is very likely to be something wrong with my set up.

benlye, 2021-04-03:
I'll play around adding an optional M997 reset before the BootCommander command. I have a program I can run on my dev board which will reset the board if it receives M997.

Will await your feedback on the rest.

benlye, 2021-04-04:
v1.11.0b3 has a configurable option to reset the board (via an M997 command) prior to running BootCommander.  This should ensure that the board goes into the OpenBLT bootloader while BootCommander is waiting for it.  The option is enabled by default.

The board will likely take ~2s to reset so the 'Attempting backdoor' status will be displayed briefly.  I changed it to text similar to the command line message, "Trying bootloader backdoor - (reset the board if this takes too long)..."

davidtgbe, 2021-04-06:
@benlye, I think I need some help here. I hace discovered that there is no data logged because the error is triggered before, in the ajax POST call.

The file I am trying to upload (firmware.srec) is 1.57 MB. However, if I try with a smaller file, it works fine. There may be some limit to the maximum file size that can be uploaded. It seems to be a limit that can be changed somewhere, but, before continuing to investigate further, perhaps you can test in your environment, to rule out if you also have this problem.

I am attaching screenshots below the POST request

[image]
[image]

benlye, 2021-04-06:
You got it - the size limit for the flash API is currently 1MB.  I can increase it easily so I'll do that (I'm thinking 5MB is probably OK), but I'll also make the plugin give a better error if the file is too large.

I'll have a new release for you to try in a few hours.

thisiskeithb, 2021-04-06:
> I'm thinking 5MB is probably OK

The Biqu BX (and the BigTreeTech SKR BX-SE motherboard inside) has a 2mb STM32H743IIT6, so 5 should be fine for a while.

benlye, 2021-04-06:
Give 1.11.0b5 a go - limit is now 5MB.  This is also now configurable by adding a configuration key in config.yaml.  I didn't expose it in the UI for now but could easily add it later on.

davidtgbe, 2021-04-06:
@benlye I have been able to upload the FW. There are also some things I have noticed:

For example, `Reset before flashing` checkbox in `Advanced` tab, is not resetting the printer. I have tried to reproduce what I think script is doing:
```
root@92c31f2530bd:/octoprint/octoprint/logs# stty -F /dev/ttyUSB0 115200
root@92c31f2530bd:/octoprint/octoprint/logs# stty -a
speed 38400 baud; rows 43; columns 137; line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^U; eof = ^D; eol = <undef>; eol2 = <undef>; swtch = <undef>; start = ^Q; stop = ^S; susp = ^Z;
rprnt = ^R; werase = ^W; lnext = ^V; discard = ^O; min = 1; time = 0;
-parenb -parodd -cmspar cs8 -hupcl -cstopb cread -clocal -crtscts
-ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr icrnl ixon -ixoff -iuclc -ixany -imaxbel -iutf8
opost -olcuc -ocrnl onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
isig icanon iexten echo echoe echok -echonl -noflsh -xcase -tostop -echoprt echoctl echoke -flusho -extproc
root@92c31f2530bd:/octoprint/octoprint/logs# echo M997 >> /dev/ttyUSB0
```
Above `echo M997` command only works if Octoprint is connected to the printer. If not, `M997` doesn't take effect and printer doesn't reboot, having to rebooting it manually.
But, on the other hand, adding a `M997` in `Pre-flash GCode` field of `Pre-flash` tab, does reboot the printer. Overall upload process doesn't work because, inspite of printer being rebooting correctly, OpenBLT has already passed control to Marlin when BootCommander is executed.

Other randome problems I hace noticed are related to timeouts and some error return codes, status messages that seems not reflecting what BootCommander is actually doing and things like this. Perhaps, the only arise when flashing large firmware files like the one I'm trying to flash. I have to test it a little more and, play with timeouts and try to find a pattern.

Thanks for your work

benlye, 2021-04-06:
> Above echo M997 command only works if Octoprint is connected to the printer. If not, M997 doesn't take effect and printer doesn't reboot, having to rebooting it manually.

Strange - I would expect the opposite - it should only work when OctoPrint is _not_ connected, otherwise the serial port is blocked.  Doing it this way works for LPC boards and other STM32 boards.

I'm not surprised that it doesn't work in the Pre-Flash command as there is too much time elapsed between sending that and running BootCommander.

I won't do anything until you've had a chance to test more.

davidtgbe, 2021-04-12:
OK @benlye. I had some time today to give a look at this.

I've been trying to figure out why the printer only reboots if Octoprint is connected. I have tried to configure the serial port and communicate with the printer.

```
pi@octopi:~/openblt/openblt/Host $ sudo stty -F /dev/ttyUSB0 115200 -echo
115200
pi@octopi:~/openblt/openblt/Host $ echo M997 >> /dev/ttyUSB0
pi@octopi:~/openblt/openblt/Host $ cat /dev/ttyUSB0                            ▒}U▒▒8▒▒aee▒(▒j▒Ԉ΍~▒▒Ĉʍv▒}▒▒▒8▒u▒▒▒
```
Getting all of those garbage chars, I then realized that Marlin connects @250000 bauds, whereas OpenBLT does @115200. That's why having Octoprint connected it does work. Octoprint configures correctly serial port speed.

I then tried to configure the serial port @250000 bauds, but it seems it is not possible with stty

```
pi@octopi:~/openblt/openblt/Host $ sudo stty -F /dev/ttyUSB0 speed 250000 -echo
stty: invalid argument ‘250000’
```

Tried this solution and it works fine: https://unix.stackexchange.com/questions/327188/how-to-monitor-a-serial-connection-250000-baud

```
pi@octopi:~ $ ./mysetbaud.py <>/dev/ttyUSB0 250000
pi@octopi:~ $ echo M997 >> /dev/ttyUSB0
```

On other concurrent terminal
```
pi@octopi:~ $ tail -f /dev/ttyUSB0
echo:Unknown command: "▒▒▒▒▒▒▒"
ok P7 B3
 T:19.34 /0.0 T:18.96 /0.00 B:18.42 /0.00 @:0 B@:0
Not SD printing
Not SD printing
 T:19.32 /0.00 B:18.87 /0.00 @:0 B@:0
Not SD printing
Not SD printing
 T:19.55 /0.00 B:18.65 /0.00 @:0 B@:0
Not SD printing
Not SD printing
 T:19.26 /0.00 B:18.93 /0.00 @:0 B@:0
Not SD printing
Not SD printing
 T:18.96 /0.00 B:19.14 /0.00 @:0 B@:0
Not SD printing
Not SD printing
 T:19.18 /0.00 B:18.91 /0.00 @:0 B@:0
Not SD printing
Not SD printing
 T:18.95 /0.00 B:18.96 /0.00 @:0 B@:0
Not SD printing
start
Marlin bugfix-2.0.x

echo: Last Updated: 2021-03-15 | Author: (davidtgbe, ET4)
echo:Compiled: Mar 28 2021
echo: Free Memory: 66791  PlannerBufferBytes: 832
//action:notification Anet ET4 TMC2208 BMG BLTOUCH Lista
EEPROM loaded from slot 18.
//action:prompt_end
pages_ready
```

benlye, 2021-04-12:
It's probably easiest if I just change the reset to use pySerial with the same baud rate is the OctoPrint connection for the `M997` command.  

The historical reason for using echo should no longer apply with recent Marlin builds (it used to be very easy to [get a Pi to crash](https://github.com/OctoPrint/OctoPrint/issues/3683) by sending `M997` to it, but a [commit ](https://github.com/MarlinFirmware/Marlin/commit/004bed8a7fc3ff9feb73a0ea9794635b50073c27 )in the bugfix-2.0.x branch a few months ago has mitigated that problem) .

Will give you a new version to test later today.

benlye, 2021-04-12:
Try 1.11.0b6.  You will be required to have the printer connected before you can start a firmware update, and the M997 command will be sent to the connected port at the connected baud rate.

The printer needs to be connected before flashing starts so that we can deterministically know the correct baud rate - without an active connection we can only guess at what it might be based on preferences.

davidtgbe, 2021-04-27:
Hi @benlye!
Resuming this again after some time...
in the end I have set up a test environment to make changes and go testing faster. 
To sum up, basically I had to change the timeout limits for the command
```
< 120 or bootcmdr_command_timeout > 180:
```
and the place where the board has to be reset. It has to be done right after the command is started, when connection is going to be established.
I have also notied that, in the programming phase, log lines are not processed in real time. I don't know if it has to do with how the command's output is processed or if some output buffer has to be flushed (my Python skills are very limited, sorry). You can see what I mean in the video. Programming stage takes a long time to complete, but is not reflected neither in the log, and therefore , nor in the progress bar.

https://streamable.com/rrkaxj

benlye, 2021-04-27:
Could you make a video of an  update just running bootcommander from the command line so I can see what 'normal' output looks like?

Getting output from commands is not always simple.  You can change the timeout that is used when reading stdout out here:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/devel/octoprint_firmwareupdater/methods/bootcmdr.py#L83

Change this:
`output = p.stdout.read(timeout=0.1).decode('utf-8')`

To something like this:
`output = p.stdout.read(timeout=0.02).decode('utf-8')`

I don't know if that will fix it - you can try experimenting with different values.

davidtgbe, 2021-04-27:
https://streamable.com/oacltw

Notice the timestamp difference between the actual command output and the logged by the plugin. Also special chars like LFs ESC and so on are included.

[bootcommander.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6387777/bootcommander.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6387778/plugin_firmwareupdater_console.log)

If I change the timeout as you suggested, it starts logging the programming phase, but ends with this status:
```
Flashing failed
BootCommander returned code -9
```
Killing the BootCommander process and leaving the flashing unfinished.

Log: 
[plugin_firmwareupdater_console_timeout_001.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6387937/plugin_firmwareupdater_console_timeout_001.log)

I can't see that output code looking at the source code: https://github.com/feaser/openblt/blob/master/Host/Source/BootCommander/main.c

benlye, 2021-04-28:
Hmm, that seems like it's hitting the command timeout and the process is getting killed. 

The intention with the timeout was only to stop it sitting waiting for the bootloader forever. I'll have a look and see if i can make it so that once erasing/programming starts the timeout doesn't trigger.

benlye, 2021-04-28:
Can you try updating bootcmdr.py with this section?  It has the shorter stdout read timeout, and the command timeout should only apply when bootcommander is waiting for the bootloader.  This should mean that the default 30s command timeout is OK.

```python

        while p.returncode is None:
            output = p.stdout.read(timeout=0.02).decode('utf-8')
            if not output:
                p.commands[0].poll()
                if (connecting and time.time() > (starttime + bootcmdr_command_timeout)):
                    self._logger.error(u"Timeout waiting for command to complete")
                    p.commands[0].kill()
                    raise FlashException("Connection timeout")
                continue

            for line in output.split("\n"):
                if line.endswith("\r"):
                    line = line[:-1]
                self._console_logger.info(u"> {}".format(line))

                if WRITING in line:
                    connecting = False
                    self._logger.info(u"Writing memory...")
                    self._send_status("progress", subtype="writing")
                if CONNECTING in line:
                    connecting = True
                    self._logger.info(u"Connecting to bootloader...")
                    self._send_status("progress", subtype="connecting")
                if BACKDOOR in line:
                    connecting = True
                    self._logger.info(u"Connecting to backdoor...")
                    self._send_status("progress", subtype="backdoor")
                if ERASING in line:
                    connecting = False
                    self._logger.info(u"Erasing memory...")
                    self._send_status("progress", subtype="erasing")
                if FINISHING in line:
                    connecting = False
                    self._logger.info(u"Finishing programming...")
                    self._send_status("progress", subtype="finishing")
```

If you still have problems can you post the plugin's console log and the octoprint log.

davidtgbe, 2021-05-03:
I think we have a fully working version.
https://streamable.com/e5ghvd

Fell free to change what you consider.
[bootcmdr.zip](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6416164/bootcmdr.zip)

benlye, 2021-05-04:
I put your changes into v1.11.0b8, please give it a go.
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/releases/tag/1.11.0b8

davidtgbe, 2021-05-06:
It's working. This is the configuration I've used.
[image]

benlye, 2021-05-16:
Firmware Updater v1.11.0 has been released.

closed 2021-05-16 by benlye