OctoPrint/OctoPrint-FirmwareUpdater #232: Advanced tab in settings doesn't show anything

cp2004, 2021-03-27:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/0b93248d1b4642ee1443cf6fa961b7b662f1d5e7/octoprint_firmwareupdater/templates/firmwareupdater_settings.jinja2#L104

It looks like there is no `#advanced` anywhere as one of the tabs.

[image]
[image]

Any more questions let me know 😄

benlye, 2021-03-27:
Can't reproduce that here...
[image]

Advanced tab is defined on line 275:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/octoprint_firmwareupdater/templates/firmwareupdater_settings.jinja2#L275

Do the other tabs work?

I wonder if there is some other element in the UI on your system with the id 'advanced' and maybe I need to use more unique identifiers?  Do you have other plugins enabled that might be causing a problem?  Can you try disabling them?

cp2004, 2021-03-27:
I have a decent number of plugins installed - not sure how I didn't find the definition of advanced tab before. I searched, promise...

For my plugins, if I remember I try and prefix the IDs I use if they are common. Something like `#FWU-advanced`?

Just checked in the devtools, it appears to be the spaghetti detective that also has an 'advanced' ID - at least, that's *one* of them, the one jquery is picking up.

benlye, 2021-03-27:
I should have seen that coming.  I use unique IDs on the modals but forgot to do it when I implemented the settings tabs.

I'll do a dev version later on today with unique IDs for those elements.  If you could test it and confirm it fixes this I'll do a stable release.

cp2004, 2021-03-27:
Thank you 😄 
I also asked TSD if they would do the same, to reduce the likelihood of things like this happening again. Advanced is a fairly common tab name.

benlye, 2021-03-27:
Should be fixed in v1.10.10b3 in the development release channel.

cp2004, 2021-03-27:
Tested, works great 😄 
Shame I can't be this fast in plugin development...

benlye, 2021-03-27:
Fixed in v1.10.10.

closed 2021-03-27 by benlye