OctoPrint/OctoPrint-FirmwareUpdater #234: [FR]

mriscoc, 2021-03-29:
**Hardware Setup**
Ender 3v2

**Describe the problem**
I use the plug-in with my firmware. In it I have enabled to list .bin files, but to be able to see the bin files in the Octoprint file list, I also must to install another plugin, then after flashing I can easily go to the file list and delete the bin file. So, can you so a config to enable the listing of binary files in the Octoprint file list? Thank you.
[image]

benlye, 2021-03-29:
What's the name of the other plugin that let's you see bin files in the list?

mriscoc, 2021-03-29:
Marlin Binary File Transfer https://plugins.octoprint.org/plugins/marlinbft/
[image]

benlye, 2021-03-29:
OK, I'll see what I can do.

mriscoc, 2021-03-29:
> OK, I'll see what I can do.

Thank you.

benlye, 2021-03-30:
Are you sure that the `fw123420.bin` file in your first screenshot is really on your SD card?

I've got the MarlinBFT plugin installed and it does not seem to enable listing .bin and .cur files from the SD card in my board.  Running `M20` in the terminal to list the SD card files only returns .gcode files, which makes me think that this isn't an OctoPrint limitation - the printer firmware isn't telling the host that those files are there.

Can you post the output of `M20` on your printer?

E.g., this is what I get from `M20`:
```
Send: M20
Recv: Begin file list
Recv: FILE.GCO 0
Recv: FILE~1.GCO 0
Recv: End file list
Recv: ok
```

This is what the File Explorer shows:
[image]

But this is what's really on the SD card:
```
pi@octopi:/ $ ls -al /media/usb/
total 206
drwxr-xr-x  2 pi   pi    16384 Jan  1  1970 .
drwxr-xr-x 10 root root   4096 Dec  4 16:03 ..
-rw-rw-rw-  1 pi   pi        0 Mar 30 09:22 file.gco
-rw-rw-rw-  1 pi   pi        0 Mar 30 09:22 file.gcode
-rw-rw-rw-  1 pi   pi   189060 Jan  1  2000 FIRMWARE.CUR
-rw-rw-rw-  1 pi   pi        0 Mar 30 09:22 foobar.bin
-rw-rw-rw-  1 pi   pi        0 Mar 30 09:22 foobar.srec
-rw-rw-rw-  1 pi   pi        0 Mar 30 09:22 fw123456.bin
```

As far as I can tell this is the way Marlin is designed to work, and non-gcode files will never be listed from the SD card.

mriscoc, 2021-03-30:
> Are you sure that the `fw124320.bin` file in your first screenshot is really on your SD card? Do you see it if you enable 'Only show files stored on SD Card'?
> 
> [image]
> 
> I've got the MarlinBFT plugin installed and it does not seem to enable listing .bin and .cur files from the SD card in my board. Running `M20` in the terminal to list the SD card files only returns .gcode files, which makes me think that this isn't an OctoPrint limitation - the printer firmware isn't telling the host that those files are there.
> 
> Can you post the output of `M20` on your printer?
> 
> E.g., this is what I get from `M20`:
> 
> ```
> Send: M20
> Recv: Begin file list
> Recv: FILE.GCO 0
> Recv: FILE~1.GCO 0
> Recv: End file list
> Recv: ok
> ```
> 
> This is what the File Explorer shows:
> [image]
> 
> But this is what's really on the SD card:
> 
> ```
> pi@octopi:/ $ ls -al /media/usb/
> total 206
> drwxr-xr-x  2 pi   pi    16384 Jan  1  1970 .
> drwxr-xr-x 10 root root   4096 Dec  4 16:03 ..
> -rw-rw-rw-  1 pi   pi        0 Mar 30 09:22 file.gco
> -rw-rw-rw-  1 pi   pi        0 Mar 30 09:22 file.gcode
> -rw-rw-rw-  1 pi   pi   189060 Jan  1  2000 FIRMWARE.CUR
> -rw-rw-rw-  1 pi   pi        0 Mar 30 09:22 foobar.bin
> -rw-rw-rw-  1 pi   pi        0 Mar 30 09:22 foobar.srec
> -rw-rw-rw-  1 pi   pi        0 Mar 30 09:22 fw123456.bin
> ```
> 
> As far as I can tell this is the way Marlin is designed to work, and non-gcode files will never be listed from the SD card.

Yes, my fork allows to see binary files on the card.

benlye, 2021-03-30:
> Yes, my fork allows to see binary files on the card.

You might have mentioned that up front. I'm hesitant to add a feature that will only work with forked firmware. 

Is there any particular reason why you need to delete the firmware file manually after flashing? The plugin will do it automatically if timestamp based filenames are enabled.

mriscoc, 2021-03-30:
> > Yes, my fork allows to see binary files on the card.
> 
> You might have mentioned that up front. I'm hesitant to add a feature that will only work with forked firmware.
> 
> Is there any particular reason why you need to delete the firmware file manually after flashing? The plugin will do it automatically the next time a firmware update is run, if timestamp based filenames are enabled.

[image]

Sorry, I did think that it was clear.

closed 2021-03-30 by mriscoc