OctoPrint/OctoPrint-FirmwareUpdater #241: [Feature Request] Support MKS Rumba 32

twproject, 2021-04-12:
Please add support to MKS Rumba32 or generally to board that need to put manually in BOOT mode.

MKS Rumba32 procedure is 
Press and hold the BOOT button, then use USB connect to the MKS RUMBA32. The motherboard enters DFU mode. 

After that i found /dev/ttyACM0 under raspberry devices

benlye, 2021-04-12:
So.... what do you need the plugin to do?

Can you put the board into DFU mode via the boot button and then do a firmware update using the existing dfu-programmer flash method?

twproject, 2021-04-12:
sorry I wrote too quickly. 

I made some test from shell and the needs is to add dfu-util support

From bash i was able to upload firmware via command:
dfu-util -a 0 -s 0x8000000:leave -D /home/pi/firmware/firmware.bin

benlye, 2021-04-12:
OK, so I need to add a new flash method for dfu-util.  That won't be too hard.  

Give me a day or two and I'll have something for you to try out.

twproject, 2021-04-12:
Thanks,
 I try "hacking" dfu-programmer mode with no luck
[image]
[image]

flash command is: sudo {dfuprogrammer} -a 0 -s 0x8000000:leave -D {firmware}

error is:
[image]

benlye, 2021-04-12:
Try changing ` {dfuprogrammer}` to the full path to `dfu-util` on your system.

twproject, 2021-04-12:
I tried passing all parameters but error is the same

```
2021-04-12 16:25:52,104 - octoprint.plugins.firmwareupdater - INFO - Printer port: undefined
2021-04-12 16:25:52,121 - octoprint.plugins.firmwareupdater - INFO - Flash method: dfuprogrammer
2021-04-12 16:25:52,145 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2021-04-12 16:25:52,159 - octoprint.plugins.firmwareupdater - INFO - Running 'echo "start flash"' in /usr/local/bin
2021-04-12 16:25:52,215 - octoprint.plugins.firmwareupdater - INFO - Running 'sudo /usr/local/bin/dfu-util -a 0 -s 0x8000000:leave -D /home/pi/firmware/firmware.bin' in /usr/local/bin
2021-04-12 16:25:52,771 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. dfu-programmer returned code 1.
```

I copied the command and run into a shell, it works correctly

```
sudo /usr/local/bin/dfu-util -a 0 -s 0x8000000:leave -D /home/pi/firmware/firmware.bin
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Match vendor ID from file: 0000
Match product ID from file: 0000
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 127800
Erase           [=========================] 100%       127800 bytes
Erase    done.
Download        [=========================] 100%       127800 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST stat
```

benlye, 2021-04-12:
Is there anything in the plugin's console log?

twproject, 2021-04-12:
yes

```
2021-04-12 16:25:52,160 echo "start flash"
2021-04-12 16:25:52,218 sudo /usr/local/bin/dfu-util -a 0 -s 0x8000000:leave -D /home/pi/firmware/firmware.bin
2021-04-12 16:25:52,769 > sudo: no tty present and no askpass program specified
2021-04-12 16:25:52,770 >
```

benlye, 2021-04-12:
I thought that might be it - you have a sudo problem.  

You could try configuring it to run without the `sudo` prefix on the command line, or add a sudoers entry allowing pi to run dfu-util without a password.  I don't know if you'll actually need sudo for the command to succeed so you'll have to try it...

twproject, 2021-04-12:
ok, with this hack it works also if i upload the firmware via browse button

this is now the command: sudo /usr/local/bin/dfu-util -a 0 -s 0x8000000:leave -D {firmware}

benlye, 2021-04-12:
Does it work without sudo?

twproject, 2021-04-12:
I didn't test, I added /etc/sudoers.d/020_firmware_updater file and it works

twproject, 2021-04-12:
It doesn't work without sudo.
I check also on shell and it require sudo to see the device

```
pi@octopi:~/.octoprint/logs $ sudo /usr/local/bin/dfu-util -l
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0483:df11] ver=2200, devnum=25, cfg=1, intf=0, path="1-1.2", alt=3, name="@Device Feature/0xFFFF0000/01*004 e", serial="STM32FxSTM32"
Found DFU: [0483:df11] ver=2200, devnum=25, cfg=1, intf=0, path="1-1.2", alt=2, name="@OTP Memory /0x1FFF7800/01*512 e,01*016 e", serial="STM32FxSTM32"
Found DFU: [0483:df11] ver=2200, devnum=25, cfg=1, intf=0, path="1-1.2", alt=1, name="@Option Bytes  /0x1FFFC000/01*016 e", serial="STM32FxSTM32"
Found DFU: [0483:df11] ver=2200, devnum=25, cfg=1, intf=0, path="1-1.2", alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,03*128Kg", serial="STM32FxSTM32"
pi@octopi:~/.octoprint/logs $
```

benlye, 2021-04-12:
Awesome.  With all the info you've pasted above I can add the new method.  Shouldn't take long.

benlye, 2021-04-14:
Please try v1.11.0b7 from the development branch.

twproject, 2021-04-14:
Hi, yes it works
Do i need to test removing /etc/sudoers.d/020_firmware_updater file?

benlye, 2021-04-14:
> Hi, yes it works

Great!

> Do i need to test removing /etc/sudoers.d/020_firmware_updater file?

No, you'll still need sudo rights to use dfu-util, so you will still need the entry in that file.

I'll write some documentation and release this version soon.

benlye, 2021-05-16:
Firmware Updater v1.11.0 has been released.

closed 2021-05-16 by benlye