OctoPrint/OctoPrint-FirmwareUpdater #244: MarlinFTP not recognized by Plugin

Gathaeryx, 2021-05-05:
**Hardware Setup**
Printer: Creality Ender3
Board: SKR Mini E3 V1.2

**Describe the problem**
The plugin keeps giving an error that the Python MBP package is not installed:
[image]

I have installed both the Python 2.7 and 3 versions of marlin-binary-protocol, and trying to reinstall them just gives the regular "Requirement already satisfied..." messages for both MBP and Heatshrink. 

**Log files**
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6426850/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6426852/plugin_pluginmanager_console.log)

benlye, 2021-05-05:
Just a guess, but are you sure you activated the Python venv (or ran `pip` from inside the venv) when you installed the packages?

Can you post the actual commands and output from your installation attempts?

Gathaeryx, 2021-05-05:
I feel like an idiot, you are 100% correct. I forgot to install in the virtual env.

closed 2021-05-05 by Gathaeryx