OctoPrint/OctoPrint-FirmwareUpdater #245: systemd-udevd not present on latest octoprint

strayr, 2021-05-05:
**Hardware Setup**
SKR Mini E3, Pi 3b

**Describe the problem**
systemd-udevd not present in latest octopi
```
strayr@edgepi:~ $ sudo systemd-
systemd-analyze                 systemd-mount
systemd-ask-password            systemd-notify
systemd-cat                     systemd-path
systemd-cgls                    systemd-resolve
systemd-cgtop                   systemd-run
systemd-delta                   systemd-socket-activate
systemd-detect-virt             systemd-stdio-bridge
systemd-escape                  systemd-sysusers
systemd-hwdb                    systemd-tmpfiles
systemd-id128                   systemd-tty-ask-password-agent
systemd-inhibit                 systemd-umount
```
Looks like your documentation is out of date, I'd love to be of help here, but last time I was current Upstart was new and exciting :-/ I suspect it's a simple change but I'm embarrassingly out of touch.

**Log Files**
n/a

**Additional context**
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/lpc176x.md#configure-systemd-udevd
I can't use the marlin binary file transfer system as it conflicts with meatpack which i'm looking at to solve a print issue, this is a too lazy for sneakernet issue :D

benlye, 2021-05-05:
It's a service, not an executable, you won't find it like that.

Follow the instructions as they are written.

strayr, 2021-05-05:
/facepalm
what i needed
`sudo service systemd-udevd --full-restart`
what happened
`sudo systemd-udevd --full-restart`

additionally almost everything else in systemd has a executable in $PATH
not YOUR problem

closed 2021-05-05 by strayr