OctoPrint/OctoPrint-FirmwareUpdater #249: Binary File Transfer: seemingly no progress when flashing from file

Gathaeryx, 2021-05-16:
Marlinbft doesn't seem to flash the firmware.bin file correctly: After clicking "Flash from file", the "Flashing firmware" bar appears, but nothing changes, the LCD on the printer does not change, and it doesn't seem like the firmware is actually flashing. How long should I be waiting for the firmware to actuallly flash?

The prerequisites are met: 
[image]
[image]

[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6489175/octoprint.log)

benlye, 2021-05-16:
The log you've attached does not include any attempts to flash firmware to the printer.  Please try again and attach a new log.

The status should change to 'Sending firmware to printer', then 'Resetting board'.  When the firmware is being sent the LCD should look something like this:

[image]

The method works, I use it all the time, so something in your setup is causing the problem.  A log containing a flash attempt will help.

Gathaeryx, 2021-05-16:
That's the problem, the flashing attempt doesn't even show up in the logs.. Here is a small excerpt from the octoprint.log file from a few minutes ago, at about 16:04:30 I tried to flash the firmware, but nothing seems to happen at all, and nothing gets logged.

`2021-05-16 16:04:17,125 - octoprint.server.util.flask - INFO - Passively logging in user Marvin from ::ffff:192.168.2.50
2021-05-16 16:04:17,126 - octoprint.access.users - INFO - Logged in user: Marvin
2021-05-16 16:04:18,863 - octoprint.server.util.sockjs - INFO - User Marvin logged in on the socket from client ::ffff:192.168.2.50
2021-05-16 16:06:44,932 - octoprint.server.api.system - INFO - Performing command for core:restart: sudo service octoprint restart
2021-05-16 16:06:45,128 - octoprint.server - INFO - Shutting down...
2021-05-16 16:06:45,660 - octoprint.server - INFO - Calling on_shutdown on plugins`

benlye, 2021-05-16:
I've no idea what's going on then.  If there's no log, the plugin isn't being called by the frontend.  The only thing I can think of is something else on your system (maybe another plugin) is upsetting the user interface / javascript side of things.

This is what it should look like:

https://user-images.githubusercontent.com/23322579/118402545-3fe5b100-b662-11eb-8d21-6214aa3ac035.mp4

Can you open the browser's developer console and look for Javascript errors (Ctrl+Shift+J in most Chrome-based browsers)? Otherwise, try disabling any other plugins you have running.

Gathaeryx, 2021-05-16:
Your video is about what I expected should happen, but it only shows the static bar in my initial post.

Trying to flash with the console open does indeed result in a 403 error:
[image]

I have also tried to flash firmware with all other plugins (besides the default Octoprint ones) disabled, with the same result.

benlye, 2021-05-16:
Can you try creating a new user account (with admin permissions)?

The only reason that /flash should give a 403 is lack of admin rights, but that should also be detected by the frontend UI.

Gathaeryx, 2021-05-16:
Created a new account with Admin Permissions, got the same 403 error message.
[image]

Gathaeryx, 2021-05-16:
O wow.. I just found out the difference between Admin Groups and Admin Permissions.. My account only had the latter. Adding my account to the Admin Group fixed it. 
Thanks for the help!

benlye, 2021-05-16:
Ah, cool. Glad you figured it out. I'll see if I can reproduce that so I can make the front end give a useful error message.

closed 2021-05-17 by benlye