OctoPrint/OctoPrint-FirmwareUpdater #251: Option to Not Rename File

jneilliii, 2021-05-30:
**Hardware Setup**
Stock FLSUN Q5 Board, MKS Robin Nano based I believe.

**Describe the problem**
With the FLSUN Q5 board, the bootloader only recognizes a specific filename, that is not firmware.bin. 

**Log Files**
[octoprint-logs (2).zip](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6565552/octoprint-logs.2.zip)

**Additional context**
For me filename is Robin_nano.bin. Just thought it would be easier to allow the plugin to have an option to not change the name of what's been uploaded.

jneilliii, 2021-05-30:
I think I may have figured out that this is possible through config.yaml, just can't seem to find it as an option in the UI.

jneilliii, 2021-06-01:
Seems this feature may still be bugged. After setting my config.yaml it appears to attempt to transfer with the correct filename, but doesn't transfer with error.

```
  firmwareupdater:
    _config_version: 2
    _selected_profile: 0
    has_bftcapability: true
    has_binproto2package: true
    profiles:
    - _name: Default
      flash_method: marlinbft
      marlinbft_use_custom_filename: true
      marlinbft_custom_filename: Robin_nano.bin
      marlinbft_no_m997_reset_wait: true
      serial_port: /dev/ttyUSB0
```

The relevant log entries are listed below.
```
2021-05-31 22:36:06,962 - octoprint.plugins.firmwareupdater - INFO - Current Baud: 250000
2021-05-31 22:36:06,962 - octoprint.plugins.firmwareupdater - INFO - Initializing Marlin BFT protocol
2021-05-31 22:36:06,964 - octoprint.plugins.firmwareupdater - INFO - pySerial Version: 3.5
2021-05-31 22:36:06,995 - octoprint.plugins.firmwareupdater - INFO - Connecting to printer at '/dev/ttyUSB0' using Marlin BFT protocol
2021-05-31 22:36:06,998 - octoprint.plugins.firmwareupdater - INFO - Connecting: Switching Marlin to binary protocol...
2021-05-31 22:36:07,024 - octoprint.plugins.firmwareupdater - INFO - Connection synced [0], binary protocol version 0.1.0, 96 byte payload buffer
2021-05-31 22:36:07,025 - octoprint.plugins.firmwareupdater - INFO - Transfering file to printer using Marlin BFT '/tmp/tmpm8bgdc2l' -> /Robin_nano.bin
2021-05-31 22:36:07,118 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Unable to transfer file.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_firmwareupdater/methods/marlinbft.py", line 98, in _flash_marlinbft
    filetransfer.copy(firmware, target, True, False)
  File "/home/pi/oprint/lib/python3.7/site-packages/binproto2/protocols.py", line 407, in copy
    self.open(dest_filename, compression_support, dummy)
  File "/home/pi/oprint/lib/python3.7/site-packages/binproto2/protocols.py", line 372, in open
    raise Exception("Cannot open file on client")
Exception: Cannot open file on client
```

Is this the reason why these settings are not exposed to the UI yet?

jneilliii, 2021-06-01:
After reading into issue https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/190 I realize that this may be because the Robin_nano.cur file remains on the SD card and deleting files only exists when the timestamp option is enabled. Wonder if with the custom filename option if the file with the same name but `.cur` extension could be removed ahead of the transfer maybe.

benlye, 2021-06-01:
I'm fairly sure that Marlin Binary File Transfer will happily overwrite an existing file, but it definitely will not transfer files with names longer than 8 characters.  This is not my limit, but something in the current version of the protocol.

You will need to use the lpc176x method if you want to write that file to the SD card.  (lpc176x exposes the option to use custom file names in the GUI.)

I guess I should add a filename length check for the marlinbft method...

jneilliii, 2021-06-01:
Thanks @benlye I wonder if @p3p would know anything about the filename length and if that's going to change or not? Especially with the long filename support coming down the line with Marlin.

p3p, 2021-06-01:
Unfortunately it's not a limitation in the transfer protocol either, it uses Marlins SD filesystem on the client directly, If Marlin can't create the file then the protocol forwards the error.

Marlins SD filesystems long filename support (at least at the time I wrote the transfer protocol) was very limited and buggy.

EvilGremlin, 2021-06-17:
+1 for UI option, unobtainium Elegoo Neptune 2 also wants elegoo.bin

benlye, 2021-06-18:
Just released v1.12.0b0 in the Development release channel with support for customizing the filename when using BFT.  The filename must be in 8dot3 format and the plugin will warn / error if it is not.

benlye, 2021-09-02:
v1.12.0 released with configurable file name for BFT uploads.

closed 2021-09-02 by benlye