OctoPrint/OctoPrint-FirmwareUpdater #252: Can't flash Marlin after flashing klipper

jexinator, 2021-05-30:
Good evening, I'll keep this as short as possible.

I have a Sovol SV01 printer with the Silent Board 2.2.1 (Atmega 2560) and a Raspberry Pi 3B+. Initial firmware was Marlin.

I wanted to try klipper and mainsail/Fluid. Followed guides online, install via KIAUH. Everything went well. Liked the environment but wanted to go back to Marlin.
Didn't know how to do it and there is no guide about that. I installed Octoprint and the Firmware Updater thinking it would solve my problem but I keep getting `stk500_getsync() attempt 1 of 10: not in sync: resp=0x00`.
Mainsail/Fluid and Octoprint (with octoklipper) connect to the printer without problem but when I attempt to flash a firmwaren nothing seems to work.
I'm tempted to go the GPIO route but not sure if it will work. Did I miss something ?

Anyway, I join my log files.
[octoprint (1).log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6566915/octoprint.1.log)
[plugin_firmwareupdater_console (2).log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6566916/plugin_firmwareupdater_console.2.log)

> 2021-05-30 21:57:56,251 /usr/bin/avrdude -v -q -p m2560 -c arduino -P /dev/ttyUSB0 -D -b 115200 -U flash:w:/tmp/tmpk5me3tpf:i
2021-05-30 21:57:56,963 > 
2021-05-30 21:57:56,964 > avrdude: Version 6.3-20171130
2021-05-30 21:57:56,965 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2021-05-30 21:57:56,966 >          Copyright (c) 2007-2014 Joerg Wunsch
2021-05-30 21:57:56,967 > 
2021-05-30 21:57:56,968 >          System wide configuration file is "/etc/avrdude.conf"
2021-05-30 21:57:56,969 >          User configuration file is "/home/pi/.avrduderc"
2021-05-30 21:57:56,970 >          User configuration file does not exist or is not a regular file, skipping
2021-05-30 21:57:56,971 > 
2021-05-30 21:57:56,972 >          Using Port                    : /dev/ttyUSB0
2021-05-30 21:57:56,973 >          Using Programmer              : arduino
2021-05-30 21:57:56,974 >          Overriding Baud Rate          : 115200
2021-05-30 21:57:56,975 > 
2021-05-30 21:58:03,021 > avrdude: stk500_recv(): programmer is not responding
2021-05-30 21:58:03,022 > avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
2021-05-30 21:58:03,023 > 
2021-05-30 21:58:13,685 
2021-05-30 21:58:13,689 /usr/bin/avrdude -v -q -p m2560 -c arduino -P /tmp/printer -D -b 115200 -U flash:w:/tmp/tmp_lx4xkom:i
2021-05-30 21:58:14,513 > 
2021-05-30 21:58:14,515 > avrdude: Version 6.3-20171130
2021-05-30 21:58:14,515 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2021-05-30 21:58:14,516 >          Copyright (c) 2007-2014 Joerg Wunsch
2021-05-30 21:58:14,517 > 
2021-05-30 21:58:14,518 >          System wide configuration file is "/etc/avrdude.conf"
2021-05-30 21:58:14,520 >          User configuration file is "/home/pi/.avrduderc"
2021-05-30 21:58:14,521 >          User configuration file does not exist or is not a regular file, skipping
2021-05-30 21:58:14,522 > 
2021-05-30 21:58:14,522 >          Using Port                    : /tmp/printer
2021-05-30 21:58:14,523 >          Using Programmer              : arduino
2021-05-30 21:58:14,524 >          Overriding Baud Rate          : 115200
2021-05-30 21:58:14,525 > ioctl("TIOCMGET"): Invalid argument
2021-05-30 21:58:14,526 > ioctl("TIOCMGET"): Invalid argument
2021-05-30 21:58:14,527 > 
2021-05-30 21:58:20,319 > avrdude: stk500_recv(): programmer is not responding
2021-05-30 21:58:20,321 > avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
2021-05-30 21:58:20,322 > 

> 2021-05-30 21:57:56,191 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile index: 0
2021-05-30 21:57:56,195 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile name: Default
2021-05-30 21:57:56,196 - octoprint.plugins.firmwareupdater - INFO - Printer port: /dev/ttyUSB0
2021-05-30 21:57:56,200 - octoprint.plugins.firmwareupdater - INFO - Flash method: avrdude
2021-05-30 21:57:56,213 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2021-05-30 21:57:56,235 - octoprint.plugins.firmwareupdater - INFO - Running '/usr/bin/avrdude -v -q -p m2560 -c arduino -P /dev/ttyUSB0 -D -b 115200 -U flash:w:/tmp/tmpk5me3tpf:i' in /usr/bin
2021-05-30 21:58:03,029 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Avrdude says: 'not in sync: resp=0x00'.
2021-05-30 21:58:13,517 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile index: 0
2021-05-30 21:58:13,521 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile name: Default
2021-05-30 21:58:13,522 - octoprint.plugins.firmwareupdater - INFO - Printer port: /tmp/printer
2021-05-30 21:58:13,528 - octoprint.plugins.firmwareupdater - INFO - Flash method: avrdude
2021-05-30 21:58:13,665 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2021-05-30 21:58:13,685 - octoprint.plugins.firmwareupdater - INFO - Running '/usr/bin/avrdude -v -q -p m2560 -c arduino -P /tmp/printer -D -b 115200 -U flash:w:/tmp/tmp_lx4xkom:i' in /usr/bin
2021-05-30 21:58:20,327 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Avrdude says: 'not in sync: resp=0x00'.

Have nice evening

benlye, 2021-05-31:
I've no idea about that specific board, but the programmer for a mega2560 is typically 'wiring'.  Try that instead of 'arduino' with the /dev/ttyUSB0 port.

jexinator, 2021-06-01:
In the end I solved my issue (with some help); I forgot to stop klipper and moonracker service before flashing firmware.
Anyway, I'am gonna close the issue.

closed 2021-06-01 by jexinator