OctoPrint/OctoPrint-FirmwareUpdater #253: Only English after flashing the firmware

Frank-3D, 2021-06-01:
I just flashed my Prusa MK3S to firmware version 3.1.0. I did that via octoprint with fimrware updater (both newest versions). The upgrade went smoothly. Then the printer presented its language menu. If I select German, the printer resets itself and ends up in the language menu again. Only when I select English I get to the main menu and everything runs normally, just in English.
This problem does not occur when flashing with the original Prusa Slicer.

benlye, 2021-06-01:
Your problem is nothing to do with the updater plugin, you need to choose the correct firmware to flash.

closed 2021-06-01 by benlye

Frank-3D, 2021-06-01:
I think that's not right! Prusa only offers one multilingual hex file for all languages ​​(version 3.1.0). The firmware is installed correctly with the prusa slicer. But it´s of course your decision to evaluate this information.