OctoPrint/OctoPrint-FirmwareUpdater #254: SKR-2 board updates from SD Card, but doesn't mount drive; Marlinbft fails

nezil, 2021-06-03:
Hi,

I think I've come to the conclusion, based on the title of this post, that the new SKR-2 board needs to use Marlin Binary File Transfer to perform updates from OctoPi because the SD Card cannot be mounted as a drive with this board.

I have set up all the required pre-requisites for this mode of operation, and the log files show the file is transferred over the link. The board also resets effectively, and LEDs appear to flash suggesting that the firmware is being loaded during the reboot, but everything stops at that point.

I believe that the firmware.bin file may be getting corrupted during the transfer, and this causes a corrupt image to load and of course prevent re-connection.

Has the Marlin Binary File Transfer option been updated to the latest implementation in Marlin firmware? I'm running 2.0.8.1 currently. 

Below is a the log file information, but it really just shows that a file was sent, and then that OctoPrint cannot reconnect after the reset:

`2021-06-03 20:26:08,974 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 99.85% 28.15KiB/s [34.22KiB/s] Errors: 0
2021-06-03 20:26:08,979 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 99.90% 28.14KiB/s [34.21KiB/s] Errors: 0
2021-06-03 20:26:08,982 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 99.95% 28.14KiB/s [34.21KiB/s] Errors: 0
2021-06-03 20:26:08,982 - octoprint.plugins.firmwareupdater - INFO - PROGRESS: 100.00% 28.14KiB/s [34.21KiB/s] Errors: 0
2021-06-03 20:26:08,997 - octoprint.plugins.firmwareupdater - INFO - File closed
2021-06-03 20:26:08,998 - octoprint.plugins.firmwareupdater - INFO - Transfer complete
2021-06-03 20:26:08,998 - octoprint.plugins.firmwareupdater - INFO - Binary file transfer complete
2021-06-03 20:26:10,009 - octoprint.plugins.firmwareupdater - INFO - Firmware update reset: attempting to reset the board
2021-06-03 20:26:10,014 - octoprint.plugins.firmwareupdater - INFO - Resetting printer at '/dev/ttyACM0'
2021-06-03 20:26:10,113 - octoprint.plugins.firmwareupdater - INFO - Waiting for printer at '/dev/ttyACM0' to reset
2021-06-03 20:26:10,575 - octoprint.plugins.firmwareupdater - INFO - Printer at '/dev/ttyACM0' is resetting
2021-06-03 20:26:33,750 - octoprint.plugins.firmwareupdater - ERROR - Timeout waiting for board reset to complete
2021-06-03 20:26:33,750 - octoprint.plugins.firmwareupdater - ERROR - Board reset failed
2021-06-03 20:26:33,757 - octoprint.plugins.firmwareupdater - ERROR - Reset failed
2021-06-03 20:26:33,762 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyACM0, baudrate=250000, profile={'axes': {'e': {'inverted': False, 'speed': 300}, 'x': {'inverted': False, 'speed': 6000}, 'y': {'inverted': False, 'speed': 6000}, 'z': {'inverted': False, 'speed': 200}}, 'color': 'default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}, 'heatedBed': True, 'heatedChamber': False, 'id': '_default', 'model': 'SV01', 'name': 'Sovol SV01', 'volume': {'custom_box': False, 'depth': 260.0, 'formFactor': 'rectangular', 'height': 300.0, 'origin': 'lowerleft', 'width': 280.0}}
2021-06-03 20:26:33,840 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"
2021-06-03 20:26:33,852 - octoprint.util.comm - INFO - Connecting to port /dev/ttyACM0, baudrate 250000
2021-06-03 20:26:33,881 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial connection" to "Error"
2021-06-03 20:26:33,883 - octoprint.util.comm - INFO - Changing monitoring state from "Error" to "Offline after error"
2021-06-03 20:26:33,885 - octoprint.util.comm - ERROR - Unexpected error while connecting to serial port /dev/ttyACM0, baudrate 250000 from hook default: SerialException: '[Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'' @ comm.py:_open_serial:3699
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyACM0'`

nezil, 2021-06-03:
I performed another test... Using the option to rename the file that is needed for the Ender 3, I was able to transfer the file, have the board reset, but not actually load the firmware because the filename is ignored by the boot loader.

The file sizes are identical, but a 'diff' command shows that they do differ.

Next I will try sending a simple text file to see if the transfer will corrupt that data as well.

benlye, 2021-06-03:
I don't know anything about your board, but I have had no other reports of file corruption using BFT. I flash my board regularly with the latest Marlin 2.0.x bugfix code and haven't had any problems.

You might want to try compiling your firmware with a lower baud rate (e.g. 115200) and configuring Octoprint to use that.

FWIW, the firmware updater plugin uses an external library to implement BFT, so if there are issues with the protocol, they are probably outside my control (hence the big warnings around using this method).

nezil, 2021-06-03:
OK... very strange... I disabled the file name check, and sent a G-Code file, again, using the Ender 3 option which resulted in the file being saved as FW211534.BIN. The board rebooted and ignored the file as expected.

I unmounted the SD Card and put it in my computer, renaming the file FW211534.gcode and then taking a look at the file. Initially I thought everything transferred fine, but here is an excerpt from the transferred file:

`^@^@; generated by PrusaSlicer 2.3.0+ on 2021-06-02 at 22:27:25 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.35mm
; first layer extrusion width = 0.42mm

M73 P0 R103
M201 X1000 Y1000 Z250 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ;^@^@ets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E4.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec`

And the same excerpt from the original file:
`; generated by PrusaSlicer 2.3.0+ on 2021-06-02 at 22:27:25 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.35mm
; first layer extrusion width = 0.42mm

M73 P0 R103
M201 X1000 Y1000 Z250 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E4.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec`

I'm not sure what's causing this, but there are sporadic broken characters throughout the file.

nezil, 2021-06-03:
> I don't know anything about your board, but I have had no other reports of file corruption using BFT. I flash my board regularly with the latest Marlin 2.0.x bugfix code and haven't had any [...]

Thanks for taking a look @benlye. My firmware is compiled with a baud rate of 115200. I do understand the warnings, and my initial reaction was that maybe that was the issue. I reached out on here in case it was a _known_ issue, and that seems not to be the case.

Unfortunately it's difficult for me to use any version of Marlin before 2.0.8.1 because support for my board was added at that point. I could use a Bugfix build, but 2.0.8.1 is only a week or two old. What data of bug fix did you last try with?

benlye, 2021-06-03:
bugfix is the development branch, it is newer than any release version.  It's probably worth trying.

Are you certain your buadrate is already 115200?  The log above shows OctoPrint trying to connect at 250k.

nezil, 2021-06-03:
> bugfix-2.0.x is the development branch, it is newer than any release version. It's probably worth trying.
> 
> Are you certain your buadrate is already 115200? The log above shows OctoPrint trying to connect at 250k.

Good point... my firmware is certainly 115200, but I'd read somewhere that when connecting over USB this was irrelevant. I'll double check this and try again. It is certainly strange that anything worked at all if there was a mis-match of baud rate.

nezil, 2021-06-03:
Unfortunately this results in exactly the same issue.... the board reboots, loads the corrupt firmware and then is unresponsive.

benlye, 2021-06-03:
Do you have any other plugins loaded?  Can you disable them?

You could try using this plugin to transfer the file:
https://plugins.octoprint.org/plugins/marlinbft/

It uses the same Python library to implement the protocol, so should suffer from the same problem.

nezil, 2021-06-03:
> Do you have any other plugins loaded? Can you disable them?
> 
> You could try using this plugin to transfer the file:
> https://plugins.octoprint.org/plugins/marlinbft/
> 
> It uses the same Python library to implement the protocol, so should suffer from the same problem.

I did look at that plugin, but it says that it's currently unmaintained and with a known issue on the latest Marlin... I can certainly try it, but I was concerned about those two points.

I'm building a Bugfix 06-02-2021 firmware now and will try that first.

nezil, 2021-06-03:
OK, same issue with the 06-02-2021 Bugfix firmware.

benlye, 2021-06-03:
What about disabling any other OctoPrint plugins?

nezil, 2021-06-03:
> What about disabling any other OctoPrint plugins?

Trying that now. 

I have just tried the other Marlin Binary Transfer plugin, but it doesn't work at all. It seems to suggest that the prerequisites aren't there, even though they clearly are, so I cannot select a file.

nezil, 2021-06-03:
OK, absolutely all plugins disabled apart from Firmware Updater... same result.

ddovod, 2021-06-08:
Have you tried to disable "Reset before flashing" flag in the Advanced settings? It helped me with skr 1.4T
#242

nezil, 2021-06-09:
I was really excited when I saw a new comment that might have been a solution, but unfortunately not... “Reset before flashing” is only an option for the LPC176X method, which doesn’t work on the SKR 2.0 because the SD card isn’t mountable over USB for whatever reason.

It would be nice if someone else could test Marlin binary transfer in the latest Marlin 2.0.8.2 build with any board and see if it is corrupting transfers. It would also be great if someone else could test with an SKR 2.0, but it’s very new so that may take a while.

benlye, 2021-06-09:
> It would be nice if someone else could test Marlin binary transfer in the latest Marlin 2.0.8.2 build with any board and see if it is corrupting transfers. It would also be great if someone else could test with an SKR 2.0, but it’s very new so that may take a while.

We've already tested builds newer than 2.0.8.2 (bugfix-2.0.x) and found that you still have the problem but I do not.  I'm fairly sure that this isn't a systemic problem, but something only in your environment / configuration - if it was systemic it's likely that there would have been other reports.

Does the file corruption always happen in the same place(s) in the files?  Is it always the same characters?  Have you tried something as simple as changing the USB cable?

@p3p, do you have any ideas?

nezil, 2021-06-09:
> We've already tested builds newer than 2.0.8.2 (bugfix-2.0.x) and found that you still have the problem but I do not. I'm fairly sure that this isn't a systemic problem, but something only in your environment / configuration - if it was systemic it's likely that there would have been other reports.

I'm sorry, it wasn't clear from your earlier post that you had tested newer builds. Bugfix-2.0.x is not sufficient information to determine that because the Bugfix versions are updated almost daily, and the version number has been Bugfix-2.0.x since the earliest days of Marlin 2.0. Stable 2.0.8.2 was only released just a few days before my original issue post, but I have tested now with 2.0.8.1, 2.0.8.2, and Bugfix from 06-02-2021. 

> Does the file corruption always happen in the same place(s) in the files? Is it always the same characters? Have you tried something as simple as changing the USB cable?

I'm going to test this now with a text file and see if that is the case or not. 

I have changed the USB cable, yes. My initial testing was with the board on a bench, it's now installed in my printer and working well in every other respect. 

I'm pretty annoyed with BTT for not supporting LPC176X mounting on this board to be honest. I'm not sure if that's a Marlin incompatibility or a real hardware issue.

> @p3p, do you have any ideas?

Any help would be most appreciated.

p3p, 2021-06-09:
I'm unsure what could cause an issue on a specific board, if I'm looking at the correct board it seems to be native USB (the USB data packets get all the way to the MCU rather than being processed into UART by another chip, if it does have native usb there's no reason it cant support host OS sdcard mounting if the STM32 framework support it)

Transmission errors are unlikely (USB protocol has internal error detection and flow control) but it all depends how the STM32 framework is processing the data from the MCUs USB peripheral before it gets to Marlin, on top of that the BFT protocol has checksums per data chunk (defaults to 96Bytes) as well, it should tell you the chunk size in the log.

So in theory the only ways to get corruption in the destination file are if the Host actually sent corrupt data or during the SD card write (though SPI transmission errors, an issue in Marlins Filesystem, or data buffer corruption)

If the corruption is on BFT chunk size boundaries (every 96 bytes by default) it may show an issue in the BFT buffers, with compression enabled it will write the sdcard in 512Byte chunks, so if the corruption is every 512Bytes it may indicate an issue with file writes.

You could disable BFT transmission compression and reduce the chunk size to see if that fixes, or changes the error distribution.

benlye, 2021-06-09:
> So in theory the only ways to get corruption in the destination file are if the Host actually sent corrupt data or during the SD card write (though SPI transmission errors, an issue in Marlins Filesystem, or data buffer corruption)

I suppose the browser upload could somehow be corrupting the file as it is being sent to the host or the Pi could be corrupting it on save. The plugin removes the temp file at the end of the process, but that could easily be disabled so that it could be checked on the Pi's file system.

nezil, 2021-06-09:
So I've done the test that @benlye suggested, and sent the same file multiple times to see if the corruption is predictable, and it is...

I used a text file filled with Lorem Ipsum text, and sent it multiple times. Every time I read the sent file, the same letters / spaces are missing.

nezil, 2021-06-09:
> I suppose the browser upload could somehow be corrupting the file as it is being sent to the host or the Pi could be corrupting it on save. The plugin removes the temp file at the end of the [...]

Aahhh, this _could_ be the issue. I'm using Safari, and a few years back I had a similar issue loading a firmware file through the browser to another device. That was an HDMI switcher / combiner system that had a web interface, but the upload would always fail. I could try using Chrome through windows and see if the result is the same. Stand by...

nezil, 2021-06-09:
Yes.... it worked from Windows & Chrome!

What's strange is that the other forms of firmware update do not have this issue with Safari. I only recently changed from 8b to 32b, and I never had this problem through Safari before.

nezil, 2021-06-09:
Now that we know this is a Safari interoperability issue, should I close this one and write up a new issue?

benlye, 2021-06-09:
> Now that we know this is a Safari interoperability issue, should I close this one and write up a new issue?

You should close this issue, yes. 

As for another issue... I need evidence that this is happening to other Safari users. 

I'm still thinking this is something in your environment - you can't be the only one out of 15k users of the plugin using Safari, and nobody else has reported this problem.

nezil, 2021-06-09:
> You should close this issue, yes.
>
> As for another issue... I need evidence that this is happening to other Safari users.
>
> I'm still thinking this is something in your environment - you can't be the [...]

I understand, but I also wouldn't be surprised... Although there are 15k users of the plugin, the combination of those 15k users that are a) macOS users b) Safari browser users c) Marlin Binary File Transfer users that d) bother to open an issue, is likely very small. Marlin Binary File Transfer alone is built in to almost no Marlin builds that get shared / posted online, so it would need to be someone who was capable of building Marlin from source as well. I know this isn't difficult, but with each permutation, the number of likely users goes down.

If I'm correct, I believe that this might be the same Safari issue that I found a couple of years back, and it was solved with that product / software.

benlye, 2021-06-09:
I just flashed my BTT board from Safari on macOS with no problems at all.

In the absence of any other  people reporting a similar problem, I'm standing firm on this being a problem with your setup.

nezil, 2021-06-09:
> I just flashed my BTT board, using BFT, from Safari, on macOS with no problems at all.
> 
> In the absence of any other people reporting a similar problem, I'm standing firm on this being a problem with your setup / environment.

Understood. Thanks for trying. I do have another Mac that I can test with. My daily driver is an Intel based MacBook Pro, running Big Sur, but I also have an M1 Mac mini, also running Big Sur.

nezil, 2021-06-10:
OK, I'm embarrassed to write this, but I tested the update using BFT using my M1 Mac mini this morning, and found no issues at all. 

I have to conclude, as @benlye said, that something is strange with my Intel based Mac system. I'll try and have a look at any differences present with Safari settings, but I'm not sure if I'll find anything. There is nothing special about my Intel Mac installation, it was a clean Big Sur install on a 2020 13" Intel MacBook Pro. 

@benlye and I have now both tested successfully with other versions of macOS (M1 in my case, Intel, but earlier than Big Sur in @benlye's case). There is still a possibility that this is an issue with Big Sur on Intel Macs, so if anyone continues to have this issue I'm documenting the status for the record.

Closing for now though.

closed 2021-06-10 by nezil