OctoPrint/OctoPrint-FirmwareUpdater #258: Firmware updater stuck on "Writing Memory"

Wimpie1000, 2021-06-16:
Hi there. I have used Firmware updater successfully on two occasions but wanted to update my Marlin to 2.0 on my Anet A8 with BL-Touch enabled. The updater starts as normal but got stuck on "Writing Memory" I then had to restart with the board obviously now bricked. I did not have all the components to flash a new bootloader so I installed my spare mainboard and hoped I did something wrong the first time. I however tried to flash it again now, but the updater is stuck again on "Writing Memory" It is now 2 hours but no change yet. Can you please help me? It will mean this board will now also be bricked. I should have my components to flash a new Bootloader in a day or 2 but then I would still like to flash my firmware with the updater. Otherwise I will have to do it with an Arduino like everyone suggests I should do.
[image]

benlye, 2021-06-16:
Need to see the logs from the flash attempts.

I no longer have an Anet A8 board, but when I did I flash it *many* times using the plugin and never had any problems.

benlye, 2021-06-16:
But the point is that this is a newly compiled hex file which you have never written successfully before...

Can you also attach the firmware updater console log, please.

benlye, 2021-06-16:
> Oops now I will humbly ask where I can get the log file for this one.

`.octoprint/logs/plugin_firmwareupdater_console.log`

Wimpie1000, 2021-06-16:
Ok cool I have located the log file but is not sure how to copy the file from within putty or is there another way to copy the file?

benlye, 2021-06-16:
You can also download it from the Logging page in he OctoPrint settings.

Wimpie1000, 2021-06-17:
Hi there Ben. Ok cool I have managed to download it, please find the 2 logs of yesterday attached. Thank you again for your help.
[octoprint-logs (1).zip](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/6670864/octoprint-logs.1.zip)

benlye, 2021-06-17:
Your firmware image is too big for the Anet A8 board with the original Anet bootloader.  The file you are flashing is 129076 bytes but the maximum file size for an Anet board with the stock bootloader is 126976 bytes - flashing this hex file is corrupting the board's bootloader.

(I'm assuming that because you have had to order the USBasp you need to recover the board you have never replaced the stock bootloader with the much smaller Optiboot bootloader - a USBasp is required to update the bootloader.)

Once you have your USBasp you have two options:
1. Put the original bootloader back then remove features from your firmware to get it below 126976 bytes
2. Flash the OptiBoot bootloader and keep your firmware below 130048 bytes

You can use the Arduino IDE along with the Anet board package here to replace the bootloader:
https://github.com/benlye/anet-board

That board supports both stock and OptiBoot bootloaders but I would **strongly** recommend flashing the OptiBoot bootloader.

Wimpie1000, 2021-06-17:
Wow that is awesome!! Can I compliment you that you are very thorough!! I have seen a video where they say we can load the optiboot to make it smaller. I think it is by crosslink. :-) I will follow your recommendation and then flash the Optiboot to be able to use the BLtouch. Thank you so much.

Wimpie1000, 2021-06-17:
Ok cool, I have linked up my USBASP and followed the instructions on your github page and was met with this error message attached. Am I doing something wrong?
[image]

benlye, 2021-06-17:
Ignore the first message - don't bother trying to update the FW on the USBasp.

Check your cabling and make sure you are on the correct pins on the board - you'll need a 10 pin to 6 pin adapter or a lot of jumper wires. There are a lot of YT videos showing the cabling. 

The drivers for these things can be tricky so maybe Google that too.

Wimpie1000, 2021-06-18:
Ok cool I tried with jumper wires but settled to get an adapter, maybe my problem lies there. ☹ I should have it in 4hrs time and then try it out. 

Meanwhile I will try and tinker and see if I can manage. One questions though, Is it necessary to have the board powered up when flashing the bootloader? 
My assumption is that it was not necessary as you get 5V from the USBASP. 👀

benlye, 2021-06-18:
Power should come just from the USBasp, you should disconnect external power and the USB cable.  This video looked helpful - wiring is around 7:30.
https://www.youtube.com/watch?v=3jm0zzIneFc

This video covers using Zadig to get the libusbk driver installed - just make sure you select the USBasp device before installing the driver:
https://www.youtube.com/watch?v=mmp9aqowEYs

Wimpie1000, 2021-06-18:
Hi there. Ok cool I have managed to successfully flash the bootloader. However when I connect the printer the COM port does not show up in the Arduino ISP, when I check my device manager it shows USBSerial. Do I need to load any special drivers?

benlye, 2021-06-18:
If memory serves me correctly, you may need the CH340G driver.

Wimpie1000, 2021-06-18:
Ok I checked around a bit and discovered I needed to install the CH341 driver. I have done it and voila!! I have now successfully flashed Marlin 2.0. :-) I just hope the BL touch will work. Baby steps but getting there. Thank you so much for your time and help, I highly appreciate it.
[image]
[image]

closed 2021-06-20 by benlye