OctoPrint/OctoPrint-FirmwareUpdater #260: binary transfer method - BTT GTR 1.0 - Marlin 2.0.9

MMoudry3, 2021-06-29:
**Hardware Setup**
BTT GTR 1.0 - Marlin 2.0.9 - Ender 5 Plus (heavily modified)

**Describe the problem**
When trying to flash the firmware using the binary transfer method it doesn't work. 

**Log Files**
octoprint.log:
2021-06-29 15:06:06,347 - octoprint.plugins.excluderegion - INFO - Setting update detected: g90InfluencesExtruder=False, clearRegionsAfterPrintFinishes=True, mayShrinkRegionsWhilePrinting=True, loggingMode=octoprint, enteringExcludedRegionGcode=None, exitingExcludedRegionGcode=None, extendedExcludeGcodes={'M73': {"gcode": "M73", "type": "ExcludedGcode", "mode": "merge", "description": "Suppress progress updates while excluding and output the most recent progress values encountered when exiting the excluded area"}, 'M117': {"gcode": "M117", "type": "ExcludedGcode", "mode": "last", "description": "Suppress display messages while excluding and output the last message encountered when exiting the excluded area"}, 'G4': {"gcode": "G4", "type": "ExcludedGcode", "mode": "exclude", "description": "Ignore all dwell commands in an excluded area to reduce delays while excluding"}, 'M205': {"gcode": "M205", "type": "ExcludedGcode", "mode": "merge", "description": "Record advanced setting changes while excluding and apply the most recent values in a single command after exiting the excluded area"}, 'M204': {"gcode": "M204", "type": "ExcludedGcode", "mode": "merge", "description": "Record default acceleration changes while excluding and apply the most recent values in a single command after exiting the excluded area"}}, atCommandActions={'ExcludeRegion': [{"action": "enable_exclusion", "type": "AtCommandAction", "command": "ExcludeRegion", "parameterPattern": "^\\s*(enable|on)(\\s|$)", "description": "Default action to enable exclusion"}, {"action": "disable_exclusion", "type": "AtCommandAction", "command": "ExcludeRegion", "parameterPattern": "^\\s*(disable|off)(\\s|$)", "description": "Default action to disable exclusion"}]}
2021-06-29 15:06:11,812 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile index: 0
2021-06-29 15:06:11,820 - octoprint.plugins.firmwareupdater - INFO - Firmware update profile name: Default
2021-06-29 15:06:11,821 - octoprint.plugins.firmwareupdater - INFO - Printer port: /dev/ttyACM0
2021-06-29 15:06:11,832 - octoprint.plugins.firmwareupdater - INFO - Flash method: marlinbft
2021-06-29 15:06:11,836 - octoprint.plugins.firmwareupdater - INFO - Python package 'marlin-binary-protocol' is installed: True
2021-06-29 15:06:11,839 - octoprint.plugins.firmwareupdater - INFO - Marlin BINARY_FILE_TRANSFER capability is enabled: True
2021-06-29 15:06:11,863 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2021-06-29 15:06:11,866 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2021-06-29 15:06:11,897 - octoprint.plugins.marlinbft - INFO - Unsetting capability BINARY_FILE_TRANSFER
2021-06-29 15:06:11,959 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2021-06-29 15:06:11,966 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2021-06-29 15:06:11,986 - octoprint.plugins.marlinbft - INFO - Unsetting capability BINARY_FILE_TRANSFER
2021-06-29 15:06:12,005 - octoprint.plugins.firmwareupdater - INFO - Current Baud: 250000
2021-06-29 15:06:12,009 - octoprint.plugins.firmwareupdater - INFO - Initializing Marlin BFT protocol
2021-06-29 15:06:12,014 - octoprint.plugins.firmwareupdater - INFO - pySerial Version: 3.5
2021-06-29 15:06:12,037 - octoprint.plugins.firmwareupdater - INFO - waiting 3s after protocol connect
2021-06-29 15:06:35,055 - octoprint.plugins.firmwareupdater - INFO - Connecting to printer at '/dev/ttyACM0' using Marlin BFT protocol
2021-06-29 15:06:35,058 - octoprint.plugins.firmwareupdater - INFO - Connecting: Switching Marlin to binary protocol...
2021-06-29 15:07:15,069 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Unable to connect to printer.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmwareupdater/methods/marlinbft.py", line 85, in _flash_marlinbft
    protocol.connect()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/binproto2/protocols.py", line 163, in connect
    self._send(0, 1)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/binproto2/protocols.py", line 221, in _send
    raise ConnectionLost()
ConnectionLost
2021-06-29 15:07:15,100 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyAMA0, baudrate=250000, profile={'heatedChamber': False, 'name': 'FDM_Ender5Plus', 'color': 'default', 'axes': {'y': {'speed': 6000, 'inverted': False}, 'x': {'speed': 6000, 'inverted': False}, 'z': {'speed': 200, 'inverted': False}, 'e': {'speed': 100, 'inverted': False}}, 'heatedBed': True, 'volume': {'origin': 'lowerleft', 'formFactor': 'rectangular', 'depth': 341.0, 'width': 350.0, 'custom_box': False, 'height': 400.0}, 'model': 'Ender 5 Plus Mipam Mod', 'id': '_default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}}
2021-06-29 15:07:15,156 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"
2021-06-29 15:07:15,164 - octoprint.util.comm - INFO - Connecting to port /dev/ttyAMA0, baudrate 250000
2021-06-29 15:07:15,170 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial connection" to "Connecting"
2021-06-29 15:07:15,176 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-06-29 15:07:15,189 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-06-29 15:07:15,205 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2021-06-29 15:07:15,226 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-06-29 15:07:15,239 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
2021-06-29 15:07:15,245 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
2021-06-29 15:07:15,321 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 2.0.9 (Jun 29 2021 15:48:38)"
2021-06-29 15:07:15,340 - octoprint.plugins.firmwareupdater - INFO - Setting BINARY_FILE_TRANSFER capability to True
2021-06-29 15:07:15,486 - octoprint.plugins.tracking - INFO - Sent tracking event printer_connected, payload: {u'printer_baudrate': 250000, u'printer_port': '/dev/ttyAMA0', u'firmware_name': u'Marlin 2.0.9 (Jun 29 2021 15:48:38)'}
2021-06-29 15:07:15,715 - octoprint.plugins.marlinbft - INFO - Setting BINARY_FILE_TRANSFER capability to True
2021-06-29 15:07:15,721 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2021-06-29 15:07:15,732 - octoprint.util.comm - INFO - Firmware states that it supports emergency GCODEs to be sent without waiting for an acknowledgement first
2021-06-29 15:07:15,740 - octoprint.util.comm - INFO - Firmware states that it supports sd status autoreporting

plugin_firmwareupdater_console.log:
- Nothing is written in it

**Additional context**
Pre-requisites are satisfied.

MMoudry3, 2021-06-29:
Just an off-topic question, do any of the other methods work with the BTT GTR 1.0? I can't seem to find any info anywhere on it.

benlye, 2021-06-29:
It looks like you are using the wrong port.

From the firmware update log:
`2021-06-29 15:06:35,055 - octoprint.plugins.firmwareupdater - INFO - Connecting to printer at '/dev/ttyACM0' using Marlin BFT protocol`

But from the connection entry:
`2021-06-29 15:07:15,164 - octoprint.util.comm - INFO - Connecting to port /dev/ttyAMA0, baudrate 250000`

Why aren't you flashing the firmware using `/dev/ttyAMA0`?

> do any of the other methods work with the BTT GTR 1.0

No idea, sorry.

MMoudry3, 2021-06-29:
Hi, Thanks so much for your reply. Changing the port fixed my problem. Also thank you for putting the effort into this plugin, hacking my printer would be so much harder without it.

M.

MMoudry3, 2021-06-29:
Resolved

closed 2021-06-29 by MMoudry3